fwupd/fwupd #1549: Fedora 31: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

luyatshimbalanga, 2019-11-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3

```

```shell
$ fwupdmgr get-devices
│
├─Samsung SSD 860 EVO 1TB:
│     Device ID:           ff5bf9f6641211b248157b73007e1b638b3f0a05
│     Summary:             ATA Drive
│     Current version:     RVT03B6Q
│     Vendor:              Advanced Micro Devices, Inc. [AMD]
│     GUIDs:               199fd920-dab0-5f9e-96fa-88130f6f4677
│                          8f72ef92-2740-56c2-8025-bac145539815
│                          bcaa2f53-dc18-5430-afaa-561ca1cff361
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           a3d4c2fe79a3b6a95e12ce4585feef633eb5773e
      Current version:     15.66.0
      Minimum Version:     15.66.0
      Vendor:              HP
      GUID:                785e408d-f2b9-469f-b7c1-519bdf9099b3
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Cryptograp
```

```shell
$ efibootmgr -v
Timeout: 0 seconds
BootOrder: 0004,0000,0001,0002,9999
Boot0000* Fedora	HD(2,GPT,918e2fac-8a28-42d0-8a3b-3b41a8827bc5,0x40800,0x82000)/File(\EFI\fedora\shimx64.efi)
Boot0001* Windows Boot Manager	HD(2,GPT,918e2fac-8a28-42d0-8a3b-3b41a8827bc5,0x40800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...s................
Boot0002* Internal Hard Disk	PciRoot(0x0)/Pci(0x8,0x2)/Pci(0x0,0x0)/Sata(1,65535,0)/HD(2,GPT,918e2fac-8a28-42d0-8a3b-3b41a8827bc5,0x40800,0x82000)..BO
Boot0004* Fedora	HD(1,GPT,f0ec8669-c099-4184-8d9f-d48026e5d383,0x800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot9999* USB Drive (UEFI)	PciRoot(0x0)/Pci(0x1d,0x0)/USB(16,0)..BO

```

```shell
$ efivar -l | grep fw
N/A
```

```shell
$ tree /boot
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtep```

Please answer the following questions:

- Operating system and version:  Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): bundled by the distribution
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No, SATA SSD.
- Is secure boot enabled (only for the UEFI plugin)? Yes

luyatshimbalanga, 2019-11-10:
According to /usr/libexec/fwup/fwup --vebose
```
02:56:45:0448 FIXME                failed to allocate dbus proxy object: Error calling StartServiceByName for com.intel.tss2.Tabrmd: Timeout was reached
ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
```

superm1, 2019-11-10:
this sounds like it might be the same as #1353.  Could you do the following:

1) Remove tpm2-abrmd and see if the issue resolves itself.
2) If it did resolve itself, please reinstall it and report the issue here: https://github.com/tpm2-software/tpm2-abrmd

luyatshimbalanga, 2019-11-10:
The issue resolved itself after removing tpm2-abrmd. By this time, I reported the issue to tpm2-abrmd.

superm1, 2019-11-11:
Great, thanks for confirming.

closed 2019-11-11 by superm1

tuxedotje, 2020-04-21:
in Fedora 32 beta still same problem,

ginsa, 2020-05-03:
Problem in Fedora 32 relaesed

ginsa, 2020-05-03:
Problem in Fedora 32 released, also after tpm2-abrmd installation

barrel0luck, 2020-05-20:
I don't mean to pester the developers, but this issue should be opened again.

hughsie, 2020-05-20:
I'm not sure what we can do, other than create our own libraries for communicating with the shared TPM.

gigitux, 2020-05-24:
Same problem on Fedora 32

Sirut, 2020-07-04:
Any solution for this trouble? everytime i turn my pc this error it's always on

drp4positive, 2026-10-04:
Still got it with fedora 44. Should I post the whole infoblock?

> $ fwupdmgr --version
compile   info.libusb                   1.0.30
compile   com.hughsie.libxmlb           0.3.29
compile   org.freedesktop.Passim        0.1.12
compile   org.freedesktop.fwupd         2.1.8
runtime   com.hughsie.libxmlb           0.3.29
runtime   org.kernel                    7.2.8-200.fc44.x86_64
runtime   org.freedesktop.fwupd         2.1.8

> ...