fwupd/fwupd #1551: No releases found for device: Firmware version formats were different, device was 'pair' and release is 'bcd'

NoSync, 2019-11-10:
It seems that the Thunderbolt controller on a Dell XPS 13 9380 (early 2019) is for some reason not upgradable:

fwupdmgr get-releases 250183f57432d0940e116ac2b40d2f4a87ce7d37
No releases found for device: Firmware version formats were different, device was 'pair' and release is 'bcd'

--

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.3.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.2
```

```shell
$ fwupdmgr get-devices
PS 13 9380
│
├─XPS 9380 Thunderbolt Controller:
│     Device ID:           250183f57432d0940e116ac2b40d2f4a87ce7d37
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     40.00
│     Vendor:              Dell (TBT:0x00D4)
│     Flags:               internal|updatable|require-ac|supported|registered
│     GUID:                0f401ed2-b847-532a-adc8-3193fc737be6
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:           f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:             NVM Express Solid State Drive
│     Current version:     1B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│   
├─System Firmware:
│     Device ID:           6c24a747f97668873b761558e322398a91dbf394
│     Current version:     0.1.8.0
│     Minimum Version:     0.1.8.0
│     Vendor:              Dell Inc.
│     Update State:        success
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                ce945437-7358-49f1-95d8-6b694a10a755
│   
└─TPM 2.0:
      Device ID:           2e59657d5e0beb043a16639968ca6178bcedff42
      Summary:             Platform TPM device
      Current version:     7.2.0.1
      Vendor:              Dell Inc.
      Update Error:        Updating disabled due to TPM ownership
      Flags:               internal|require-ac|registered
      GUIDs:               d621e687-b00d-539a-aeac-ac3eb530cbc5
                           7d65b10b-bb24-552d-ade5-590b3b278188
                           6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d
                           fe462d4a-e48f-5069-9172-47330fc5e838
```

Please answer the following questions:

- Operating system and version: Manjaro stable
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-11-11:
Can you have a try with refreshing your metadata and running that again?

```
# fwupdmgr refresh
# fwupdmgr get-releases 250183f57432d0940e116ac2b40d2f4a87ce7d37
```

NoSync, 2019-11-11:
After a refresh it works as expected:

─XPS 9380 Thunderbolt Thunderbolt Controller:
      New version:         44.00
      Remote ID:           lvfs
      Summary:             Update for the Thunderbolt host controller in a XPS 9380
      License:             proprietary
      Size:                241,7 kB
      Vendor:              Dell Inc.
      Flags:               is-upgrade
      Description:         Dfu64W, Dfu32W 4.0.23: Fixes/Enhancements
      
      ------------------------------------------
      
       • Missing Windows Manifest to require Administrator Privileges
       • Update MST Version Format on DFU to match with vendor release version
       • Dock Tools did show the update-to-date and the exit button
      
      when the RBU flag of dock package is not set in policy
      
       • Multiple EC payloads to show in Flash Utility UI
       • Add staging firmware update support
      
      DfuParser 1.0.7: Fixes/Enhancements

closed 2019-11-11 by superm1