fwupd/fwupd #1555: No Remotes Configured

Queuecumber, 2019-11-11:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**INSERT OUTPUT HERE**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
client version:	1.3.3
compile-time dependency versions
	gusb:	0.2.11
	efivar:	36
daemon version:	1.3.3
```

Please answer the following questions:

- Operating system and version: Clear Linux 
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Came with OS
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes 
- Is secure boot enabled (only for the UEFI plugin)? No

I'm giving clear linux a shot and when I try to run any relevant fwupdmgr commands I get "No remotes configured" as the output. This came with the OS and I think they botched the install, there is no directory `/var/lib/fwupd/remotes.d` for example. I copied that from the repository and restarted `fwupd.service` but no luck still the same output. Is there a straightforward way to fix this without a complete reinstall?

superm1, 2019-11-11:
It's very recent that they got fwupd enabled again (it was only because of some features added to fwupd 1.3.3 that they were willing to do it).

> This came with the OS and I think they botched the install, there is no directory /var/lib/fwupd/remotes.d

It's possible.  they've done some weird stuff in the past (See the crash they cause by returning `NULL` without setting `error` here: https://github.com/clearlinux-pkgs/fwupd/blob/master/no-poking-in-config-space.patch)

Could you do the following:

```
# fwupdmgr modify-config VerboseDomains "*"
# fwupdmgr get-remotes
# fwupdmgr refresh
```

And then share the journal output for `fwupd.service` and the command line output from those commands?  I think that should help us figure out where the mistake is made.

superm1, 2019-11-11:
Although looking at their spec, it's a little interesting.  They um, don't use `/etc`.  They set sysconfdir to `--sysconfdir=/usr/share/fwupd/`

And then they put remotes here into `/usr/share/fwupd/fwupd/remotes.d/`
https://github.com/clearlinux-pkgs/fwupd/blob/master/fwupd.spec#L250

Which I guess works, but any package updates will blow away your configured settings.
They have

Queuecumber, 2019-11-11:
yeah the more time i spend with Clear the more I think its a good idea that is executed in a really weird way (and with a lot of lawyers involved)

```
fwupdmgr modify-config VerboseDomains "*" 
Restart the daemon to make the change effective? [Y|n]: y
Successfully modified configuration value
```

```
fwupdmgr get-remotes
No remotes configured
```

```
fwupdmgr refresh
No remotes configured
```

Quite verbose journal output:

```
Nov 11 16:57:57 wigglytuff systemd[1]: Starting Firmware update daemon...
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 Fwupd                looking for os-release at /etc/os-release
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 FuConfig             loading /usr/share/fwupd/metainfo
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 XbSilo               file: 72ba2454-72bf-54c5-88d9-4d5fa136e1e2, current:72ba2454-72bf-54c5-88d9-4d5fa136e1e2, cached: (null)
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 XbSilo               loading silo with file contents
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 FuConfig             using config path of /etc/fwupd
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 FuConfig             path /etc/fwupd/remotes.d does not exist
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuEngine             client certificate exists and working
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHistory            got schema version of 5
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuIdle               setting timeout to 0s
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property Manufacturer=Dell Inc.
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property EnclosureKind=a
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property Family=XPS
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property ProductName=XPS 15 9560
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property ProductSku=07BE
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BiosVendor=Dell Inc.
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BiosVersion=1.16.0
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BiosMajorRelease=01
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BiosMinorRelease=10
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0306 FuHwids              smbios property BaseboardProduct=05FFDN
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0308 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               file: 3149702c-8044-5819-831d-2b6a7c6cead2, current:3149702c-8044-5819-831d-2b6a7c6cead2, cached: (null)
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               loading silo with file contents
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               0.00ms      parse blob
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               0.03ms      loaded file
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               file: 00000000-0000-0000-0000-000000000000, current:00000000-0000-0000-0000-000000000000, cached: (null)
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               loading silo with file contents
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               0.00ms      parse blob
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0309 XbSilo               ignoring index: silo has no data
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0314 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_solokey.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0314 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_solokey.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0315 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_dfu.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0315 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_dfu.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0315 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_thelio_io.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0315 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_thelio_io.so
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0315 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_jabra.so
[... cut ...]
Nov 11 16:57:58 wigglytuff fwupd[31291]: 21:57:58:0739 FuMain               Started with locale en_US.UTF-8
Nov 11 16:57:58 wigglytuff fwupd[31291]: 21:57:58:0739 FuMain               Daemon ready for requests
Nov 11 16:57:58 wigglytuff systemd[1]: Started Firmware update daemon.
Nov 11 16:57:58 wigglytuff fwupd[31291]: 21:57:58:0740 FuMain               FuMain: acquired name: org.freedesktop.fwupd
Nov 11 16:57:58 wigglytuff fwupd[31291]: 21:57:58:0745 FuMain               Called GetRemotes()
Nov 11 16:58:03 wigglytuff fwupd[31291]: 21:58:03:0458 FuMain               Called GetRemotes()
```

superm1, 2019-11-11:
OK so the first problem looks like this:
```
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0301 FuConfig             using config path of /etc/fwupd
```

If you trace the code back it goes to getting this from the environment variable `CONFIGURATION_DIRECTORY`: https://github.com/fwupd/fwupd/blob/679d1c0f9a49de15e247657d1fb284205bc34f5e/src/fu-common.c#L1028

Well `CONFIGURATION_DIRECTORY` gets set from [systemd via the unit ](https://www.freedesktop.org/software/systemd/man/systemd.exec.html) to be what was set by their systemd config. https://github.com/fwupd/fwupd/blob/master/data/fwupd.service.in#L13

In this case their systemd has `ConfigurationDirectory` set to come out as `/etc` not `/usr/share` like their fwupd package does.

So there are two potential solutions:
1. Modify the systemd unit `fwupd.service` and comment out the `ConfigurationDirectory` directive.  Reboot and you should be in good shape (or reload unit files and restart the daemon).
This is the easier solution; a 1 line change.

2. Fix the clear implementation of the systemd package so that `ConfigurationDirectory` is properly populated.  This means fixing systemd, and then rebuilding the clear fwupd package.  This is the more correct change, but obviously a few steps to make happen.

I think you can try 1, and if it works can bring this to Clear as a bug for discussion.

Queuecumber, 2019-11-11:
I think there's a third option which is for them to use /etc like everyone
else right

superm1, 2019-11-11:
Well yeah 😜 .  I was going on the assumption they made this decision for a good reason.

Other notable thing from your log, it looks like the TPM PCR register reading isn't working on Clear.  Maybe they need to move up to a newer version of the tpm2-tss stack?
```
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0397 FuDevice             using f49999b6e3af5b4bf8ad1f2ed2922b76df457271 for UEFI-34578c72-11dc-4378-bc7f-b643866f598c-dev0
Nov 11 16:57:57 wigglytuff fwupd[31291]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Nov 11 16:57:57 wigglytuff fwupd[31291]: 21:57:57:0398 FuPluginUefi         failed to initialize TPM library
```
That should be a separate bug to them though.

Queuecumber, 2019-11-12:
You know what I was getting TPM errors my last system too I have a feeling its something with my machine but I'm not sure

Queuecumber, 2019-11-12:
Actually my systemd unit has `ConfigurationDirectory=fwupd`

```
[Unit]
Description=Firmware update daemon
Documentation=https://fwupd.org/
After=dbus.service
Before=display-manager.service

[Service]
Type=dbus
BusName=org.freedesktop.fwupd
ExecStart=/usr/libexec/fwupd/fwupd
StateDirectory=fwupd
CacheDirectory=fwupd
ConfigurationDirectory=fwupd
PrivateTmp=yes
ProtectHome=yes
ProtectSystem=full
RestrictAddressFamilies=AF_NETLINK AF_UNIX
SystemCallFilter=~@mount
ProtectControlGroups=yes
ProtectKernelModules=yes
RestrictRealtime=yes
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI
```

Queuecumber, 2019-11-12:
Yes commenting that line out did fix it though, I'm not entirely sure why. Is it that their fwupd is hard coded to use `/usr/share` so not providing a directive allows it to pick the default?

superm1, 2019-11-12:
> ConfigurationDirectory=fwupd

Yes that means the subdirectory of whatever systemd configures the base directory for configuration. Which in this case systemd.pc has `/etc`

closed 2019-11-12 by superm1