fwupd/fwupd #1563: Cant update Star Drive SATA SSD in 1.3.3

Sean-StarLabs, 2019-11-15:
```
fwupdmgr get-devices 
Lite
│
├─Star Drive SATA SSD:
│     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
│     Summary:             ATA Drive
│     Current version:     SBFM61.2
│     Vendor:              Intel Corporation
│     GUIDs:               3a36ddd0-e013-5dec-a4bb-a2204096d765
│                          0966c9ae-3ad1-57f1-a328-3cee48807461
│                          1bd403d3-8f7e-5dca-b0c1-6535d48f4e85
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─SuperIO IT8987:
│     Device ID:           1f664ad149f2151fb750b52751dbf9ba045a7846
│     Summary:             Embedded Controller
│     Current version:     14.07
│     GUID:                0aa07091-dc02-5e62-a0cd-9d3880721b64
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Update requires a reboot
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│   
└─System Firmware:
      Device ID:           b25a2c32299c1278de5a9fbbf369e11d4c669358
      Current version:     1.0.5
      Minimum Version:     1.0.5
      Vendor:              Star Labs
      GUID:                797f8bae-0ea2-4c0f-8a30-7d10ccfacbc0
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
    

```
```
No upgrades for SuperIO IT8987, current is 14.07: 14.07=same, 14.06=older
No upgrades for System Firmware, current is 1.0.5: 1.0.5=same, 1.0.4=older
```
```
fwupdmgr install c7d85a5516e61d294053baa522fb7f0ed2918b0a-SBFM.cab 
Decompressing…           [***************************************]
No supported devices found
```

```
Nov 15 15:32:53 star-Lite fwupd[1663]: 15:32:53:0928 FuMain               Called GetHistory()
Nov 15 15:32:53 star-Lite fwupd[1663]: 15:32:53:0924 FuMain               Called GetDevices()
Nov 15 15:32:49 star-Lite dbus-daemon[808]: [system] Rejected send message, 2 matched rules; type="method_call", sender=":1.72" (uid=1000 pid=1660 comm="fwupdmgr modify-config VerboseDomains * " label="unconfined") interface="org.freedesktop.systemd1.Unit" member="Stop" error name="(unset)" requested_reply="0" destination=":1.1" (uid=0 pid=1 comm="/sbin/init splash " label="unconfined")
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0843 XbSilo               2.09ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='3a36ddd0-e013-5dec-a4bb-a2204096d765']/../..|components/component/provides/firmware[@type='flashed'][text()='0966c9ae-3ad1-57f1-a328-3cee48807461']/../..|components/component/provides/firmware[@type='flashed'][text()='1bd403d3-8f7e-5dca-b0c1-6535d48f4e85']/../..` limit=1 -> 0 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0840 XbSilo               1.25ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='7ee0c805-ca7f-51ea-8bbd-e20e623b9323']/../..|components/component/provides/firmware[@type='flashed'][text()='8257b546-4d2d-5c32-adc5-fc54b8fdfe53']/../..` limit=1 -> 0 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0838 XbSilo               0.68ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='797f8bae-0ea2-4c0f-8a30-7d10ccfacbc0']/../..` limit=1 -> 1 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0837 XbSilo               1.66ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='0aa07091-dc02-5e62-a0cd-9d3880721b64']/../..` limit=1 -> 1 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0835 XbSilo               0.18ms      query on / with `components/component/provides/firmware` limit=0 -> 379 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0834 XbSilo               0.31ms      query on / with `components/component/provides/firmware` limit=0 -> 379 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0834 FuEngine             319 components now in silo
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0834 XbSilo               3.11ms      query on / with `components/component` limit=0 -> 319 results
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0831 XbSilo               0.01ms      parse blob
[... 10 lines cut ...]
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0830 FuConfig             ordering lvfs-testing=lvfs+1
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0830 FuConfig             ignoring unfound remote fwupd
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0830 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0830 FuConfig             using config path of /usr/lib/fwupd
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0830 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0829 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0829 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0829 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0829 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 FuConfig             using config path of /etc/fwupd
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 XbSilo               loading silo with file contents
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 XbSilo               file: 2d5ee269-e243-5875-8d4a-2ce6443b91e7, current:2d5ee269-e243-5875-8d4a-2ce6443b91e7, cached: (null)
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 FuConfig             loading /usr/share/fwupd/metainfo
Nov 15 15:32:46 star-Lite fwupd[1663]: 15:32:46:0828 Fwupd                looking for os-release at /etc/os-release
Nov 15 15:32:46 star-Lite polkitd(authority=local)[878]: Operator of unix-session:c2 successfully authenticated as unix-user:star to gain TEMPORARY authorization for action org.freedesktop.fwupd.modify-config for system-bus-name::1.72 [fwupdmgr modify-config VerboseDomains *] (owned by unix-user:star)
Nov 15 15:32:44 star-Lite dbus-daemon[808]: [system] Successfully activated service 'org.freedesktop.fwupd'
Nov 15 15:32:43 star-Lite fwupd[1663]: ERROR:esys:src/tss2-esys/esys_context.c:67:Esys_Initialize() Initialize default tcti. ErrorCode (0x00070002)
Nov 15 15:32:43 star-Lite fwupd[1663]: ERROR:esys:src/tss2-esys/esys_tcti_default.c:213:get_tcti_default() No standard TCTI could be loaded
Nov 15 15:32:43 star-Lite dbus-daemon[808]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.72' (uid=1000 pid=1660 comm="fwupdmgr modify-config VerboseDomains * " label="unconfined")
Nov 15 15:32:14 star-Lite systemd[1]: Started Refresh fwupd metadata regularly.
Nov 15 15:31:47 star-Lite systemd[1]: Stopped Refresh fwupd metadata regularly.
Nov 15 15:31:39 star-Lite systemd[1]: Started Refresh fwupd metadata regularly.
Nov 15 15:31:24 star-Lite sudo[5319]:     star : TTY=pts/0 ; PWD=/home/star ; USER=root ; COMMAND=/usr/local/bin/apt install --reinstall fwupd libfwupd2
```

superm1, 2019-11-15:
Well so I guess that's this CAB file: https://fwupd.org/lvfs/devices/com.starlabs.ata.firmware
```
    <!-- IDE\SATA_SSD________________________________SBFM61.3 -->
    <firmware type="flashed">8d12d74a-14a9-5dc8-a87d-b21471c3f128</firmware>
```

Can you run `fwupdmgr get-devices` with `sudo` so that we can see the GUIDs that were used to build the device?

superm1, 2019-11-15:
Also you have a hwid requirement; 
```
<hardware>b31aa89d-cfb8-5c67-8502-ce76a9e7b580</hardware>
```

So can we see `fwupdtool hwids` for your system?

Sean-StarLabs, 2019-11-18:
Seems to be the ID of the SSD - strange as there have been successful updates. 

Thank you!

Sean-StarLabs, 2019-11-19:
Added the other guid into the cab and can now install it the update from the cab but it still doesn't show as a supported device.

I'm guessing I've missed something but can't figure out what.

```
Lite
│
├─SATA SSD:
│     Device ID:           5b9c910f15a42ff2925b267353c990e5d355d969
│     Summary:             ATA Drive
│     Current version:     SBFM61.3
│     Vendor:              Intel Corporation
│     Serial Number:       A462078A198900535930
│     GUIDs:               a8e0709c-5214-53df-bd41-5faf36394126 ← IDE\SATA_SSD________________________________SBFM61.3
│                          e6fd9e43-61db-5303-9fb5-8a997cbbe585 ← IDE\0SATA_SSD________________________________
│                          ef0645fb-0011-5566-b655-ab24bb26b354 ← SATA SSD
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─SuperIO IT8987:
│     Device ID:           1f664ad149f2151fb750b52751dbf9ba045a7846
│     Summary:             Embedded Controller
│     Current version:     14.07
│     GUID:                0aa07091-dc02-5e62-a0cd-9d3880721b64 ← SuperIO-IT8987
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Update requires a reboot
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│   
└─System Firmware:
      Device ID:           b25a2c32299c1278de5a9fbbf369e11d4c669358
      Current version:     1.0.5
      Minimum Version:     1.0.5
      Vendor:              Star Labs
      GUID:                797f8bae-0ea2-4c0f-8a30-7d10ccfacbc0
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
```
```
Computer Information
--------------------
BiosVendor: American Megatrends Inc.
BiosVersion: V1.0.05
BiosMajorRelease: 1
BiosMinorRelease: 0
Manufacturer: Star Labs
Family: APL_NB_116
ProductName: Lite
ProductSku: L3-U
EnclosureKind: a
BaseboardManufacturer: Star Labs Online Limited
BaseboardProduct: Star Lite Mk II

Hardware IDs
------------
{9804963b-ebc1-5b5c-aac8-4b2e5415d8ca}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{a1a9c893-666a-55fe-9c09-db43b2b1ed40}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{aece3866-c02c-5e23-91ee-222e2a1e5b13}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{dbd75976-289b-5d5c-b264-48ea4ebc08ec}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{4fcfe2f1-dabf-5799-b8ea-fc44e8224051}   <- Manufacturer + Family + ProductName + ProductSku
{7bf15b3e-cb72-52b8-ae8d-09fc58a0eb58}   <- Manufacturer + Family + ProductName
{30387f87-c03d-5fa9-9b9c-cf374bfa8246}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{013b60e5-1023-5bee-8ae5-14cae21377b7}   <- Manufacturer + ProductSku
{c9b921c7-6c8d-5f7f-86c5-926a58ba414a}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{b29597d2-f144-52f1-85c3-b40567f2e0bf}   <- Manufacturer + ProductName
{7e0802d9-2a1d-5682-b969-fea039a73602}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{80f1fd54-f581-571f-90e6-00bf8a67f627}   <- Manufacturer + Family
{69ae119e-0cc0-5d66-93db-3687e326f907}   <- Manufacturer + EnclosureKind
{d1a64840-4307-58fb-a62c-de28a07c0151}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{b31aa89d-cfb8-5c67-8502-ce76a9e7b580}   <- Manufacturer

```

https://fwupd.org/lvfs/devices/com.starlabs.ata.firmware

hughsie, 2019-11-19:
> Added the other guid into the cab

I think you should be using a8e0709c-5214-53df-bd41-5faf36394126

> it still doesn't show as a supported device.

"Supported" is when the firmware is on the LVFS. Perhaps it's not a very good translated name.

superm1, 2019-11-19:
> I think you should be using a8e0709c-5214-53df-bd41-5faf36394126

So I suspect the biggest challenge is going to be that this GUID has the version encoded into it.  So downgrading firmware won't work, and the upgrade ot the next one needs to list all the GUIDs it can upgrade "from".

> "Supported" is when the firmware is on the LVFS. Perhaps it's not a very good translated name.

I thought we did fine with that: "                           • Supported on remote server"

Sean-StarLabs, 2019-11-19:
> So I suspect the biggest challenge is going to be that this GUID has the version encoded into it. So downgrading firmware won't work, and the upgrade ot the next one needs to list all the GUIDs it can upgrade "from".
It's strange though, we've only ever had 2 version for these SSD's (61.2 and 61.3). The cab was uploaded at least 6 months ago and tested heavily by us (and I think @hughsie tested it too)

> "Supported" is when the firmware is on the LVFS. Perhaps it's not a very good translated name.
Makes sense to me if a device with a specific firmware version isn't supported.

superm1, 2019-11-19:
> It's strange though, we've only ever had 2 version for these SSD's (61.2 and 61.3). The cab was uploaded at least 6 months ago and tested heavily by us (and I think @hughsie tested it too)

I see it as:
61.2 needs to have the GUID that corresponds to 61.2 and 61.3 in it's metadata.  Then 61.3 needs to have the GUID that corresponds to 61.2 and 61.3 in it's metadata.  And then when you introduced 61.4 it has to have 61.3 and 61.4.

This will accomplish the goal of it continuing to be marked supported, allow reinstalls to work.  If you want to support downgrades or upgrades without going through intermediary releases you'll need to modify all XML every time you release something.

Sean-StarLabs, 2019-11-19:
Gotcha.

Is it possible to modify the cabs? (Rather than re-upload)

superm1, 2019-11-19:
>Is it possible to modify the cabs? (Rather than re-upload)

As in modify the XML to add a new GUID and repack the CABs from LVFS side?  Not with the way things are currently set up in LVFS and fwupd.

A better idea to me is to use the OEM area to encode information to build a GUID.  Dell does this for both ATA and NVME.  Example: https://github.com/fwupd/fwupd/blob/master/plugins/ata/fu-ata-device.c#L139

It's sorta wild west in this area though, so try to do it in a way that won't conflict with Dell's implementation and I don't see a problem with it going into fwupd.

hughsie, 2019-11-19:
If Dell published the spec, I'm sure other companies even rhyming with Kenovo would implement it...

superm1, 2019-11-19:
Outside of firmware update ability Dell has _other_ reasons that the disk firmware is different, and this is the reason for adding information to the OEM block in the first place.  The OEM and IHV provided firmware are completely incompatible even on the same model disk.

I was under the presumption other OEMs did this too, but maybe they just take IHV firmware.

hughsie, 2019-11-19:
> I was under the presumption other OEMs did this too, but maybe they just take IHV firmware.

No, they have OEM firmware too, just with no unique PID or vendor block...

Sean-StarLabs, 2019-11-19:
SSD firmware isn't my strong suit, maybe we can add an OEM block. But would it work in theory, to derive a GUID say the controller and info block? 

(Simply matching the `Model` field on the info block would be specific enough for us but I don't think we are "normal" in that regard)

superm1, 2019-11-19:
> No, they have OEM firmware too, just with no unique PID or vendor block...

Hmm, with the way fwupd is set up today if a firmware was pushed from another OEM using a "generic" GUID it _could_ actually push out to a non-OEM system that means.  The bootloader of the drive is supposed to reject it from what I understand, but I certainly haven't tested that myself :)

Conversely if an IHV pushed generic firmware to LVFS it could try to install on OEM disks.
So yes, I think other OEMs should start using the vendor blocks like Dell does.

Dell's implementation can mostly be interpreted from the fwupd code.  I don't think anyone should use the exact same one, as it will build `STORAGE-DELL-%s` GUIDs.  But similar principles sure.

superm1, 2019-11-19:
> Conversely if an IHV pushed generic firmware to LVFS it could try to install on OEM disks.

I wonder if we need some sort of device flag that it requires a CHID restriction to be set on LVFS so it's only targeted to certain systems?  That would at least help some cases, but still if you go buy an off the shelf drive and pop it in your system you'll hit this limitation.

hughsie, 2019-11-19:
> it requires a CHID restriction to be set on LVFS so it's only targeted to certain systems

That seems fragile; I don't know about you but I have dozens of laptops here and I've lost track of what drive came from what laptop originally as they get swapped out so often.

superm1, 2019-11-19:
Yeah I completely agree.  It's not scalable in this regard.

superm1, 2019-11-21:
Since this issue is just needing to put "correct" GUIDs into CAB closing it.

closed 2019-11-21 by superm1