fwupd/fwupd #1566: Update process shows message that shim isn't installed and nothing is updated after reboot

tomastzn, 2019-11-15:
Running Linux Mint 19.2 on Lenovo X1 Carbon 6th gen. Dual boot with Windows (on the same disk, in different partitions). 
In BIOS all settings are enabled (like update from Windows), including secure boot being enabled.
The application `fwdupd` installed from Snap store.

I executred following commands:
```shell
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr -v update
```
and the last one produced following output:
```
Upgrade available for System Firmware from 0.1.41 to 0.1.43
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 0.1.43 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[                    \                  ]
Scheduling…              [***************************************]
Secure boot is enabled, but shim isn't installed to the EFI system partition
```

The last message sort of indicates that something is not as expected and during reboot there was no update done.

Here is the info you might need:
```shell
$ fwupd.fwupdmgr --version
client version:	1.3.3
checkout info:	1.3.3
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.3
```

```shell
$ fwupd.fwupdmgr get-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           f6ea21352e9263d5a08d7559dd26e8dae24f8622
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          09bc8035-c26c-56c8-af2d-5b0fd7e1efe1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─ADATA SX8200PNP:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     S0118C
│     Vendor:              Intel Corporation (NVME:0x1CC1)
│     GUIDs:               5a22253e-625f-57bb-80fe-4717c59f7fc2
│                          80968304-c171-5944-8e11-18a1fb4203b9
│                          a41282e2-8560-5da8-bb61-738fd992a09f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.41
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Update State:        failed-transient
│     Update Error:        Secure boot is enabled, but shim isn't installed to the EFI system partition
│     Last modified:       2019-11-15 22:17
│     GUIDs:               a4b51dca-8f97-4310-8821-3330f83c9135
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a
│                          6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.05_B0029
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

```shell
$ efibootmgr -v
efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0018,0019,001A,001B,001C,001D,001E,001F
Boot0000* Windows Boot Manager	HD(2,GPT,b94afefa-d7b3-4dec-9128-e418703df9bd,0xfa000,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...0................
Boot0001* ubuntu	HD(2,GPT,b94afefa-d7b3-4dec-9128-e418703df9bd,0xfa000,0x32000)/File(\EFI\ubuntu\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-a4b51dca-8f97-4310-8821-3330f83c9135-0
```

```shell
$ tree /boot
/boot
├── config-4.15.0-70-generic
├── efi
│   └── EFI
│       ├── Boot
│       │   ├── bootx64.efi
│       │   └── fbx64.efi
│       ├── linuxmint
│       │   └── fw
│       │       └── fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
│       ├── Microsoft
│       │   ├── Boot
│       │   │   ├── ...
│       │   │   ├── bootmgfw.efi
│       │   │   ├── bootmgr.efi
│       │   │   ├── BOOTSTAT.DAT
│       │   │   ├── boot.stl
│       │   │   ├── memtest.efi
│       │   │   └── ...
│       │   └── Recovery
│       │       └── ...
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   └── ...
│   ├── unicode.pf2
│   └── x86_64-efi
│       └── ...
├── initrd.img-4.15.0-70-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.15.0-70-generic
└── vmlinuz-4.15.0-70-generic
```

Please answer the following questions:

- Operating system and version: Linux Mint 19.2
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

superm1, 2019-11-15:
Can you share /etc/os-release?

And did this work previously on mint? Is it a regression?

tomastzn, 2019-11-17:
Below is the content of `/etc/os-release`. I can't tell whether this worked on Linux Mint before, I'm trying this for the first time on this laptop.
```
NAME="Linux Mint"
VERSION="19.2 (Tina)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 19.2"
VERSION_ID="19.2"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.ubuntu.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=tina
UBUNTU_CODENAME=bionic
```

superm1, 2019-11-18:
Well it looks like it's caused by the `ID_LIKE` field and `ID` field differences.  So mint is a derivative of Ubuntu, so it's like Ubuntu.

Similarly on Ubuntu I see this:
```
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.2 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

The problem is mint actually uses Ubuntu's bootloaders in `\efi\ubuntu` but fwupd is expecting `\efi\linuxmint` based on the `/etc/os-release` reading code.

Changing generically from `ID` to `ID_LIKE` won't be a good idea because Ubuntu doesn't use the `\efi\debian\` paths.

superm1, 2019-11-18:
I've adjusted the fallback logic in https://github.com/fwupd/fwupd/commit/e888b58ccd3d7532ee2fbff2c15188982414947c which will be part of the next fwupd release.

Your options for _now_ to use fwupd are:
1. Turn off secure boot and run the update.
2. Switch to the `edge` snap instead.  This is like running git master.

tomastzn, 2019-11-20:
Is the timeframe for next release known? Just trying to figure out whether I should wait for next release or be brave and try it with `edge`.

hughsie, 2019-11-20:
> Is the timeframe for next release known

I'm thinking Friday, as in two days time.

superm1, 2019-11-20:
I would encourage you to test with the edge snap.  It's essentially what will turn into the next release right now, so if this is still broken for you it would be nice to fix before we tag.

tomastzn, 2019-11-20:
Fair point, I will give it a try tonight and report back.

tomastzn, 2019-11-20:
Unfortunately the change did not help.
I've refreshed to `fwupdmgr` from `edge`:

superm1, 2019-11-20:
Have you rebooted or restarted the daemon since then? I would guess yes, but please confirm. And also share a verbose log again.

superm1, 2019-11-21:
Oh I think I know why .. it's because you had one run with bad data the first time.

Please remove the Linuxmint folder from your efi system partition and try again

tomastzn, 2019-11-21:
Before I remove anything, just want to confirm which folder needs to be removed.
This is the output from `sudo tree /boot` (some parts removed for brevity):
```
/boot
├── config-4.15.0-70-generic
├── efi
│   └── EFI
│       ├── Boot
│       │   ├── bootx64.efi
│       │   └── fbx64.efi
│       ├── linuxmint
│       │   └── fw
│       │       └── fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
│       ├── Microsoft
│       │   ...
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi
├── grub
│   └── ...
├── initrd.img-4.15.0-70-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-4.15.0-70-generic
└── vmlinuz-4.15.0-70-generic
```
Is it the folder `/boot/efi/EFI/linuxmint` that can be removed?

superm1, 2019-11-21:
Yep

tomastzn, 2019-11-21:
Success!
After removing the `/boot/efi/EFI/linuxmint` folder, the update process worked fine. Well, it did not automatically reboot or offer a reboot but I think that issue might have been caused by the Logitech Unifying receiver that did not reconnect after upgrade. You can see it in the log below.

When the `sudo fwupdmgr -v update` was done, I ran `fwupdmgr get-updates` again and it showed that a reboot is pending, so I rebooted the system. At boot time, there were couple of messages like "adding capsule" (I wasn't fast enough to snap a pic), then the system rebooted again, this time showing Lenovo logo for a minute or two, then rebooted again once or twice and finally rebooted back to grub and back to Linux.

Awesome job everyone, I really appreciate the effort you've put into making firmware updates from Linux possible. Thank you.

The log of the whole update process (long):
```
sudo fwupdmgr -v update  
Upgrade available for System Firmware from 0.1.41 to 0.1.43
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 0.1.43 for System Firmware...
(fwupdmgr:5233): FuCommon-DEBUG: 19:38:40.524: creating path /root/snap/fwupd/common/.cache/fwupd
(fwupdmgr:5233): FuMain-DEBUG: 19:38:40.562: skpping download as file already exists
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.567: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.601: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.601: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.602: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.614: Emitting ::status-changed() [scheduling]
Updating System Firmware…[ -                                     ]
Scheduling…              [ -                                     ](fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.614: Emitting ::device-changed(1c53551e7da69d896138fac1ae131c83ad46d923)
Scheduling…              [   -                                   ](fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.710: Emitting ::status-changed() [idle]
Scheduling…              [***************************************]
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.711: Emitting ::changed()
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.714: Emitting ::status-changed() [unknown]
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:40.714: Emitting ::device-changed(1c53551e7da69d896138fac1ae131c83ad46d923)
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
Upgrade available for UEFI Device Firmware from 0.1.17 to 0.1.18
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 0.1.18 for UEFI Device Firmware...
(fwupdmgr:5233): FuCommon-DEBUG: 19:38:52.359: creating path /root/snap/fwupd/common/.cache/fwupd
(fwupdmgr:5233): Fwupd-DEBUG: 19:38:52.360: looking for os-release at /etc/os-release
(fwupdmgr:5233): FuMain-DEBUG: 19:38:52.374: downloading https://fwupd.org/downloads/de71b7264413d9e78e6f38dcc5e8eb1fa0a7f52e-Lenovo-ThinkPad-X1Carbon6th-ECFirmware-1.18.cab to /root/snap/fwupd/common/.cache/fwupd/de71b7264413d9e78e6f38dcc5e8eb1fa0a7f52e-Lenovo-ThinkPad-X1Carbon6th-ECFirmware-1.18.cab
Fetching firmware https://fwupd.org/downloads/de71b7264413d9e78e6f38dcc5e8eb1fa0a7f52e-Lenovo-ThinkPad-X1Carbon6th-ECFirmware-1.18.cab
(fwupdmgr:5233): FuMain-DEBUG: 19:38:53.138: progress: 0%
Downloading…             [    \                                  ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.139: progress: 1%
Downloading…             [                                       ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.140: progress: 3%
Downloading…             [*                                      ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.141: progress: 4%
Downloading…             [*                                      ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.240: progress: 5%
Downloading…             [*                                      ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.241: progress: 6%
Downloading…             [**                                     ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.242: progress: 7%
Downloading…             [**                                     ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.243: progress: 8%
Downloading…             [***                                    ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.245: progress: 9%
Downloading…             [***                                    ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.246: progress: 10%
Downloading…             [***                                    ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.248: progress: 11%
Downloading…             [****                                   ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.249: progress: 12%
Downloading…             [****                                   ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.342: progress: 13%
Downloading…             [*****                                  ](fwupdmgr:5233): FuMain-DEBUG: 19:38:53.343: progress: 14%
[... cut ...]
Authenticating…          [  -                                    ](fwupdmgr:5233): Fwupd-DEBUG: 19:39:01.784: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
Updating Unifying Receiver…
Idle…                    [***************************************]
(fwupdmgr:5233): Fwupd-DEBUG: 19:39:03.267: Emitting ::device-changed(672c087de09848d9e7ee32aa1dea2fbeb8b81e6b)
failed to get device after detach: failed to wait for detach replug: device fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 did not come back
```

superm1, 2019-11-21:
Cool, glad that worked for your UEFI issue.

closed 2019-11-21 by superm1

cybericius, 2024-11-28:
I have simply replicated the folder content under `/boot/efi/EFI/linuxmint` from `/boot/efi/EFI/ubuntu`. Not sure if that's fine and was only needed by `fwupdmgr`, but I had space on the mount point still, so I gave it a try and the update worked. I think it is still a workaround only. ;)

superm1, 2024-11-28:
I'm guessing that your `/etc/os-release` looks something like this?

```
ID=linuxmint
ID_LIKE=debian
```