fwupd/fwupd #1569: Can't update firmware on DELL XPS 13 9380

jalelegenda, 2019-11-19:
```shell
$ fwupdmgr update
No upgrades for XPS 9380 Thunderbolt Controller, current is 44.00: 44.00=same
ignoring XPS 13 9380 System Firmware [6c24a747f97668873b761558e322398a91dbf394] as not updatable
```

```shell
$ fwupdmgr --version
client version: 1.2.11
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.11
```

```shell
$ fwupdmgr get-devices
XPS 9380 Thunderbolt Controller
  DeviceId:             c66301b3d8e487f26dd5f0b7a7f1c12c8c7c4fef
  Guid:                 0f401ed2-b847-532a-adc8-3193fc737be6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              44.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-11-19

XPS 13 9380 System Firmware
  DeviceId:             6c24a747f97668873b761558e322398a91dbf394
  Guid:                 ce945437-7358-49f1-95d8-6b694a10a755
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.1.3.2
  VersionLowest:        0.1.3.2
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-11-19
  UpdateError:          /sys/firmware/efi/efivars was not mounted

XPS 13 9380 TPM 2.0
  DeviceId:             2e59657d5e0beb043a16639968ca6178bcedff42
  Guid:                 d621e687-b00d-539a-aeac-ac3eb530cbc5
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              7.2.0.2
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-11-19

PM981 NVMe Samsung 512GB
  DeviceId:             f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 ad4ad443-6f4d-5735-ba86-17a1583a2403
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              EXA74D1Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-11-19
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,0000,0003,0005,0006
Boot0000  Windows Boot Manager  HD(1,GPT,7ac303db-0700-1e44-b589-df3fb1b8a8b8,0x800,0x154000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* void_grub     HD(1,GPT,7ac303db-0700-1e44-b589-df3fb1b8a8b8,0x800,0x7d000)/File(\EFI\void_grub\grubx64.efi)
Boot0002  Windows Boot Manager  HD(2,GPT,06c213a1-7f07-4809-a46c-1427c6e4d928,0x96800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...&................
Boot0003* Diskette Drive        BBS(Floppy,Diskette Drive,0x0)..BO
Boot0005* CD/DVD/CD-RW Drive    BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0006* USB NIC       BBS(128,Realtek PXE B39 D00,0x0)..BO
```
I had to `sudo` for this ^

```shell
$ efivar -l | grep fw
(empty)
```

```shell
$ tree /boot
[...]
efi
│   └── EFI
│       ├── Dell
│       │   └── logs
│       │       ├── diags_current.xml
│       │       └── diags_previous.xml
│       └── void_grub
│           └── grubx64.efi
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   └── x86_64-efi
[...]
```

- Operating system and version: void linux (rolling), kernel version: 5.3.11_1
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): xbps
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

I'm aware of [this issue](https://github.com/fwupd/fwupd/issues/627), and have tried stuff applicable from it (like editing uefi.conf `OverrideESPMountPoint=/boot`), but I'm not running systemd.

hughsie, 2019-11-19:
Can you try a newer fwupd version, and then reget the "fwupdmgr get-devices --show-all-devices" output please.

superm1, 2019-11-19:
```
UpdateError:          /sys/firmware/efi/efivars was not mounted
```

It looks like your kernel is missing efivarfs?

jalelegenda, 2019-11-19:
@superm1 Thank you, sir! It works now.

closed 2019-11-19 by jalelegenda