fwupd/fwupd #1571: Lenovo T480s failed to reboot after update, pin-hole reset necessary

liquidat, 2019-11-19:
What happened:

I executed `fwupdmgr  update` on November 12. It was attached to a docking station (not listed below since not available right now), and I left for lunch.
When I came back, the laptop was shut down, and didn't react to any key presses, long-pressing of power button or anything else. It only flashed the loading LEDs three times when I attached a power source.

I was only able to reboot the machine after pressing the pin hole reset button.

The same happend a few days later when the machine was attached to another docking station of the same type and I again launched the update command.

If needed I can execute the commands listed below while the machine is attached to the docking station again, but that will only be possible at the end of this week.

```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             c0d1db0f0d2ba3ba27cc0be55b332e9eb9a53316
  Guid:                 3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc <- TBT-01091708
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 thunderbolt
  Created:              2019-11-19

20L8S2N809 System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(939bc5d03076798764dc0eeb2569c06ea6a375e0)
  Checksum:             SHA256(e458f28eb0b8c91e6df21c9cd14cd2785d8172a0d24887c9fe09c83081d0c979)
  Vendor:               LENOVO
  Version:              0.1.36
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-11-19
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
  Guid:                 39a39e45-38ce-4b05-809b-60351bf88ec4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        184.65.3590
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-11-19
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.18
  VersionLowest:        0.1.18
  VersionFormat:        triplet
[... 200 lines cut ...]
             clock: 33MHz
             capabilities: pci normal_decode bus_master cap_list
             configuration: driver=pcieport
             resources: irq:125 memory:dc000000-dc0fffff
           *-nvme
                description: Non-Volatile memory controller
                product: NVMe SSD Controller SM981/PM981/PM983
                vendor: Samsung Electronics Co Ltd
                physical id: 0
                bus info: pci@0000:3e:00.0
                logical name: /dev/nvme0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: nvme nvm_express bus_master cap_list
                configuration: driver=nvme latency=0
                resources: irq:16 memory:dc000000-dc003fff
              *-disk
                   description: NVMe disk
                   product: SAMSUNG MZVLB256HAHQ-000L7
                   physical id: 0
                   logical name: /dev/nvme0n1
                   version: 0L2QEXD7
                   serial: S41GNB0K538402
```

superm1, 2019-11-19:
This sounds like it's firmly in the camp of BIOS bug, all that can be done is demote the firmware from LVFS stable until Lenovo can debug it.

Can you please share the `fwupdmgr get-history` output?  We can figure out which UEFI firmware update needs to be demoted.

hughsie, 2019-11-19:
To clarify slightly, did the update succeed if the laptop was *not* docked?

liquidat, 2019-11-20:
@hughsie This behavior happened twice, each time on a different docking station. Since then I only tried it once more, but without a docking station, and there are no available updates listed.

@superm1 
```
$ sudo fwupdmgr get-history
UEFI Device Firmware
  DeviceId:             118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
  Guid:                 39a39e45-38ce-4b05-809b-60351bf88ec4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|needs-reboot|reported
  Version:              184.55.3510
  Created:              2019-03-28
  UpdateState:          success
  
  [Release]
  RemoteId:             lvfs
  Version:              184.60.3561
  Checksum:             SHA1(b8c8dd775700134a7adc323d3eb20b4ac2d24702)
  Flags:                none
  BootMgrDesc:          legacy
  DistroVersion:        29
  CompileVersion(com.redhat.fwupdate): 12
  CompileVersion(org.freedesktop.fwupd): 1.2.5
  RuntimeVersion(com.redhat.fwupdate): 12
  SecureBoot:           Disabled
  CompileVersion(com.redhat.efivar): 37
  UEFIUXCapsule:        Disabled
  ESPMountPoint:        /boot/efi
  MissingCapsuleHeader: False
  CompileVersion(org.freedesktop.gusb): 0.3.0
  RuntimeVersion(com.dell.libsmbios): 2.4
  DistroId:             fedora
  CpuArchitecture:      x86_64
  BootTime:             1553782067
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  DistroVariant:        workstation
  RuntimeVersion(org.freedesktop.fwupd): 1.2.5
  KernelVersion:        5.0.3-200.fc29.x86_64

Unifying Receiver
  DeviceId:             e2419b15f48575f22e60e51ba08d9c1bc8c2b04b
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Plugin:               unifying
  Flags:                updatable|supported|reported
  Version:              RQR24.01_B0023
  Created:              2019-05-15
  UpdateState:          success
  
  [Release]
  RemoteId:             lvfs
  Version:              RQR24.06_B0030
  Checksum:             SHA1(fe037400f0977a13c215d94e427f6be14ae0be3e)
  Flags:                none
  DistroVariant:        workstation
  RuntimeVersion(com.redhat.fwupdate): 12
  CompileVersion(org.freedesktop.gusb): 0.3.0
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.8
  DistroId:             fedora
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  DistroVersion:        30
  CompileVersion(org.freedesktop.fwupd): 1.2.8
  BootTime:             1557920658
  RuntimeVersion(com.dell.libsmbios): 2.4
  CompileVersion(com.redhat.efivar): 37
  KernelVersion:        5.0.14-300.fc30.x86_64
  CompileVersion(com.redhat.fwupdate): 12

UEFI Device Firmware
  DeviceId:             118b3d8cfafa8781890c4d3e23a4deb9fcb18e5a
[... 314 lines cut ...]
  
  [Release]
  RemoteId:             lvfs
  Version:              0.1.18
  Checksum:             SHA1(61194dd0f02dd8e0620d3f3688c410c31ec817ac)
  Flags:                none
  BootMgrDesc:          legacy
  CompileVersion(com.redhat.fwupdate): 12
  DistroVersion:        31
  CompileVersion(org.freedesktop.fwupd): 1.2.11
  SecureBoot:           Disabled
  CompileVersion(com.redhat.efivar): 37
  UEFIUXCapsule:        Enabled
  ESPMountPoint:        /boot/efi
  CompileVersion(org.freedesktop.gusb): 0.3.0
  RuntimeVersion(com.dell.libsmbios): 2.4
  DistroId:             fedora
  CpuArchitecture:      x86_64
  BootTime:             1573752556
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  RuntimeVersion(com.redhat.fwupdate): 12
  DistroVariant:        workstation
  RuntimeVersion(org.freedesktop.fwupd): 1.2.11
  KernelVersion:        5.3.8-300.fc31.x86_64
```

superm1, 2019-11-21:
All those histories look accurate, so i'm not sure.

liquidat, 2019-11-29:
So I was finally able to attach the laptop again to the docking station. I launched the update as shown below, and after a restart and applying firmware the machine would not boot anymore.

```
**$ fwupdmgr update
No upgrades for 20L8S2N809 System Firmware, current is 0.1.36: 0.1.36=same, 0.1.34=older, 0.1.33=older, 0.1.31=older, 0.1.30=older
Downloading 184.70.3626 for UEFI Device Firmware...
Fetching firmware https://fwupd.org/downloads/a7d95bacee4c374cd6f1112b35bb933e31902cc2-Lenovo-ThinkPad-T480s-CorporateMEFirmware-11.8.70.3626.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
No upgrades for UEFI Device Firmware, current is 0.1.18: 0.1.18=same, 0.1.17=older, 0.1.16=older, 0.1.13=older

An update requires a reboot to complete. Restart now? [Y|n]: n
```

After this however fwupdmgr reports an updated system:

```
$ fwupdmgr get-updates
No upgrades for System Firmware, current is 0.1.36: 0.1.36=same, 0.1.34=older, 0.1.33=older, 0.1.31=older, 0.1.30=older
No upgrades for UEFI Device Firmware, current is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.18: 0.1.18=same, 0.1.17=older, 0.1.16=older, 0.1.13=older
```

```
$ fwupdmgr get-devices
20L8S2N809
│
├─ThinkPad Thunderbolt 3 Dock:
│     Device ID:           00ea9a1c7c8b1ee1a31ca82721799606065d8c56
│     Current version:     36.02
│     Vendor:              Lenovo (TBT:0x0108)
│     GUID:                75039175-c354-5df9-a798-5da41461c007
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Thunderbolt Controller:
│     Device ID:           c0d1db0f0d2ba3ba27cc0be55b332e9eb9a53316
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     14.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
│                          03f688aa-3369-58aa-94cf-83a11d9c3fe2
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SAMSUNG MZVLB256HAHQ-000L7:
│     Device ID:           c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:             NVM Express Solid State Drive
│     Current version:     0L2QEXD7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          9657ce89-450f-58d2-ade0-2c6ae667541a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.36
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
[... 603 lines cut ...]
        Summary:           Lenovo ThinkPad T480s Corporate ME Firmware
        License:           Proprietary
        Size:              7.5 MB
        Vendor:            Lenovo Ltd.
        Description:       • 8 Q2'19 Intel Platform Update
        
        Version 11.8.70.3626 (LVFS: 184.70.3626)
        
        Problem Fixes
        
         • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
        
        Security issues fixed:
        
         • CVE-2019-0168
         • CVE-2019-0169
         • CVE-2019-11087
         • CVE-2019-11090
         • CVE-2019-11101
         • CVE-2019-11104
         • CVE-2019-11106
         • CVE-2019-11110
         • CVE-2019-11131
         • CVE-2019-11132
```

closed 2020-01-05 by stale[bot]