fwupd/fwupd #1572: UEFI firmware did not update

klingtnet, 2019-11-20:
There is an update for the UEFI firmware that is scheduled but not applied on reboot:

```sh
$ fwupdmgr get-updates
No upgrades for Thunderbolt Controller, current is 43.00: 43.00=same
No upgrades for System Firmware, current is 0.1.43: 0.1.43=same, 0.1.42=older, 0.1.41=older, 0.1.40=older, 0.1.39=older
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same, 0.1.19=older, 0.1.12=older, 0.1.11=older, 0.1.10=older
20HRCTO1WW
│
└─UEFI Device Firmware:
  │   Device ID:           6461040534f2259a0439361986adf73979fd836e
  │   Current version:     184.65.3590
  │   Minimum Version:     184.65.3590
  │   Update State:        success
  │   GUID:                c35736d2-9e47-4578-93e9-68d5b04ea77e
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 5th Consumer ME Update:
        New version:       184.70.3626
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 5th Consumer ME Firmware
        License:           Proprietary
        Size:              2.2 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 8 Q2'19 Intel Platform Update
        
        Version 11.8.70.3626 (LVFS: 184.70.3626)
        
        Problem Fixes
        
         • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
         • Mitigated the following security vulnerabilities under issues:
        
        Security issues fixed:
        
         • CVE-2019-0168
         • CVE-2019-0169
         • CVE-2019-11087
         • CVE-2019-11090
         • CVE-2019-11101
         • CVE-2019-11104
         • CVE-2019-11106
         • CVE-2019-11110
         • CVE-2019-11131
         • CVE-2019-11132
$ fwupdmgr update              
No upgrades for Thunderbolt Controller, current is 43.00: 43.00=same
No upgrades for System Firmware, current is 0.1.43: 0.1.43=same, 0.1.42=older, 0.1.41=older, 0.1.40=older, 0.1.39=older
Upgrade available for UEFI Device Firmware from 184.65.3590 to 184.70.3626
20HRCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same, 0.1.19=older, 0.1.12=older, 0.1.11=older, 0.1.10=older

An update requires a reboot to complete. Restart now? [Y|n]: n
$ sudo efibootmgr -v
BootNext: 0001
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000,0001
Boot0000* Linux-Firmware-Updater
```

This issue could be related to https://github.com/fwupd/fwupd/issues/1538 .

---

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
```
```shell
$ fwupdmgr get-devices
20HRCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           0489035fa9e9cd1c7b4b19b221e52c859ffffd62
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               89d9d1e6-9e4c-5f07-b5c1-603da3d61835
│                          a6473cf8-043d-5a3a-a661-862c2d332fd1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.7.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1cbe298fc17877b6883e85560778ac0812f0a385
│     Current version:     0.1.43
│     Minimum Version:     0.1.14
│     Vendor:              LENOVO
│     GUID:                798ffd60-f10e-4ac4-8939-c8beabfe55b4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           6461040534f2259a0439361986adf73979fd836e
│     Current version:     184.65.3590
│     Minimum Version:     184.65.3590
│     Update State:        success
│     GUID:                c35736d2-9e47-4578-93e9-68d5b04ea77e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
│     Current version:     0.1.20
│     Minimum Version:     0.0.1
│     GUID:                74997a6b-1adf-4b12-b994-401f06ea8c72
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─VMM0000:
      Device ID:           7c2dd454e6f29453896a4824e8234317d237f8c5
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.000
      Vendor:              Synaptics
      GUIDs:               42addef4-40f9-5e89-b925-d564e35ed368
                           cf8c03c5-18bf-53c4-971f-4a08f88932b5
                           e9427b6a-7389-5461-a592-1da5f8ec99fd
      Device Flags:        • Updatable
```

```shell
$ efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-c35736d2-9e47-4578-93e9-68d5b04ea77e-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   └── fwupd-c35736d2-9e47-4578-93e9-68d5b04ea77e.cap
│   │   └── fwupdx64.efi
│   ├── boot
│   │   └── bootx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-defaults.conf
│   │   └── arch-lts.conf
│   ├── loader.conf
│   └── random-seed
├── vmlinuz-linux
└── vmlinuz-linux-lts

7 directories, 17 files
```

Please answer the following questions:

- Operating system and version: Arch Linux: `Linux m604 5.3.11-arch1-1 #1 SMP PREEMPT Tue, 12 Nov 2019 22:19:48 +0000 x86_64 GNU/Linux`
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman`
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

hughsie, 2019-11-20:
This looks very odd:

    $ sudo efibootmgr -v
    BootNext: 0001
    BootCurrent: 0019
    Timeout: 0 seconds
    BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000,0001
    Boot0000* Linux-Firmware-Updater

Can you paste the full output please.

klingtnet, 2019-11-20:
```sh
$ sudo efibootmgr
BootNext: 0001
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000,0001
Boot0000* Linux-Firmware-Updater
Boot0001* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* IDER BOOT CDROM
Boot0020* IDER BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
```

hughsie, 2019-11-20:
> Linux-Firmware-Updater

I suspect Lenovo firmware is freaking out with both entries the same, which it shouldn't but we've seen similar bugs before... Can you try

* Deleting Boot0001 with efibootmgr and retrying the update please? If that still fails then we know that Boot0000 is somehow special in the Lenovo firmware and we can avoid using index 0000 in the future. I'll give you a patch set to try that avoids 0000 in that case.

If it works, we just have to recover this the best we can.

Of course, the real bug is that we're creating more than one of the same, but I'm looking for that bug now.

Richard.

klingtnet, 2019-11-20:
Thank you for the quick response!

It says now that there is nothing to update, so, maybe the update was applied?

```sh
$ sudo efibootmgr --delete-bootnum --bootnum 0001
BootNext: 0001
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* IDER BOOT CDROM
Boot0020* IDER BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
$ fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
```
Successfully downloaded new metadata: 4 local devices supported
$ fwupdmgr update 
No upgrades for Thunderbolt Controller, current is 43.00: 43.00=same
No upgrades for System Firmware, current is 0.1.43: 0.1.43=same, 0.1.42=older, 0.1.41=older, 0.1.40=older, 0.1.39=older
No upgrades for UEFI Device Firmware: A reboot is pending
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same, 0.1.19=older, 0.1.12=older, 0.1.11=older, 0.1.10=older
```

hughsie, 2019-11-20:
> No upgrades for UEFI Device Firmware: A reboot is pending

Urg, you'll need to set BootNext to 0019, reboot, then try to schedule it again. Sorry, I missed that bit.

klingtnet, 2019-11-20:
Ah alright, after a reboot it said there was an update available (but it did not update, though).
I will set the `BootNext` parameter and try again.

klingtnet, 2019-11-20:
It did not help, to set `--bootnext`, probably because it was already set to `0019`.

hughsie, 2019-11-20:
Can you delete any of the remaining `Linux-Firmware-Updater` entries please, and then try with the wip/hughsie/uefi branch please?

klingtnet, 2019-11-20:
I think you forgot to push the branch.

hughsie, 2019-11-20:
> I think you forgot to push the branch.

Kinda, we pushed it, reviewed it, then merged it to master :)

joakim-tjernlund, 2019-11-20:
> > I think you forgot to push the branch.
> 
> Kinda, we pushed it, reviewed it, then merged it to master :)

I don't see that commit in master either

joakim-tjernlund, 2019-11-20:
I deleted my Boot0000* Linux-Firmware-Updater and then I got thesame update as above using 
fwupdmgr get-updates

I then patched my sys-apps/fwupd-1.3.3-r1 with the above patch and did an update and it
installed this update. Now fwupd this my T480s is uptodate.
```
 efibootmgr 
BootCurrent: 001A
Timeout: 0 seconds
BootOrder: 0010,0011,0012,0013,001D,0019,0017,0018,001A,001B,001C,001E,0023,0001
Boot0001* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* USBR BOOT CDROM
Boot0020* USBR BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
Boot0023* PCI LAN
```
Should the Boot0001* Linux-Firmware-Updater be there afterwards?
When PXE booting, this entry is next to my PCI LAN and one could use it again by mistake

klingtnet, 2019-11-20:
@hughsie 

> Can you delete any of the remaining Linux-Firmware-Updater entries please, and then try with the wip/hughsie/uefi branch please?

Sorry, took me a while to build the package but, luckily, I can confirm that it works now.
Thank you very much for your help!

I will keep the ticket open because of the comment of @joakim-tjernlund

klingtnet, 2019-11-20:
Only for completeness here is the output of `efibootmgr` after the update:

```
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater
Boot0010  Setup
Boot0011  Boot Menu
Boot0012  Diagnostic Splash Screen
Boot0013  Lenovo Diagnostics
Boot0014  Startup Interrupt Menu
Boot0015  Rescue and Recovery
Boot0016  MEBx Hot Key
Boot0017* USB CD
Boot0018* USB FDD
Boot0019* NVMe0
Boot001A* ATA HDD0
Boot001B* USB HDD
Boot001C* PCI LAN
Boot001D  Other CD
Boot001E  Other HDD
Boot001F* IDER BOOT CDROM
Boot0020* IDER BOOT Floppy
Boot0021* ATA HDD
Boot0022* ATAPI CD
```

joakim-tjernlund, 2019-11-20:
You still have Boot0000* ...
That * flag should be removed I after the update is done, possibly the whole entry

superm1, 2019-11-20:
Specifically cc72260 was _not_ merged to master intentionally due to some review comments.  The hope was the other fixes on the branch that were merged to master would be enough to fix this issue.

superm1, 2019-11-20:
@klingtnet can you confirm the last few commits on the branch when you tested?  Given you still have a `Boot0000` entry you probably tested after that commit was dropped (which would confirm those were enough to fix it)

klingtnet, 2019-11-20:
It was built against the latest state on `master`:

```sh
fwupdmgr --version
client version:	1.3.3-58-gfdb4c450
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3-58-gfdb4c450
```

superm1, 2019-11-20:
Thanks for confirming.  @hughsie so it sounds like we probably don't need that other commit then, agree?

joakim-tjernlund, 2019-11-20:
I will delete my boot0001 then.
Just want to point out that I didn't get the update until I had delete my Linux-Firmware-Updater.
The update is from September and I know I have checked for updates between now and then

hughsie, 2019-11-20:
> Thanks for confirming. @hughsie so it sounds like we probably don't need that other commit then, agree?

Agree.

joakim-tjernlund, 2019-11-20:
What about the active Linux-Firmware-Updater entry remaining after update?

hughsie, 2019-11-20:
> What about the active Linux-Firmware-Updater entry remaining after update?

That's by design, and pretty harmless. I think it's best not to remove it afterwards.

closed 2019-11-20 by superm1

mpfz0r, 2021-04-28:
> I will delete my boot0001 then.
> Just want to point out that I didn't get the update until I had delete my Linux-Firmware-Updater.
> The update is from September and I know I have checked for updates between now and then

Had the same issue. This helped. Thanks :+1: