fwupd/fwupd #1575: synaptics rmi plugin doesn't make windows compatible HWID

superm1, 2019-11-21:
In discussion with Synaptics about releasing firmware they were telling me that the GUIDs fwupd is creating are not compatible with the Windows ones.  They were hoping to use the exact same string on both sides.

I double checked and sure enough this is the case.
```
      GUIDs:               424bd00e-9789-5cdf-a12a-3c81bc4676d6 ← HIDRAW\VEN_06CB&DEV_76AF
                           140f4458-951b-5bb9-85e2-879bd5b02615 ← SYNAPTICS_RMI\TM3038-003
                           b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d ← SYNAPTICS_RMI\TM3038
```

On Windows:
[image]

So notable differences:

- Different prefix (`HID` vs `HIDRAW`)
- Using the ACPI information to build the strings, not the HID descriptor
- Includes an attribute corresponding to I think the collection.

I think that some of that at least some information is easily accessible by looking at parents in udev:
```
P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00
E: SUBSYSTEM=i2c
E: DRIVER=i2c_hid
E: MODALIAS=acpi:DELL08AF:PNP0C50:
E: USEC_INITIALIZED=6392113
E: ID_VENDOR_FROM_DATABASE=Dell Inc.

P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001
E: SUBSYSTEM=hid
E: DRIVER=hid-multitouch
E: HID_ID=0018:000006CB:000076AF
E: HID_NAME=DELL08AF:00 06CB:76AF
E: HID_PHYS=i2c-DELL08AF:00
E: HID_UNIQ=
E: MODALIAS=hid:b0018g0004v000006CBp000076AF

P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001/hidraw/hidraw0
N: hidraw0
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001/hidraw/hidraw0
E: SUBSYSTEM=hidraw
E: DEVNAME=/dev/hidraw0
E: MAJOR=247
E: MINOR=0

P: /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001/input/input14
L: 0
E: DEVPATH=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-9/i2c-DELL08AF:00/0018:06CB:76AF.0001/input/input14
E: SUBSYSTEM=input
E: PRODUCT=18/6cb/76af/100
E: NAME="DELL08AF:00 06CB:76AF Touchpad"
E: PHYS="i2c-DELL08AF:00"
E: UNIQ=""
E: PROP=5
E: EV=1b
E: KEY=e520 10000 0 0 0 0
E: ABS=2e0800000000003
E: MSC=20
E: MODALIAS=input:b0018v06CBp76AFe0100-e0,1,3,4,k110,145,148,14A,14D,14E,14F,ra0,1,2F,35,36,37,39,m5,lsfw
E: USEC_INITIALIZED=6931241
E: ID_INPUT=1
E: ID_INPUT_TOUCHPAD=1
E: ID_SERIAL=noserial
E: ID_PATH=pci-0000:00:15.1-platform-i2c_designware.1
E: ID_PATH_TAG=pci-0000_00_15_1-platform-i2c_designware_1
E: ID_FOR_SEAT=input-pci-0000_00_15_1-platform-i2c_designware_1
E: TAGS=:seat:
```

hughsie, 2019-11-22:
> I think that some of that at least some information is easily accessible by looking at parents in udev:

Are those Instance IDs documented by Microsoft? That's literally the first time I've seen things like that.

>  They were hoping to use the exact same string on both sides.

I don't think that's possible, I was told by Syanaptics that the `VEN` and `DEV` IDs were not specific enough, as sometimes different chips (with unique firmware_ are used without changing the HID IDs. That's literally the reason the `SYNAPTICS_RMI\TM3038` ID exists to cover this exact case.

> Different prefix (HID vs HIDRAW)

I guess that makes sense to change, as long as it doesn't break existing firmware. I'd like to see some MS specifications about all the HID IDs first tho...

Richard.

superm1, 2019-11-22:
> Are those Instance IDs documented by Microsoft? That's literally the first time I've seen things like that.
> I guess that makes sense to change, as long as it doesn't break existing firmware. I'd like to see some MS specifications about all the HID IDs first tho...

https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hidclass-hardware-ids-for-top-level-collections

> I don't think that's possible, I was told by Syanaptics that the VEN and DEV IDs were not specific enough, as sometimes different chips (with unique firmware_ are used without changing the HID IDs. That's literally the reason the SYNAPTICS_RMI\TM3038 ID exists to cover this exact case.

Right so that would be the reason to use the ACPI device IDs instead (or also).

hughsie, 2019-11-22:
> https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hidclass-hardware-ids-for-top-level-collections

If you want to add those, ACK from me. I think the ones based from the chip ID are still the safest ones to use...

superm1, 2019-11-22:
The problem with the chip ID is the same chip gets used in different machines and you don't always want to update firmware across a bunch of machines.

So unless you end up setting up restrictions you could end up updating firmware on a Dell machine from a Lenovo payload (for example).

So I guess this comes back to my point that we might want to make restrictions mandatory for certain device classes. If you do that for synaptics RMI for example the OEM can decide whether to use a CHID that is "Dell Inc" or one that is Dell Inc + model number.

hughsie, 2019-11-22:
> The problem with the chip ID is the same chip gets used in different machines and you don't always want to update firmware across a bunch of machines.

That's exactly the opposite of what Synaptics told me, apparently there are assigned chip IDs for each vendor. It's less "silicon chip kind" and more "an ID reserved internally" as i understand it.

superm1, 2019-12-12:
In more discussions we decided this wasn't required and will rely upon the existing strings.

closed 2019-12-12 by superm1