fwupd/fwupd #1584: SuperIO IT8987 device not found after update to 1.3.3

thomasives, 2019-11-23:
On my Star LabTop Mk3 fwupd no longer finds my SuperIO IT8987 device after I have updated to version 1.3.3.   I am running Fedora 31 and I updated via dnf to fwupd.x86_64-1.3.3-1.fc31 from fwupd.x86_64-1.2.11-2.fc31.

systemd has captured the following output during startup of the daemon:

```
Nov 23 11:35:08 bercilak systemd[1]: Starting Firmware update daemon...
Nov 23 11:35:09 bercilak fwupd[2521]: 11:35:09:0107 GLib-GObject         invalid cast from 'FuSuperioIt89Device' to 'FuUdevDevice'
Nov 23 11:35:09 bercilak fwupd[2521]: 11:35:09:0107 FuUdevDevice         fu_udev_device_pread: assertion 'FU_IS_UDEV_DEVICE (self)' failed
Nov 23 11:35:09 bercilak fwupd[2521]: 11:35:09:0107 FuPlugin             unset error in plugin superio for coldplug()
Nov 23 11:35:09 bercilak fwupd[2521]: 11:35:09:0268 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:02.0: failed to open (null): Bad address
Nov 23 11:35:09 bercilak systemd[1]: Started Firmware update daemon.
Nov 23 11:44:54 bercilak systemd[1]: fwupd.service: Succeeded.
```

I suspect this has been caused by https://github.com/fwupd/fwupd/commit/025fc77d7a9e6e3ff78206883cfae26a10b88c79.

I have tried rebooting and I am using an NVMe disk.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
```

```shell
$ fwupdmgr get-devices
LabTop
│
├─Star Drive PCIe SSD:
│     Device ID:           03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:             NVM Express Solid State Drive
│     Current version:     ECFM12.3
│     Vendor:              Phison Electronics Corporation (NVME:0x1987)
│     GUIDs:               8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
│                          a44eb54c-5441-56f2-8cc0-5e48964c6457
│                          12d61493-5ac4-57f7-a2c5-96878d115455
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs shutdown after installation
│                          • Device is usable for the duration of the update
│   
└─System Firmware:
      Device ID:           5984fe62b2de5dfa0c7980660e230707d696c8dd
      Current version:     1.0.9
      Minimum Version:     1.0.9
      Vendor:              Star Labs
      GUID:                8265d473-a6c2-42b4-897b-bc220faa2d32
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
```
Above is output from fwupd after I have updated.  If I downgrade to the previous version I have the following output:

```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

```shell
$ fwupdmgr get-devices
SuperIO IT8987
  DeviceId:             1f664ad149f2151fb750b52751dbf9ba045a7846
  Guid:                 0aa07091-dc02-5e62-a0cd-9d3880721b64
  Summary:              Embedded Controller
  Plugin:               superio
  Flags:                internal|updatable|only-offline|require-ac|supported|registered|needs-reboot
  Version:              03.08
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-11-23

LabTop System Firmware
  DeviceId:             5984fe62b2de5dfa0c7980660e230707d696c8dd
  Guid:                 8265d473-a6c2-42b4-897b-bc220faa2d32
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(107a02c13e7e52763ae5639638d4ab492a9beed7)
  Checksum:             SHA256(c5f07bb75975a550fa49a83a47ccb5ea3551b0496665b387a3d5b2e7b554b564)
  Vendor:               Star Labs
  Version:              1.0.9
  VersionLowest:        1.0.9
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-11-23

Star Drive PCIe SSD
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
  Guid:                 a44eb54c-5441-56f2-8cc0-5e48964c6457
  Guid:                 12d61493-5ac4-57f7-a2c5-96878d115455
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-shutdown
  Vendor:               Phison Electronics Corporation
  VendorId:             NVME:0x1987
  Version:              ECFM12.3
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-11-23
```
I don't believe I am using the UEFI plugin.

hughsie, 2019-11-23:
I believe this was fixed in 1.3.4

thomasives, 2019-11-23:
I have installed 1.3.4 and can confirm this has fixed the issue.

Thanks for your help.

closed 2019-11-23 by thomasives