fwupd/fwupd #1589: running "fwupd.fwupdtool" failed: timeout waiting for snap system profiles to get updated

jarnos, 2019-11-24:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.3
checkout info:	1.3.3
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.3
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
OptiPlex 745
│
├─KINGSTON SV300S37A60G:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:             ATA Drive
│     Current version:     541ABBF0
│     Vendor:              Intel Corporation
│     GUIDs:               e7b0d5b5-8f0b-5879-9d21-368669d3ebc6
│                          d11b9be5-59e1-501e-92b2-8e59cb468696
│                          e236e9f8-4338-506b-ac1c-67daa0ec642a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─ST3500418AS:
      Device ID:           2396250036142d895b79f25cc30512a624fc3d2e
      Summary:             ATA Drive
      Current version:     CC49
      Vendor:              Intel Corporation
      GUIDs:               b76a358e-5827-539c-927f-2597901e4cfd
                           8afbb3c5-0441-5ae6-85c7-dcb0669f4c1a
                           c05ca493-4a9a-57a3-b718-3a709c326590
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Xubuntu 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): sudo snap install
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)?

There is some delay in shutdown:
```shell
$ LC_ALL=C journalctl -b -1
...
Nov 23 18:48:06 jarnos-OptiPlex-745 systemd[1]: Received SIGRTMIN+20 from PID 4308 (plymouthd).
Nov 23 18:49:13 jarnos-OptiPlex-745 fwupd.fwupdtool[4240]: internal error, please report: running "fwupd.fwupdtool" failed: timeout waiting for snap system profiles to get updated
Nov 23 18:49:13 jarnos-OptiPlex-745 systemd[1]: fwupd-activate.service: Control process exited, code=exited status=46
Nov 23 18:49:13 jarnos-OptiPlex-745 systemd[1]: fwupd-activate.service: Failed with result 'exit-code'.
Nov 23 18:49:13 jarnos-OptiPlex-745 systemd[1]: Stopped Activate fwupd updates
...
```

superm1, 2019-11-25:
Can you please try with the snap from the `candidate` channel and see if this persists?

jarnos, 2019-11-28:
Updating to the snap from the candidate channel did not help.

superm1, 2019-12-02:
I would recommend to report this to the snap forums then.  https://forum.snapcraft.io/

That error isn't coming from fwupd, but rather coming from snapd.

closed 2019-12-02 by superm1