fwupd/fwupd #1614: On Fedora31: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

MycroftHolmes1989, 2019-11-28:
Problem details:
On a fresh installation of Fedora 31, the system repeatedly throws this error on startup:
```
Failed to Update: Detailed errors from the package manager follow:
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```
Running `sudo /usr/libexec/fwupd/fwupd --verbose` gives:
```
23:48:40:0303 FuDebug              Verbose debugging enabled (on console 1)
23:48:40:0303 FuConfig             loading config values from /etc/fwupd/daemon.conf
23:48:40:0304 Fwupd                looking for os-release at /etc/os-release
23:48:40:0304 FuConfig             loading /usr/share/fwupd/metainfo
23:48:40:0304 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
23:48:40:0304 XbSilo               file: 0ae864be-38a3-5efc-824a-64c7fed63c86, current:0ae864be-38a3-5efc-824a-64c7fed63c86, cached: (null)
23:48:40:0304 XbSilo               loading silo with file contents
23:48:40:0304 FuConfig             using config path of /etc/fwupd
23:48:40:0304 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
23:48:40:0304 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
23:48:40:0305 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
23:48:40:0305 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
23:48:40:0305 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
23:48:40:0305 FuConfig             ignoring unfound remote fwupd
23:48:40:0305 FuConfig             ordering lvfs-testing=lvfs+1
23:48:40:0305 FuConfig             ignoring unfound remote fwupd
23:48:40:0305 FuConfig             ignoring unfound remote fwupd
23:48:40:0305 FuConfig             ignoring unfound remote fwupd
23:48:40:0312 FuEngine             client certificate exists and working
23:48:40:0312 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
23:48:40:0312 FuHistory            got schema version of 5
23:48:40:0312 FuIdle               setting timeout to 7200s
23:48:40:0312 FuHwids              smbios property Manufacturer=System manufacturer
23:48:40:0312 FuHwids              smbios property EnclosureKind=3
23:48:40:0312 FuHwids              smbios property Family=To be filled by O.E.M.
23:48:40:0312 FuHwids              smbios property ProductName=System Product Name
23:48:40:0312 FuHwids              smbios property ProductSku=SKU
23:48:40:0312 FuHwids              smbios property BiosVendor=American Megatrends Inc.
23:48:40:0312 FuHwids              smbios property BiosVersion=5220
23:48:40:0312 FuHwids              smbios property BiosMajorRelease=05
23:48:40:0312 FuHwids              smbios property BiosMinorRelease=0d
23:48:40:0312 FuHwids              smbios property BaseboardManufacturer=ASUSTeK COMPUTER INC.
23:48:40:0312 FuHwids              smbios property BaseboardProduct=ROG STRIX X370-F GAMING
23:48:40:0315 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
23:48:40:0315 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
23:48:40:0315 XbSilo               file: 30b5cf4d-149b-5110-a783-29e035d4aa32, current:30b5cf4d-149b-5110-a783-29e035d4aa32, cached: (null)
23:48:40:0315 XbSilo               loading silo with file contents
23:48:40:0315 FuEngine             remote lvfs-testing not enabled, so skipping
23:48:40:0315 FuEngine             remote vendor not enabled, so skipping
23:48:40:0315 FuEngine             remote vendor-directory not enabled, so skipping
23:48:40:0315 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
23:48:40:0315 XbSilo               0.03ms      parse blob
23:48:40:0315 XbSilo               0.06ms      loaded file
23:48:40:0315 XbSilo               file: 109468ec-079d-5e3f-bf71-85ae0aed5e47, current:109468ec-079d-5e3f-bf71-85ae0aed5e47, cached: (null)
23:48:40:0315 XbSilo               loading silo with file contents
23:48:40:0315 XbSilo               0.01ms      parse blob
23:48:40:0316 XbSilo               0.23ms      query on / with `components/component` limit=0 -> 334 results
23:48:40:0316 FuEngine             334 components now in silo
23:48:40:0316 XbSilo               0.30ms      query on / with `components/component/provides/firmware` limit=0 -> 396 results
23:48:40:0316 XbSilo               0.11ms      query on / with `components/component/provides/firmware` limit=0 -> 396 results
23:48:40:0320 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_fastboot.so
23:48:40:0320 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_fastboot.so
23:48:40:0320 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_colorhug.so
23:48:40:0321 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_colorhug.so
23:48:40:0321 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_wacom_usb.so
23:48:40:0321 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_wacom_usb.so
23:48:40:0321 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_vli_usbhub.so
23:48:40:0322 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_vli_usbhub.so
23:48:40:0322 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_thunderbolt.so
23:48:40:0322 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_thunderbolt.so
23:48:40:0323 FuEngine             adding plugin /usr/lib64/fwupd-plugins-3/libfu_plugin_optionrom.so
23:48:40:0323 FuPlugin             performing init() on /usr/lib64/fwupd-plugins-3/libfu_plugin_optionrom.so
[... 390 lines cut ...]
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-c91fc7f39d9849a69f9fa316d63617b0.img
├── initramfs-5.3.12-300.fc31.x86_64.img
├── initramfs-5.3.7-301.fc31.x86_64.img
├── loader
│   └── entries
│       ├── c91fc7f39d9849a69f9fa316d63617b0-0-rescue.conf
│       ├── c91fc7f39d9849a69f9fa316d63617b0-5.3.12-300.fc31.x86_64.conf
│       └── c91fc7f39d9849a69f9fa316d63617b0-5.3.7-301.fc31.x86_64.conf
├── memtest86+-5.01
├── System.map-5.3.12-300.fc31.x86_64
├── System.map-5.3.7-301.fc31.x86_64
├── vmlinuz-0-rescue-c91fc7f39d9849a69f9fa316d63617b0
├── vmlinuz-5.3.12-300.fc31.x86_64
└── vmlinuz-5.3.7-301.fc31.x86_64

14 directories, 101 files
```

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Preinstalled
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? Secure boot set to Other OS (Asus X370 motherboard)

MycroftHolmes1989, 2019-11-28:
Possible duplicate of [1549](https://github.com/fwupd/fwupd/issues/1549).

Is it safe to ignore the error for the time being?

superm1, 2019-11-28:
Remove tpm2-abrmd and see if it helps

MycroftHolmes1989, 2019-11-28:
Removing it resolves the issue, but reinstalling it brings the problem back.

superm1, 2019-12-02:
Definitely a duplicate of #1549 then.  As mentioned in https://github.com/tpm2-software/tpm2-abrmd/issues/668 it's caused by  https://github.com/tpm2-software/tpm2-abrmd/issues/667 which will be fixed in the 2.3.1 release.

closed 2019-12-02 by superm1

aadunn, 2020-02-24:
> Definitely a duplicate of #1549 then. As mentioned in [tpm2-software/tpm2-abrmd#668](https://github.com/tpm2-software/tpm2-abrmd/issues/668) it's caused by [tpm2-software/tpm2-abrmd#667](https://github.com/tpm2-software/tpm2-abrmd/issues/667) which will be fixed in the 2.3.1 release.

I'm still seeing this behaviour on Fedora 31!

superm1, 2020-02-24:
Probably a duplicate of #1805 now

r03, 2020-03-05:
On fedora 31 this works for me now: sudo dnf autoremove

tuxedotje, 2020-04-21:
in Fedora 32 beta still the same

Jerther, 2020-05-11:
F32 here and r03 's solution worked! Although it also removed a lot of other stuff like Wireshark and my desktop background (Fedora's default). Nothing important I think.

jsg2021, 2020-06-24:
Fedora 32 r03's solution didn't work for me. It didn't remove anything. Still get this warning every restart.

jsg2021, 2020-06-24:
this made the error go away...
```
sudo dnf remove tpm2-abrmd
```

I have an intel nvme ssd on a pcie-nvme adapter board... wonder if that's related.

superm1, 2020-06-24:
Do you know what version of tpm2-abrmd you had?  If it's newer than 2.3.1 then please report this upstream to them. https://github.com/tpm2-software/tpm2-abrmd

jsg2021, 2020-06-24:
I failed to record what version it was. Is there a history log?

superm1, 2020-06-24:
I'm sure dnf has some sort of log like that, but I'm a debian/ubuntu guy, I don't know where it would be.

superm1, 2020-06-24:
Assuming this is right https://src.fedoraproject.org/rpms/tpm2-abrmd it's with 2.3.1 that it happened.

superm1, 2020-06-24:
And that fix was in 2.3.1, so yes please bring it upstream to tpm2-abrmd guys.
https://github.com/tpm2-software/tpm2-abrmd/issues/667