fwupd/fwupd #1615: Firmware update fails with "start_image() returns unsupported" on Lenovo T470s

aravindhp, 2019-11-28:
```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3
```

```shell
$ fwupdmgr get-devices
20HGS22D00
│
├─Thunderbolt Controller:
│     Device ID:           9829435011463170d5c4af2a48d441d23f4f7a2e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               96c3f95f-611a-56f8-be95-0f912cb84657 ← TBT-01091609
│                          e2b1fec6-8e86-51f6-89a6-afd49e79c64b ← TBT-01091609-0000:03:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Samsung SSD 960 EVO 500GB:
│     Device ID:           e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Summary:             NVM Express Solid State Drive
│     Current version:     3B7QCXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S3X4NB0K232179V
│     GUIDs:               5b3df2da-f745-5fd0-81de-5dafd7f0bf8c ← NVME\VEN_144D&DEV_A804
│                          841f7890-a450-5aa7-8e16-9a4ad2bffedb ← Samsung SSD 960 EVO 500GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           84149fb78009cf27cc1dd5520911f46f8792dbfe
│     Current version:     0.1.35
│     Minimum Version:     0.1.7
│     Vendor:              LENOVO
│     GUID:                7a176688-0960-47ba-931b-7829849e8347
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cbede50560aaa37366326cd54923421517427f39
│     Current version:     184.50.3425
│     Minimum Version:     0.0.1
│     Update State:        success
│     GUID:                e9124c4a-fdff-42e5-b2ad-f745db345953
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
      Current version:     0.1.20
[... 120 lines cut ...]
20HGS22D00 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 0.1.20: 0.1.20=same

An update requires a reboot to complete. Restart now? [Y|n]: Y
```

On the reboot, I see the following which I have transcribed below:
```
dpath: \EFI\fedora\shimx64.efi
path: \fwupdx64.efi
Found update fwupd-e9124c4a-fdff-42e5-b2ad-f745db345953-0
Adding new capsule
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported
```
On rebooting I see the same available firmware update. Is there a workaround or fix for this?

joakim-tjernlund, 2019-12-02:
vaguelyrembering something similar, you might need fwupd >= 1.3.4

hughsie, 2019-12-06:
Unsupported means the existing uefi firmware is unable to update. If you update to something newer using a CD-RW from Lenovo manually, then the bios will be new enough to use with fwupd.

aravindhp, 2019-12-06:
I already did that for the BIOS (system-firmware) and am at the latest version.

aravindhp, 2019-12-06:
I don't see a way to update the `UEFI Device Firmware` using an ISO, and this is what is failing for me through `fwupdatemgr`.

joakim-tjernlund, 2019-12-06:
download the ISO ig, use geteltorito to convert it to USB format, then in BIOS mode you update from your USB stick

aravindhp, 2019-12-06:
There is no ISO for updating the `UEFI Device Firmware` at [Lenovo support](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470s/20hg/20hgs22d00/downloads/driver-list/). I could only find an ISO for [System Firmware](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t470s/20hg/20hgs22d00/downloads/DS120418) and I already used the ISO to update that.

joakim-tjernlund, 2019-12-06:
hmm, and there is no EFI mode you can select in that BIOS ?

aravindhp, 2019-12-06:
There is and that is what I am using. Otherwise I wouldn't even get far enough for `fwupdmgr` to pick up that there are updates available.

cryptomilk, 2020-01-06:
I'm running into the same issue with a Lenovo X1 Carbon Gen6!

fwupd is version 1.3.6.
Updating the firmware fails with the described error from above.

I've updated the BIOS via the CD image from the Lenovo website a month ago. Updating to the latest version fails again.

cryptomilk, 2020-01-06:
The System Firmware update with fwupd got correctly installed. However the UEFI Firmware always fails.

```
Upgrade available for UEFI Device Firmware from 184.55.3510 to 184.70.3626
Upgrade available for UEFI Device Firmware from 0.1.18 to 0.1.19
```

It could also be a bug in the Lenovo update routine. Is there a way to report bugs to Lenovo?

cryptomilk, 2020-01-29:
Just updated with [BIOS Update from Bootable CD](https://pcsupport.lenovo.com/de/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/downloads/driver-list/component?name=BIOS%2FUEFI) and if I run `fwupdmgr update` it offers an update for:

```
20KGS23S10                                                                                    
│                                                                                                                                                                                            
└─UEFI Device Firmware:                                                                       
  │   Device ID:           e563ad307df81c99f0de8c26292afd71cf409673
  │   Current version:     184.55.3510
```

If I try to update this it fails with the mentioned error.

cryptomilk, 2020-03-01:
...

Jamstah, 2020-04-05:
Seeing the same error on a Lenovo P52:

```
dpath: \EFI\fedora\shimx64.efi
path: \fwupdx64.efi
Found update fwupd-<id>
Adding new capsule
Adding new capsule
Adding new capsule
WARNING: Could not query capsule capabilities: Unsupported
WARNING: Could not apply capsules: Unsupported
start_image() returned Unsupported
```

Running Fedora 31:
```
[jammy@ibm007470 ~]$ rpm -qa | grep fwupd
fwupd-1.3.9-2.fc31.x86_64
```

Based on the recommendations here I updated the BIOS with ISO to make sure it was up to date, so that's at latest.

This one says it updated, then on reboot, I get the message above:
```
[jammy@ibm007470 ~]$ sudo fwupdmgr get-updates
[sudo] password for jammy: 
• Thunderbolt Controller has the latest available firmware version
• CT500BX100SSD1 has no available firmware updates
• SAMSUNG MZVLB512HAJQ-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
20MAS1721M
│
└─UEFI Device Firmware:
  │   Device ID:           619057f1d0da67b3fa4e9f74e09fc9218ce32643
  │   Current version:     192.10.1127
  │   Minimum Version:     0.0.1
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                971682af-21a0-46e3-91c0-825702c2ba70
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │ 
  ├─ThinkPad P52/ThinkPad P72 Corporate ME Update:
  │     New version:       192.47.1524
  │     Remote ID:         lvfs
  │     Summary:           Lenovo ThinkPad P52/P72 Corporate ME Firmware
  │     Licence:           Proprietary
  │     Size:              12.2 MB
  │     Vendor:            Lenovo Ltd.
  │     Flags:             is-upgrade
  │     Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)
  │     
  │     Version 12.0.47.1524 (LVFS: 192.47.1524)
  │     
  │     Problem Fixes
  │     
  │      • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
  │      • Mitigated the following security vulnerabilities under issues:
  │     
  │     Security issues fixed:
  │     
  │      • CVE-2019-0086
  │      • CVE-2019-0090
  │      • CVE-2019-0092
  │      • CVE-2019-0093
  │      • CVE-2019-0094
  │      • CVE-2019-0096
  │      • CVE-2019-0097
  │      • CVE-2019-0098
  │      • CVE-2019-0153
  │      • CVE-2019-0170
  │   
  └─ThinkPad P52/ThinkPad P72 Corporate ME Update:
        New version:       192.35.1427
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad P52/P72 Corporate ME Firmware
        Licence:           Proprietary
        Size:              12.2 MB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       • 0 Quarterly Security Release
        
        Version 12.0.35.1427 (LVFS: 192.35.1427)
        
        Problem Fixes
        
         • Intel CSME PSIRT-TA-201901-002 Q1 2019 QSR:
        
        Security issues fixed:
        
         • CVE-2019-0086
         • CVE-2019-0090
         • CVE-2019-0092
         • CVE-2019-0093
         • CVE-2019-0094
         • CVE-2019-0096
         • CVE-2019-0097
         • CVE-2019-0098
         • CVE-2019-0153
         • CVE-2019-0170
```

aravindhp, 2020-05-02:
Continue to see this issue with newer versions of `fwupdmgr`
```
 ~  fwupdmgr --version
client version:	1.4.1
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.1
```

mmckinst, 2020-06-01:
It is still happening.

bliemli, 2020-06-02:
Same problem here on a Lenovo Thinkpad T480s:

```
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-676af093-2a5c-4238-9c29-db8063a33532-0
../plugins/uefi/efi/fwupdate.c:577:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(1,MBR,0xE222E51F)/\EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:415:fwup_add_update_capsule(): Read file; 57538 bytes
../plugins/uefi/efi/fwupdate.c:416:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../plugins/uefi/efi/fwupdate.c:417:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../plugins/uefi/efi/fwupdate.c:434:fwup_add_update_capsule(): Checking GOP for ux capsule
../plugins/uefi/efi/fwupdate.c:577:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(1,MBR,0xE222E51F)/\EFI\fedora\fw\fwupd-676af093-2a5c-4238-9c29-db8063a33532.cap...
../plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Pci(0x0,0x0)/Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and Path(131,23,0x01000000A0329941DA01129684012C00)/HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:231:fwup_search_file(): Comparing: HD(1,MBR,0xE222E51F) and HD(1,MBR,0xE222E51F)
../plugins/uefi/efi/fwupdate.c:240:fwup_search_file(): Match up! Returning \EFI\fedora\fw\fwupd-676af093-2a5c-4238-9c29-db8063a33532.cap
../plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../plugins/uefi/efi/fwupdate.c:415:fwup_add_update_capsule(): Read file; 2210336 bytes
../plugins/uefi/efi/fwupdate.c:416:fwup_add_update_capsule(): updates guid: 676AF093-2A5C-4238-9C29-DB8063A33532
../plugins/uefi/efi/fwupdate.c:417:fwup_add_update_capsule(): File guid: 676AF093-2A5C-4238-9C29-DB8063A33532
../plugins/uefi/efi/fwupdate.c:593:efi_main(): n_updates: 2
../plugins/uefi/efi/fwupdate.c:458:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported
../plugins/uefi/efi/fwupdate.c:608:efi_main(): Could not apply capsules: Unsupported
```

Running Fedora 32:

```
rpm -q fwupd
fwupd-1.4.2-2.fc32.x86_64
```

Manually updated BIOS so everything else is on the latest version except the "ThinkPad T480s Consumer ME Update" which cannot be applied because of above error.

```
sudo fwupdmgr get-updates 
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• LENSE30512GMSP34MEAT3TA has the latest available firmware version
• System Firmware has the latest available firmware version
• UEFI Device Firmware has no available firmware updates
20L7001LMZ
│
└─Intel Management Engine:
  │   Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  │   Current version:     184.70.3626
  │   Minimum Version:     184.70.3626
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                676af093-2a5c-4238-9c29-db8063a33532
[... cut ...]
        2020.01 Q1 Intel Platform Update (IPU), formerly known as Quarterly security release (QSR)
        
        Several security fixes and enhancements are on this release.
        
        Security issues fixed:
        
         • CVE-2020-0531
         • CVE-2020-0532
         • CVE-2020-0533
         • CVE-2020-0535
         • CVE-2020-0536
         • CVE-2020-0537
         • CVE-2020-0538
         • CVE-2020-0539
         • CVE-2020-0540
         • CVE-2020-0545
         • CVE-2020-0594
         • CVE-2020-0595
         • CVE-2020-0596
         • CVE-2020-8674
```

joakim-tjernlund, 2020-06-02:
Maybe the Consumer ME Update is broken? My T480s works with the Enterprise ME Update

bliemli, 2020-06-02:
> Maybe the Consumer ME Update is broken? My T480s works with the Enterprise ME Update

Firmware updates using fwupd haven't been working for me for several months now with the exact same error message as I reported above.