fwupd/fwupd #1617: fwupdmgr asked me if i wanted to reboot, i said yes, and nothing happened

albert-astals-cid-kdab, 2019-11-30:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

```shell
$ fwupdmgr get-devices
$ LC_ALL=en_US.UTF-8 fwupdmgr get-devices
XPS 15 9570
│
├─Thunderbolt Controller:
│     Device ID:           6075dae7d92d8a13e157f179fe41d008c140c71a
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               560774f3-821b-5ebb-9f1a-97cf34ce266f
│                          d3826540-45c3-5f3c-89d9-b21b27c6ad77
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─KXG50ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     AADA4105
│     Vendor:              Toshiba America Info Systems (NVME:0x1179)
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9
│                          2ce5641c-516e-5c6b-908f-6279409807b5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           66352e92895dc6d018071ad4ecfae249b0b8af0d
│     Current version:     0.1.13.0
│     Minimum Version:     0.1.13.0
│     Vendor:              Dell Inc.
│     GUID:                e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│     Device ID:           f80c30caae6a1e4a4f8b403e53e5c726320e224c
│     Summary:             Platform TPM device
│     Current version:     7.2.0.1
│     Vendor:              Dell Inc.
│     Update Error:        Updating disabled due to TPM ownership
│     GUIDs:               abe4e2d2-3fd5-5f0c-9613-7a04e8e048a2
│                          7d65b10b-bb24-552d-ade5-590b3b278188
│                          6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d
│                          fe462d4a-e48f-5069-9172-47330fc5e838
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
└─Touchpad:
      Device ID:           881d0bf9902910255d18c15a368c4106b26fee62
      Current version:     1.7.2378871
      Bootloader Version:  54.0
      Vendor:              Synaptics Inc (HIDRAW:0x06CB)
      GUIDs:               1d6705d0-3fa6-58d2-bd0e-806ed6792c18
                           2aa17264-7ef0-5cc5-9368-8595974a5e31
                           741692a4-ae19-5ecc-9e48-e644604c2345
                           a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8
      Device Flags:        • Internal device
                           • Updatable
    
```

Please answer the following questions:

- Operating system and version: 
ArchLinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
yes
- Are you using an NVMe disk?
i think so

After updating the firmware of my laptop (well wasn really updated, needed to reboot so the actual update took place), fwupdmgr asked me if i wanted to reboot, i said yes, and nothing happened.

I was running fwupdmgr as root. Desktop is Plasma 5.17.3

How do you issue a reboot command?

hughsie, 2019-11-30:
> How do you issue a reboot command?

https://github.com/fwupd/fwupd/blob/master/src/fu-util-common.c#L350

So it depends on how you compiled fwupd.

albert-astals-cid-kdab, 2019-12-02:
the archlinux one uses fwupd and calling 
`qdbus --system org.freedesktop.login1 /org/freedesktop/login1 org.freedesktop.login1.Manager.Reboot true`
definitely forces a reboot here.

So let's assume i probably said "no" instead of "yes" and it's just a PBKAC.

I'll watch for the next time it happens and try to reproduce it again.

albert-astals-cid-kdab, 2019-12-11:
It actually happened again.

I think i found the problem, code says

```
g_print ("\n%s %s [Y|n]: ",
				 /* TRANSLATORS: explain why we want to reboot */
				 _("An update requires a reboot to complete."),
				 /* TRANSLATORS: reboot to apply the update */
				 _("Restart now?"));
			if (!fu_util_prompt_for_boolean (FALSE))
				return TRUE;
```

Saying that if i just press enter it will be a Yes `[Y|n]` but the `gboolean def` passed to  `fu_util_prompt_for_boolean` is `FALSE`.

Did i read the code correctly?

hughsie, 2019-12-11:
@superm1 ?

superm1, 2019-12-11:
Just pressing enter should avoid the reboot - hence the FALSE passed for default.

Pressing Y followed by enter should reboot it.

albert-astals-cid-kdab, 2019-12-11:
Then the code should say `[y|N]` instead of `[Y|n]`, right?

superm1, 2019-12-11:
The code actually doesn't care about the case. It will treat Y and y the same.

albert-astals-cid-kdab, 2019-12-11:
i think you don't understand the nomenclature that has been long established in CLI apps, of the capital letter being the one that is the default.

So [Y|n] means yes is the default and [y|N] means no is the default

superm1, 2019-12-11:
Ah yes thanks for explaining that.  I wasn't aware, PR incoming.

commit 5d94afae (2019-12-11): fu-util-common: correct the default prompt for reboot/shutdown (Fixes: #1617)

Y/n means that Y is the default if you press enter.
y/N means that N is the default if you press enter.
closed 2019-12-11 by superm1