fwupd/fwupd #1618: fwupdmgr upgrade gives a message that is hard to understand

tsdgeos, 2019-11-30:
```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

```shell
$ fwupdmgr get-devices
LC_ALL=en_US.UTF-8 fwupdmgr get-devices 
20EM000QSP
│
├─LITEON LCH-256V2S:
│     Device ID:           5dbeb140337f610d54913c0f43d15fdaa0eafb51
│     Summary:             ATA Drive
│     Current version:     3C87902
│     Vendor:              Intel Corporation
│     Serial Number:       SD0F66157L2TH6110FN0
│     GUIDs:               ae7ea9e0-b497-5b7b-bce1-fc358bdf84ad ← IDE\LITEON_LCH-256V2S_______________________3C87902
│                          f5caef3f-05a8-5bb3-8e25-b03a18619f24 ← IDE\0LITEON_LCH-256V2S_______________________
│                          ad6b029a-e346-545e-a360-fa076902a208 ← LITEON LCH-256V2S
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           8be4cbd5f33a57bd5ac69de9004c4e5ecc91325f
│     Current version:     0.0.49
│     Minimum Version:     0.0.49
│     Vendor:              LENOVO
│     GUID:                b8d73eb5-feba-42e1-a106-04540b6ae7d6
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           aff24b50c0a8c292c6c2a81d3827c36d7a86f62a
      Current version:     184.50.3425
      Minimum Version:     184.50.3425
      GUID:                03d3297b-3851-4379-95ad-12e21a96c80a
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    
```

Please answer the following questions:

- Operating system and version:
Archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
yes
- Are you using an NVMe disk?
no idea

So i run
```shell
 # fwupdmgr upgrade
No releases found for device: no HWIDs matched 90706264-e399-575b-a9fb-077ead03b9b4
```
What does that mean? I don't even have any decide with that ID according to the get-devices call.

hughsie, 2019-11-30:
>What does that mean?

@superm1 -- I'm thinking for "valid" failures like this, I think we should just show "no updates available" and then only show this message in debug mode or something? Or maybe just show it in the daemon logs?

superm1, 2019-11-30:
My interpretation of this error is that there is an upgrade that matches the GUID of something but it's restricted to a CHID that doesn't match this system.  If that's the case yes; I agree we should not be showing that in `get-updates` or `upgrade/update` output.  It should be relegated to debugging logs in that case.

tsdgeos, 2019-11-30:
FWIW i am 94% sure that for a moment i was suggested an update, then said no and now i can't get an update anymore :D

Not sure if a newer data unlisted my device from the update or something

superm1, 2019-12-01:
@tsdgeos 

Was this message before you ran `fwupdmgr refresh` for the very first time?

tsdgeos, 2019-12-01:
You mean the "hard to understand" or the "you have an update one"?

The hard to understand is now, all the time even if i run refresh.

The "you have an update" i am not really sure, i had installed (and probably run) fwupdmgr months ago, and yesterday i ran it again, but not sure if i ran get-upgrades or upgrade first, sorry :/

superm1, 2019-12-09:
fixed by https://github.com/fwupd/fwupd/pull/1647

closed 2019-12-09 by superm1