fwupd/fwupd #1619: Logitech K850 bricked after firmware update

flaphoschi, 2019-11-30:
Hello!

I received an update for my Logitech K850 (without M, I'm not using the mouse from the combo) keyboard, not the Unifying-Receiver. The update failed with an error message and bricked my keyboard, it doesn't work anymore. There is no reset pin or something like that.

During update I got a failed message from *gnome-software* and the LEDs on the keyboard just kept flashing.

[image]

```shell
$ fwupdmgr --version
client version:	1.3.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.3

```

```shell
$ fwupdmgr get-devices
MS-7C02
│
├─Samsung SSD 970 EVO 250GB:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:             NVM Express Solid State Drive
│     Current version:     2B2QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          c198ada2-cf92-5e2e-a7d0-fb7fa0fde599
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           9d9b6d05ba1c30b381f02df0dc8bca472b60400e
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server

```
Not visible on this output is the Non-Unifying-Receiver of my mouse, a Logitech G603.

- Fedora 31 X86_64:
- already installed by distribution
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

[Final result for your amusement. I'm sorry for MP4.](https://i.imgur.com/UBwfOH1.mp4)

Despite christmas time is coming, I don't appreciate this light show. I will contact Logitech, they are responsible for the firmware and the security issues.

Thanks

FFY00, 2019-11-30:
What likely happened is that the device was put into bootloader mode. Please replug your device and see if the issue persists.

flaphoschi, 2019-12-01:
That doesn't impress the keyboard a lot :wink:

closed 2020-01-07 by stale[bot]