fwupd/fwupd #1626: fwupd build issues with flashrom support

kloczek, 2019-12-01:
Last release 1.1 of the flashrom https://review.coreboot.org/cgit/flashrom.git is not installing .pc file however I've been looking on git version which already has meson build suit. I've added on top of 1.1 
https://review.coreboot.org/cgit/flashrom.git/patch/?id=cb97368 which adds meson builds suite and I found some issues:
- flashrom does not provide flashrom.pc but libflashrom.pc
- after building flashrom with that patch and correcting in meson.buid s/flashrom/libflashrom/ build fails with
<pre>[tkloczko@barrel fwupd-1.3.5]$ /usr/bin/ninja -v -j48 -C x86_64-redhat-linux-gnu
ninja: Entering directory `x86_64-redhat-linux-gnu&apos;
[1/2] gcc -Iplugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha -Iplugins/flashrom -I../plugins/flashrom -I. -I../ -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/gudev-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 &apos;-DMM_REQUIRED_VERSION=&quot;1.10.0&quot;&apos; &apos;-DQMI_REQUIRED_VERSION=&quot;1.23.1&quot;&apos; -O2 -g -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -fPIC -pthread &apos;-DG_LOG_DOMAIN=&quot;FuPluginFlashrom&quot;&apos; &apos;-DLOCALSTATEDIR=&quot;/var&quot;&apos; -MD -MQ &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o&apos; -MF &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o.d&apos; -o &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o&apos; -c ../plugins/flashrom/fu-plugin-flashrom.c
FAILED: plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o 
gcc -Iplugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha -Iplugins/flashrom -I../plugins/flashrom -I. -I../ -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -I/usr/include/libxmlb-1 -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/gudev-1.0 -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=c99 -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 &apos;-DMM_REQUIRED_VERSION=&quot;1.10.0&quot;&apos; &apos;-DQMI_REQUIRED_VERSION=&quot;1.23.1&quot;&apos; -O2 -g -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none -fPIC -pthread &apos;-DG_LOG_DOMAIN=&quot;FuPluginFlashrom&quot;&apos; &apos;-DLOCALSTATEDIR=&quot;/var&quot;&apos; -MD -MQ &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o&apos; -MF &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o.d&apos; -o &apos;plugins/flashrom/1ac62b6@@fu_plugin_flashrom@sha/fu-plugin-flashrom.c.o&apos; -c ../plugins/flashrom/fu-plugin-flashrom.c
In file included from <b>../plugins/flashrom/fu-plugin-flashrom.c:28</b>:
<b>/usr/include/libflashrom.h:57:72:</b> <font color="#CC0000"><b>error: </b></font>unknown type name ‘<b>bool</b>’
   57 | void flashrom_flag_set(struct flashrom_flashctx *, enum flashrom_flag, <font color="#CC0000"><b>bool</b></font> value);
      |                                                                        <font color="#CC0000"><b>^~~~</b></font>
<b>/usr/include/libflashrom.h:58:1:</b> <font color="#CC0000"><b>error: </b></font>unknown type name ‘<b>bool</b>’
   58 | <font color="#CC0000"><b>bool</b></font> flashrom_flag_get(const struct flashrom_flashctx *, enum flashrom_flag);
      | <font color="#CC0000"><b>^~~~</b></font>
<b>/usr/include/libflashrom.h:69:37:</b> <font color="#CC0000"><b>error: </b></font>unknown type name ‘<b>uint8_t</b>’
   69 |   struct flashrom_flashctx *, const <font color="#CC0000"><b>uint8_t</b></font> *buf, size_t len);
      |                                     <font color="#CC0000"><b>^~~~~~~</b></font>
ninja: build stopped: subcommand failed.
</pre>

So question is what is the current status of the flashrom plugin support?
Whichj one version of the flashrom should be used?  latest git tree?

superm1, 2019-12-02:
Pretty sure you need master still. Upstream hasn't tagged a release since we got all the fwupd build stuff fixed.

superm1, 2019-12-02:
Yep, minimum commit you need is  https://github.com/flashrom/flashrom/commit/2f6936bd926b6d4f21680e2cdc160fc580c3ecb3

superm1, 2019-12-02:
there are a handful of others commits needed too for it to build on 1.1.  Notable ones:
https://github.com/flashrom/flashrom/commit/d954d5d700d7aa2ff144e7f8a5a2b6df08b07590
https://github.com/flashrom/flashrom/commit/2a8d4390fa85160b79f9ccf3d2a8c676595611e7

closed 2019-12-06 by hughsie