fwupd/fwupd #1631: Can't update Dell OptiPlex 7060 due to different firmware version formats

sbstnk, 2019-12-03:
I get an error message about different firmware version formats when trying to update the firmware on this machine:

```shell
$ fwupdmgr update
No releases found for device: Firmware version formats were different, device was 'triplet' and release is 'quad'
```

```shell
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5
```

```shell
$ fwupdmgr get-devices
OptiPlex 7060
│
└─System Firmware:
      Device ID:           c2a15756c7442750fba0ce09a503930d64db5c7a
      Current version:     0.1.772
      Minimum Version:     0.1.772
      Vendor:              Dell Inc.
      Update State:        success
      GUID:                df35f706-c014-49e1-81e7-029945a45b12
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Cryptographic hash verification is available
                           • Device is usable for the duration of the update
```

```shell
fwupdmgr get-history
OptiPlex 7060
│
├─OptiPlex 7060 System Firmware:
│ │   Device ID:           c2a15756c7442750fba0ce09a503930d64db5c7a
│ │   Previous version:    0.1.1.4
│ │   Update State:        needs-reboot
│ │   Last modified:       2019-02-11 14:39
│ │   GUID:                df35f706-c014-49e1-81e7-029945a45b12
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │ 
│ └─  New version:       0.1.534
│       Remote ID:         lvfs
│       License:           Unknown
│     
└─OptiPlex 7060 System Firmware:
  │   Device ID:           c2a15756c7442750fba0ce09a503930d64db5c7a
  │   Previous version:    0.1.2.22
  │   Update State:        needs-reboot
  │   Last modified:       2019-02-11 14:39
  │   GUID:                df35f706-c014-49e1-81e7-029945a45b12
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │ 
  └─  New version:       0.1.3.4
        Remote ID:         lvfs
        License:           Unknown
```

This is on Fedora 31, the exact package version is `fwupd-1.3.5-1.fc31.x86_64`. The issue persists across reboots.

superm1, 2019-12-03:
Are you running a system with modified SMBIOS tables?  Can you please share the output of `/usr/libexec/fwupd/fwupdtool hwids`?

sbstnk, 2019-12-03:
I did not change anything about the SMBIOS tables.
```shell
$ sudo /usr/libexec/fwupd/fwupdtool hwids
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.3.4
BiosMajorRelease: 1
BiosMinorRelease: 3
Manufacturer: Dell Inc.
Family: OptiPlex
ProductName: OptiPlex 7060
ProductSku: 85A
EnclosureKind: 3
BaseboardManufacturer: Dell Inc.
BaseboardProduct: NC2VH

Hardware IDs
------------
{ba2d1af7-ba41-5ba0-9a31-ea465b7b9817}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{50b75c2c-858a-5c2c-a3c7-29e3a7c07985}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{c146f951-9561-5c23-ae47-0e7a0c8779d7}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{0a3ce07d-ef86-519d-8e53-20aed0891e86}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{24ab10ef-1e62-55ca-8a6d-1eea6ba257a1}   <- Manufacturer + Family + ProductName + ProductSku
{4d61a1b6-f2b6-5deb-ad8b-1b0d74c7c3cf}   <- Manufacturer + Family + ProductName
{5fe15cfc-7469-5bb2-a882-df05934d073e}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{e391450c-c4ac-5442-81e9-1407826748d5}   <- Manufacturer + ProductSku
{2c06ca9e-9792-5c7e-bc52-507a5a316b4f}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{f1181f97-f3ad-5297-b434-1a16e7e60d8d}   <- Manufacturer + ProductName
{6ab099b2-25fd-50e1-b278-f1030c6a2326}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{91251071-a01b-5067-9dca-81de39f555f9}   <- Manufacturer + Family
{529aea4b-4247-5705-89e2-edc2edd08c4e}   <- Manufacturer + EnclosureKind
{14d0daea-80e7-51ee-9fdf-0adba458e7cf}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

Also I'm seeing the same error message on my Dell XPS 13 9370 as well. fwupd used to work on both of these machines until a few weeks ago.

Here are all the outputs for the XPS 13 9370 as well:
```shell
$ fwupdmgr get-devices
XPS 13 9370
│
├─Thunderbolt Controller:
│     Device ID:           a817ac92399b77c3cdfa2a439925a88d94397abb
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
│                          31dedd69-1429-5224-87d4-a8fb8072a50e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─System Firmware:
│     Device ID:           8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│     Current version:     0.1.2817
│     Minimum Version:     0.1.2817
│     Vendor:              Dell Inc.
│     Update State:        success
│     GUID:                7ceaf7a8-0611-4480-9e30-64d8de420c7c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
[... 116 lines cut ...]
Family: XPS
ProductName: XPS 13 9370
ProductSku: 7E6
EnclosureKind: a
BaseboardManufacturer: Dell Inc.
BaseboardProduct: F6P3V

Hardware IDs
------------
{9803e4f0-6cdb-5ead-a2de-9351e214ffb0}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{1d03431b-bebb-5bda-bbbf-c78a7c73d937}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{7bd4b150-cf72-5056-bbbb-95b529aa07de}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{a00e4d4d-247b-55f9-8869-e83553e74570}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{d72b8452-80b6-5b66-b95c-8dd75519a30c}   <- Manufacturer + Family + ProductName + ProductSku
{184da599-9986-5212-9dba-226d2d233133}   <- Manufacturer + Family + ProductName
{f51ef121-403c-5ebf-8ccc-28b7a3e36b44}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{74bb5080-7e51-5373-aea1-bfc218a2caaf}   <- Manufacturer + ProductSku
{04840070-fce8-57d3-bec4-2ecebebdecaf}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{f33883b9-e996-539e-b35a-73568b903d13}   <- Manufacturer + ProductName
{b350638b-bc1b-57d9-9946-5a0f555d95e7}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{61178075-a8fd-563c-9045-44227d8c121f}   <- Manufacturer + Family
{5c163113-9296-5c8d-a92d-ae04eb59a0f7}   <- Manufacturer + EnclosureKind
{7ec53000-14a8-55ad-9834-388b0101c8e3}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

superm1, 2019-12-03:
It almost looks like the SMBIOS quirk file isn't getting loaded anymore.  Can you see if you have `/usr/share/fwupd/quirks.d/dell.quirk` on your system?

sbstnk, 2019-12-03:
The file exists and is readable on both machines. The relevant lines in that file also are present on both machines:
```
[SmbiosManufacturer=Dell Inc.]
UefiVersionFormat = quad
```

superm1, 2019-12-03:
Hmm, can you please run the daemon verbose and share it's output?

sbstnk, 2019-12-03:
You were right about the quirk not getting loaded, but the reason for that is completely unexpected:
```
19:36:08:0551 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
19:36:08:0551 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
19:36:08:0551 XbSilo               failed to load silo: Failed to open file “/var/cache/fwupd/quirks.xmlb”: open() failed: No such file or directory
19:36:08:0551 XbSilo               compiling /usr/share/fwupd/quirks.d/altos.quirk:ctime=1575394227.650069…
19:36:08:0551 XbSilo               compiling /usr/share/fwupd/quirks.d/ata.quirk:ctime=1575394227.650069…
19:36:08:0552 FuQuirks             failed to build silo: failed to compile /usr/share/fwupd/quirks.d/ata.quirk:ctime=1575394227.650069: cannot process content type chemical/x-turbomole-vibrational
```
After uninstalling `chemical-mime-data` fwupd works again. And the reason for that is found in `/usr/share/mime/packages/chemical-mime-data.xml` in that package:
```
  <mime-type type="chemical/x-turbomole-vibrational">
    <comment>Turbomole Vibrational Data File</comment>
    <glob pattern="???"/>
    <sub-class-of type="text/plain"/>
  </mime-type>
```

I'm guessing fwupd or libxmlb is trying to create a xmlb file with an exactly 3 character long file name and if `chemical-mime-data` is installed, this gets assigned the `chemical/x-turbomole-vibrational` mime type which is not considered a supported mime type.

I'm not sure if anything needs to be done from the fwupd side here, so feel free to close this. On the other hand it seems like the upstream bug tracker for `chemical-mime-data` is gone along with their mailing list, so this might need to be fixed by all downstreams individually. In that case changing the file name or the mime type check could avoid some spurious bug reports.

hughsie, 2019-12-03:
> I'm guessing fwupd or libxmlb is trying to create a xmlb file with an exactly 3 character long file name

Nahh, it's getting the content type of a three letter filename, e.g. `ata.quirk`

> I'm not sure if anything needs to be done from the fwupd side here, so feel free to close this.

I suspect quite a lot of other weird things are happening on your computer right now. Try opening `foo.txt` and see what happens :)

closed 2019-12-03 by hughsie

sbstnk, 2019-12-03:
> Nahh, it's getting the content type of a three letter filename, e.g. ata.quirk

That should be working fine:
```
$ gio list -a 'standard::content-type' /usr/share/fwupd/quirks.d/
dfu.quirk	8394	(regular)	standard::content-type=text/plain
coreboot.quirk	71	(regular)	standard::content-type=text/plain
colorhug.quirk	1555	(regular)	standard::content-type=text/plain
ebitdo.quirk	2009	(regular)	standard::content-type=text/plain
emmc.quirk	84	(regular)	standard::content-type=text/plain
fastboot.quirk	93	(regular)	standard::content-type=text/plain
logitech-hidpp.quirk	1116	(regular)	standard::content-type=text/plain
optionrom.quirk	87	(regular)	standard::content-type=text/plain
rts54hid.quirk	413	(regular)	standard::content-type=text/plain
steelseries.quirk	127	(regular)	standard::content-type=text/plain
synaptics-prometheus.quirk	183	(regular)	standard::content-type=text/plain
synapticsmst.quirk	1323	(regular)	standard::content-type=text/plain
altos.quirk	247	(regular)	standard::content-type=text/plain
dell-dock.quirk	3463	(regular)	standard::content-type=text/plain
dell.quirk	979	(regular)	standard::content-type=text/plain
synaptics-cxaudio.quirk	645	(regular)	standard::content-type=text/plain
uefi-recovery.quirk	137	(regular)	standard::content-type=text/plain
nvme.quirk	161	(regular)	standard::content-type=text/plain
vli-usbhub.quirk	2459	(regular)	standard::content-type=text/plain
ata.quirk	83	(regular)	standard::content-type=text/plain
csr-aiaiai.quirk	337	(regular)	standard::content-type=text/plain
jabra.quirk	547	(regular)	standard::content-type=text/plain
nitrokey.quirk	218	(regular)	standard::content-type=text/plain
rts54hub.quirk	139	(regular)	standard::content-type=text/plain
solokey.quirk	90	(regular)	standard::content-type=text/plain
superio.quirk	665	(regular)	standard::content-type=text/plain
synaptics-rmi.quirk	58	(regular)	standard::content-type=text/plain
thelio-io.quirk	90	(regular)	standard::content-type=text/plain
uefi.quirk	751	(regular)	standard::content-type=text/plain
vli-usbhub-lenovo.quirk	3984	(regular)	standard::content-type=text/plain
wacom-raw.quirk	2243	(regular)	standard::content-type=text/plain
wacom-usb.quirk	873	(regular)	standard::content-type=text/plain
modem-manager.quirk	1164	(regular)	standard::content-type=text/plain
```

The problematic file does not seem to come from that directory, but maybe somewhere the file name without the extension is used to create a temporary file. In that case `ata.quirk` might indirectly be the problematic file name.

> I suspect quite a lot of other weird things are happening on your computer right now. Try opening foo.txt and see what happens :)

Actually I never had any noticeable issues and I've had this package installed on many machines over several years. Also it won't affect `foo.txt`, the pattern applies to the entire file name including the extension. And it only seems to affect non-empty files:
```
$ echo 'hello' > foo.txt
$ gio info -a 'standard::content-type' foo.txt 
uri: file:///home/skeller/foo.txt
attributes:
  standard::content-type: text/plain

$ touch foo
$ gio info -a 'standard::content-type' foo
uri: file:///home/skeller/foo
attributes:
  standard::content-type: text/plain

$ echo 'hello' > foo
$ gio info -a 'standard::content-type' foo
uri: file:///home/skeller/foo
attributes:
  standard::content-type: chemical/x-turbomole-vibrational
```

I will open a bug report against the `chemical-mime-data` Fedora package.

superm1, 2019-12-03:
> The problematic file does not seem to come from that directory, but maybe somewhere the file name without the extension is used to create a temporary file

I think this might be a side effect of libxmlb if i'm following code right:

Compile the file:
https://github.com/hughsie/libxmlb/blob/ec3c881e8850ffb815052b06351fd9c547cbc89c/src/xb-builder.c#L686

Compile source:
https://github.com/hughsie/libxmlb/blob/ec3c881e8850ffb815052b06351fd9c547cbc89c/src/xb-builder.c#L209

Get input stream:
https://github.com/hughsie/libxmlb/blob/ec3c881e8850ffb815052b06351fd9c547cbc89c/src/xb-builder.c#L237
https://github.com/hughsie/libxmlb/blob/31fb23fb49617b9dc7f9c7edd55b35ecf89a3b4a/src/xb-builder-source.c#L435

It removes the last extension:
https://github.com/hughsie/libxmlb/blob/31fb23fb49617b9dc7f9c7edd55b35ecf89a3b4a/src/xb-builder-source.c#L485

So that means `ata.quirk` gets passed into the lookup later as `ata`.

@hughsie maybe need another clause here https://github.com/hughsie/libxmlb/blob/31fb23fb49617b9dc7f9c7edd55b35ecf89a3b4a/src/xb-builder-source.c#L468 for if you get the text/plain content type.

hughsie, 2019-12-04:
> So that means ata.quirk gets passed into the lookup later as ata.

It kinda has to. If you have `<xml>foo</xml>` in a file called `ata.desktop` then `g_content_type_guess` will only return `application-x-desktop` without actually sniffing the contents. We remove one layer of the file extension each time so that for the last iteration we actually **have** to sniff the contents rather than doing a glob match.