fwupd/fwupd #1633: Unsupported Daemon version number in proportion to client number

hokuco, 2019-12-03:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.10
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Unsupported daemon version 1.3.5, client version is 1.2.10
```

```shell
$ efibootmgr -v
BootCurrent: 0003
Timeout: 2 seconds
BootOrder: 0003
Boot0000* Windows Boot Manager  HD(1,GPT,3f849dbe-9a65-418c-a407-cbfb60c3909a,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0003* ubuntu        HD(1,GPT,b6de2bd2-709f-4820-a6e4-4de2082644b8,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0004* Windows Boot Manager  HD(2,GPT,6e02f841-2b1c-4da3-80e3-51e05479aced,0xfa000,0xf9800)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
**This is only required if you use the UEFI plugin**
```
I'm pretty sure I have the UEFI plugin because the readme [here](https://github.com/fwupd/fwupd/tree/master/plugins/uefi) says it is installed by default. Also, I'm not sure why anything to do with windows is listed on the efibootmanager, as this is a fresh SSD that has only ever had linux installed on it.
```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-53f51f56-eff2-4516-9dda-ed501b3f0f8a-0
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
there were no cap files or any mention of fwupx64.efi
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: kubuntu 19.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Snap
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? yes

My problem is fairly apparent from what I've provided, but in short, my problem is that I get this error when trying to scan for devices: `Unsupported daemon version 1.3.5, client version is 1.2.10`. My laptop, the Dell Latitude 7280 is [listed on your products page](https://fwupd.org/lvfs/devices/com.dell.tbta2bfa54c.firmware), and should be supported, as your readme file states that any version greater than or equal to 16.04 should be supported.

hughsie, 2019-12-03:
If you're using snap you should be using `fwupd.fwupdmgr` rather than `fwupdmgr` directly.

superm1, 2019-12-03:
Yes; as mentioned on the wiki: https://github.com/fwupd/fwupd/wiki/fwupd-snap

1. either remove the apt packaged version first (`snap remove fwupd`; `apt purge fwupd`; `snap install fwupd --classic`)
or
2. Use the prefixed version as mentioned above.

hokuco, 2019-12-03:
Thank you! That worked, but I'm also wondering if
A. It's abnormal not to have cap/fwup files show up when run `tree /boot`
B. there's any way to tell if I have the UEFI extension installed, and if not if there's anything I can do about it

superm1, 2019-12-03:
> A. It's abnormal not to have cap/fwup files show up when run tree /boot

They're only going to be there after you've run an upgrade.

> B. there's any way to tell if I have the UEFI extension installed, and if not if there's anything I can do about it

It's shipped in the snap.

hokuco, 2019-12-03:
Ok thanks!

closed 2019-12-03 by hokuco