fwupd/fwupd #1634: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

scareduck, 2019-12-04:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
[rlm@localhost ~]$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Inspiron 3847
│
├─System Firmware:
│     Device ID:           1d70a31dd3ff0941761550a38436fc990c6c3ff0
│     Current version:     0.1.0.6
│     Minimum Version:     0.1.0.6
│     Vendor:              Dell Inc.
│     GUID:                7039436b-6acf-433b-86a1-368ec2ef7e1f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─WDC WD10EZEX-75M2NA0:
      Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
      Summary:             ATA Drive
      Current version:     01.01A01
      Vendor:              Intel Corporation
      GUIDs:               d1dd99cb-6def-574e-a5d9-79d4f5111f62
                           fa5ea645-df36-5c29-8bc0-473b151320a5
                           ea47efd1-c30d-55fd-8e50-1d8d7a00e0cc
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

```shell
$ efibootmgr -v
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0008,0000,0001
Boot0000  P0: WDC WD10EZEX-75M2NA0      	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
Boot0001  P4: PLDS DVD+/-RW DH-16AES    	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
Boot0002* Fedora	HD(1,GPT,781f86bc-9c91-4b7e-83d4-34e43aa35e87,0x800,0x2800000)/File(\EFI\fedora\shim.efi)
Boot0008* UEFI OS	HD(1,GPT,781f86bc-9c91-4b7e-83d4-34e43aa35e87,0x800,0x2800000)/File(\EFI\BOOT\BOOTX64.EFI)

**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
<no output>
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
/boot
├── config-5.3.12-300.fc31.x86_64
├── config-5.3.13-300.fc31.x86_64
├── config-5.3.8-300.fc31.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── f3603cfae66044dda9d024b8fa5aec6e
│   └── 4.11.8-300.fc26.x86_64
├── grub2 [error opening dir]
├── initramfs-0-rescue-2dde7c61ae444335b5ff6030b0eb959a.img
├── initramfs-5.3.12-300.fc31.x86_64.img
├── initramfs-5.3.13-300.fc31.x86_64.img
├── initramfs-5.3.8-300.fc31.x86_64.img
├── loader
│   └── entries [error opening dir]
├── memtest86+-5.01
├── System.map-5.3.12-300.fc31.x86_64
├── System.map-5.3.13-300.fc31.x86_64
├── System.map-5.3.8-300.fc31.x86_64
├── vmlinuz-0-rescue-2dde7c61ae444335b5ff6030b0eb959a
├── vmlinuz-5.3.12-300.fc31.x86_64
├── vmlinuz-5.3.13-300.fc31.x86_64
└── vmlinuz-5.3.8-300.fc31.x86_64

**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): sudo dnf update
- Have you tried rebooting? Multiple times
- Are you using an NVMe disk? Not to my knowledge?
- Is secure boot enabled (only for the UEFI plugin)? UEFI not in use AFAIK

Thanks. I did look at one of the other cases with this error, but was unable to make any headway in terms of what the resolution was.

superm1, 2019-12-04:
Check if tpm2 abrmd is installed. If so remove it

joh-ku, 2019-12-10:
I currently get the same error message, also on Fedora 31.

(system language: german)
```
Siehe nachfolgend die detaillierten Fehlermeldungen von der Paketverwaltung:

Fehler beim Aufruf von StartServiceByName für org.freedesktop.fwupd: Zeitüberschreitung wurde erreicht
```
```
$ sudo systemctl status fwupd
 fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Tue 2019-12-10 10:17:06 CET; 8min ago
     Docs: https://fwupd.org/
 Main PID: 11917 (fwupd)
    Tasks: 5 (limit: 19098)
   Memory: 5.0M
      CPU: 143ms
   CGroup: /system.slice/fwupd.service
           └─11917 /usr/libexec/fwupd/fwupd

Dez 10 10:16:41 fedora systemd[1]: Starting Firmware update daemon...
Dez 10 10:17:06 fedora fwupd[11917]: 09:17:06:0437 FIXME                failed to allocate dbus proxy object: Fehler beim Aufruf von StartServiceByName für com.intel.tss2.Tabrmd: Zeitüberschreitung wurde erreicht
Dez 10 10:17:06 fedora fwupd[11917]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Dez 10 10:17:06 fedora fwupd[11917]: 09:17:06:0448 FuDevice             failed to set quirk key Flag=needs-unlock-legacy813: quirk key not supported
Dez 10 10:17:06 fedora fwupd[11917]: 09:17:06:0577 FuEngine             failed to add USB device 2109:0813: failed to read SPI chip ID: failed to read chip ID: endpoint stalled or request not supported
Dez 10 10:17:06 fedora systemd[1]: Started Firmware update daemon.
```
```
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5
```
```
$ fwupdmgr get-devices
System Product Name
│
├─CT500BX100SSD1:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Zusammenfassung:     ATA Drive
│     Current version:     MU02
│     Vendor:              Intel Corporation
│     GUIDs:               cd4e91be-75b1-5c2a-bf52-dc06bcc0c1e9
│                          f1fad2d9-2e86-596e-b178-11beb0673f84
│                          7e548f41-4d9b-5772-befc-c649124a1cca
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─Samsung SSD 970 EVO 500GB:
│     Device ID:           0fba4a8bedba53076fc9b147655e59455bc25a87
│     Zusammenfassung:     NVM Express Solid State Drive
│     Current version:     1B2QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          5fc1d3d7-f6cf-5f1c-ba7e-18a56f00f262
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│
├─System Firmware:
│     Device ID:           d4e14cbbbbbcb8d9488555c8d96effd32a84defb
│     Current version:     0.0.14338
│     Minimum Version:     0.0.14338
│     Vendor:              System manufacturer
│     GUID:                79da5ebf-fea3-5a75-98a1-f037f03ef79d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│
└─WDC WD40EZRZ-00WN9B0:
      Device ID:           612faa85309fec62ddfdce9f4635cdfef838d74e
      Zusammenfassung:     ATA Drive
      Current version:     80.00A80
      Vendor:              Intel Corporation
      GUIDs:               3c5871e6-eff9-5eb6-a11a-d9a006ca25c5
                           0cb5394c-f06b-5c68-b3cf-205a91644656
                           ca79edc2-1991-5aa0-bd0e-5d7233ae3521
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```
Uninstalling `tpm2-abrmd` indeed resolves the timeout issue (still a few other error messages in my case):

```
wupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Tue 2019-12-10 10:39:25 CET; 526ms ago
     Docs: https://fwupd.org/
 Main PID: 8423 (fwupd)
    Tasks: 6 (limit: 19098)
   Memory: 11.3M
      CPU: 185ms
   CGroup: /system.slice/fwupd.service
           └─8423 /usr/libexec/fwupd/fwupd

Dez 10 10:39:20 fedora systemd[1]: Starting Firmware update daemon...
Dez 10 10:39:20 fedora fwupd[8423]: ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a)
Dez 10 10:39:25 fedora fwupd[8423]: 09:39:25:0499 FuDevice             failed to set quirk key Flag=needs-unlock-legacy813: quirk key not supported
Dez 10 10:39:25 fedora fwupd[8423]: 09:39:25:0630 FuEngine             failed to add USB device 2109:0813: failed to read SPI chip ID: failed to read chip ID: endpoint stalled or request not supported
Dez 10 10:39:25 fedora systemd[1]: Started Firmware update daemon.
```

joh-ku, 2019-12-10:
Hi @superm1,
thank you for the quick reply. I already removed `tpm2-abrmd` with success (see the end of my first post - maybe too much noise for one single comment).

superm1, 2019-12-10:
> Hi @superm1,
> thank you for the quick reply. I already removed `tpm2-abrmd` with success (see the end of my first post - maybe too much noise for one single comment).

Yeah I missed the end. Okay then it's a duplicate of #1549 and #1614 then.

superm1, 2019-12-10:
@hughsie did you see those last few messages? Is this a via problem?

```
Dez 10 10:39:25 fedora fwupd[8423]: 09:39:25:0499 FuDevice             failed to set quirk key Flag=needs-unlock-legacy813: quirk key not supported
Dez 10 10:39:25 fedora fwupd[8423]: 09:39:25:0630 FuEngine             failed to add USB device 2109:0813: failed to read SPI chip ID: failed to read chip ID: endpoint stalled or request not supported
```

closed 2019-12-10 by superm1

hughsie, 2019-12-10:
> Is this a via problem

No, it's my fat fingers: https://github.com/fwupd/fwupd/commit/038d8f463e32220303b05614e1a620f17b461e9c#

Thanks for the heads up!

scareduck, 2019-12-10:
That appears to have done the trick. Thanks!