fwupd/fwupd #1635: fwupd.shutdown failed with exit status 2

dkadioglu, 2019-12-04:
When shutting down the system, I can see an error message stating that /usr/lib/systemd/system-shutdown/fwupd.shutdown failed with exit status 2. If I run fwupd.shutdown manually I get the following output:
```No firmware to activate```
This makes sense, as there is no pending firmware update.
If I run ```echo $?``` afterwards to get the last exit code, I get 2. I think that the case, when there is no pending firmware update should not be an error case. Is there a reason that this case is handled as an error? Would it be possible to change the exit code in this case?

```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

```shell
$ fwupdmgr get-devices
20HGS00V00
│
├─Thunderbolt Controller:
│     Device ID:           9cf44cb16f407badb950570b6c96d46f00f310b8
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               96c3f95f-611a-56f8-be95-0f912cb84657 ← TBT-01091609
│                          e2b1fec6-8e86-51f6-89a6-afd49e79c64b ← TBT-01091609-0000:03:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Lenovo ThinkPad Dock:
│     Device ID:           73ef80b60058b4f18549921520bfd94eaf18710a
│     Zusammenfassung:     USB 3.x Hub
│     Current version:     50.41
│     Vendor:              LENOVO (USB:0x17EF)
│     Install Duration:    10 seconds
│     GUIDs:               3a5bdad1-2726-5194-aa7d-04e1daf9946b ← USB\VID_17EF&PID_1010&REV_5041
│                          8d708000-7d3b-5e85-ac9d-8ef32fb2d445 ← USB\VID_17EF&PID_1010
│                          6201fecc-1641-51f6-a6d2-38a06d5476bf ← VLI_USBHUB\SPI_C220
│                          b435be36-7b59-5463-ab34-abfe22cf7a53 ← USB\VID_17EF&PID_1010&SPI_C220&REV_5041
│                          06cdabc6-2059-5daa-87ab-2f6fc31fd74b ← USB\VID_17EF&PID_1010&SPI_C220
│     Device Flags:        • Updatable
│                          • Cryptographic hash verification is available
│   
├─SAMSUNG MZVLW1T0HMLH-000L7:
│     Device ID:           e11623b2caa18fee292058a5c09ca4e6152f7ecf
│     Zusammenfassung:     NVM Express Solid State Drive
│     Current version:     5L7QCXY7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S35ANX0J100208
│     GUIDs:               f87b9ac8-1cb3-5c0a-ae57-7144f211fe5e ← NVME\VEN_144D&DEV_A804&REV_00
│                          5b3df2da-f745-5fd0-81de-5dafd7f0bf8c ← NVME\VEN_144D&DEV_A804
│                          1f807912-a8fa-548b-af96-117b372f6a42 ← SAMSUNG MZVLW1T0HMLH-000L7
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           84149fb78009cf27cc1dd5520911f46f8792dbfe
│     Current version:     0.1.35
│     Minimum Version:     0.1.7
│     Vendor:              LENOVO
│     GUID:                7a176688-0960-47ba-931b-7829849e8347
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
[... 449 lines cut ...]
│       ├── wrmsr.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

12 directories, 362 files
```

Please answer the following questions:

- Operating system and version: Arch Linux (daily updated)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2019-12-04:
I guess this makes sense. It should just be the case to replace the lump of code in `src/fu-tool.c` of:

    if (!has_pending) {
        g_set_error_literal (error,
                     FWUPD_ERROR,
                     FWUPD_ERROR_NOTHING_TO_DO,
                     "No firmware to activate");
        return FALSE;
    }

with:

    if (!has_pending) {
        g_printerr ("No firmware to activate\n");
        return TRUE;
    }

Could you open a PR with that change please?

superm1, 2019-12-04:
The other option is in the fwupd shutdown script to look at the return code of the activate call. If it's 0 or 2 return 0.

dkadioglu, 2019-12-04:
@hughsie done
@superm1 I made a pull request following @hughsie idea, as I still think, that the case "no firmware to be updated" should not be an error.

superm1, 2019-12-04:
@dkadioglu OK.
I think you forgot to hit the open PR button after you pushed to your personal branch.

dkadioglu, 2019-12-04:
@superm1 Indeed, thank you very much!

superm1, 2019-12-04:
Closed in #1639, thanks!!!

closed 2019-12-04 by superm1