fwupd/fwupd #1648: installed-tests failing in Debian (sometimes)

superm1, 2019-12-10:
This isn't happening everytime, but occasionally there are failures on DebCI in Debian and Autopkgtest in Ubuntu with the fwupd 1.3.x stuff.  The related debugging commits were loaded into Debian's latest package so we now have a full verbose daemon log of the failure.  Copying it here since inevitably it's going to run again and pass.

The log came from here though: https://ci.debian.net/data/autopkgtest/unstable/amd64/f/fwupd/3622576/log.gz
```
Running test: fwupd/fwupdmgr.test
Getting the list of remotes...
HVM domU
│
├─Linux Vendor Firmware Service (testing):
│     Remote ID:           lvfs-testing
│     Type:                download
│     Keyring:             gpg
│     Enabled:             false
│     Priority:            1
│     Filename:            /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz
│     Filename Signature:  /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz.asc
│     Metadata URI:        https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
│     Metadata Signature:  https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
│     Report URI:          https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting: false
│   
├─Enable UEFI capsule updates on Dell systems:
│     Remote ID:           dell-esrt
│     Type:                local
│     Keyring:             none
│     Enabled:             true
│     Filename:            /usr/share/fwupd/remotes.d/dell-esrt/metadata.xml
│   
├─fwupd test suite:
│     Remote ID:           fwupd-tests
│     Type:                local
│     Keyring:             none
│     Enabled:             false
│     Filename:            /usr/share/installed-tests/fwupd/fwupd-tests.xml
│   
├─Linux Vendor Firmware Service:
│     Remote ID:           lvfs
│     Type:                download
│     Keyring:             gpg
│     Enabled:             true
│     Filename:            /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz
│     Filename Signature:  /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz.asc
│     Metadata URI:        https://cdn.fwupd.org/downloads/firmware.xml.gz
│     Metadata Signature:  https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
│     Report URI:          https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting: false
│   
├─Vendor:
│     Remote ID:           vendor
│     Type:                local
│     Keyring:             none
│     Enabled:             false
│     Filename:            /usr/share/fwupd/remotes.d/vendor/vendor.xml.gz
│   
└─Vendor (Automatic):
      Remote ID:           vendor-directory
      Type:                directory
      Keyring:             none
      Enabled:             false
      Filename:            /usr/share/fwupd/remotes.d/vendor/firmware
    
Enabling fwupd-tests remote...
Successfully enabled remote
Update the device hash database...
Successfully updated device checksums
Getting devices (should be one)...
HVM domU
│
└─Integrated Webcam™:
      Device ID:           08d460be0f1f9f128413f816022a6439e0078018
[... 310 lines cut ...]
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.55ms      get size nodetab
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.03ms      adding strtab element
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.02ms      adding strtab attr name
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.03ms      adding strtab attr value
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.04ms      adding strtab text
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0489 XbSilo               0.03ms      writing nodetab
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0490 XbSilo               0.02ms      fixing ->parent and ->next
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0490 XbSilo               0.01ms      appending strtab
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0490 XbSilo               0.01ms      parse blob
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0492 XbSilo               2.62ms      save file
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0492 XbSilo               0.02ms      parse blob
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0492 XbSilo               0.10ms      loaded file
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0493 XbSilo               0.01ms      query on / with `components/component` limit=0 -> 4 results
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0493 FuEngine             4 components now in silo
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0493 XbSilo               0.01ms      query on / with `components/component/provides/firmware` limit=0 -> 4 results
Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0493 XbSilo               0.05ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='b585990a-003e-5270-89d5-3705a17f9a43']/../..` limit=1 -> 1 results
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
SUMMARY: total=1; passed=0; skipped=0; failed=1; user=0.1s; system=0.0s; maxrss=12052
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
autopkgtest [01:56:48]: test ci: -----------------------]
autopkgtest [01:56:48]: test ci:  - - - - - - - - - - results - - - - - - - - - -
ci                   FAIL non-zero exit status 2
autopkgtest [01:56:48]: @@@@@@@@@@@@@@@@@@@@ summary
ci                   FAIL non-zero exit status 2
```

hughsie, 2019-12-10:
> Dec 10 01:56:48 autopkgtest-unstable-amd64 fwupd[4983]: 01:56:48:0493 XbSilo               0.05ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='b585990a-003e-5270-89d5-3705a17f9a43']/../..` limit=1 -> 1 results

Does the daemon actually segfault? can we get a backtrace if so?

superm1, 2019-12-10:
No I don't believe backend segfaults. We should have seen that in the backend logs if so.  The client process ends with error code 2, so it's nothing to do:
```
Getting updates (should be one)...
No updatable devices
```

superm1, 2019-12-10:
It seems to me like unfortunate timing maybe with the client moving quicker than the backend does.

What we're seeing in the logs is:

- GetRemotes
- ModifyRemote
- UpdateHash
- GetDevices
- GetDevices
- Daemon detects conffile change and reloads everything

I don't think that we should be seeing those GetDevices calls until after the remote is fully enabled.
We probably need to make it atomic that the `ModifyRemote` command immediately reloads conffiles rather than waiting for the inotify.

hughsie, 2019-12-10:
> We probably need to make it atomic that the ModifyRemote command immediately reloads conffiles rather than waiting for the inotify

Yes, that. On windows the inotify "poll" is 30 seconds, and so we don't want to rely on the event if we can trigger it manually at the right time to avoid a race.

hughsie, 2019-12-10:
See https://github.com/fwupd/fwupd/pull/1650 too

superm1, 2019-12-10:
closed by #1650

superm1, 2020-01-08:
Re-opening.
Unfortunately there still seems to be some failures happening in an unpatched 1.3.6.

```
Running test: fwupd/fwupdmgr.test
Getting the list of remotes...
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
OpenStack Nova
│
├─Linux Vendor Firmware Service (testing):
│     Remote ID:           lvfs-testing
│     Type:                download
│     Keyring:             gpg
│     Enabled:             false
│     Priority:            1
│     Filename:            /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz
│     Filename Signature:  /var/lib/fwupd/remotes.d/lvfs-testing/metadata.xml.gz.asc
│     Metadata URI:        https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
│     Metadata Signature:  https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
│     Report URI:          https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting: false
│   
├─Enable UEFI capsule updates on Dell systems:
│     Remote ID:           dell-esrt
│     Type:                local
│     Keyring:             none
│     Enabled:             true
│     Filename:            /usr/share/fwupd/remotes.d/dell-esrt/metadata.xml
│   
├─fwupd test suite:
│     Remote ID:           fwupd-tests
│     Type:                local
│     Keyring:             none
│     Enabled:             false
│     Filename:            /usr/share/installed-tests/fwupd/fwupd-tests.xml
│   
├─Linux Vendor Firmware Service:
│     Remote ID:           lvfs
│     Type:                download
│     Keyring:             gpg
│     Enabled:             true
│     Filename:            /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz
│     Filename Signature:  /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz.asc
│     Metadata URI:        https://cdn.fwupd.org/downloads/firmware.xml.gz
│     Metadata Signature:  https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
│     Report URI:          https://fwupd.org/lvfs/firmware/report
│     Automatic Reporting: false
│   
├─Vendor:
│     Remote ID:           vendor
│     Type:                local
│     Keyring:             none
│     Enabled:             false
│     Filename:            /usr/share/fwupd/remotes.d/vendor/vendor.xml.gz
│   
└─Vendor (Automatic):
      Remote ID:           vendor-directory
      Type:                directory
      Keyring:             none
      Enabled:             false
      Filename:            /usr/share/fwupd/remotes.d/vendor/firmware
    
Enabling fwupd-tests remote...
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Successfully enabled remote
Update the device hash database...
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
Successfully updated device checksums
Getting devices (should be one)...
WARNING: The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
[... 23 lines cut ...]
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0725 FuMain               Called GetRemotes()
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0742 FuMain               Called GetRemotes()
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0744 FuMain               Called ModifyRemote(fwupd-tests,Enabled=true)
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0744 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0746 FuMain               Emitting PropertyChanged('Status'='idle')
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0764 FuMain               Called GetDevices()
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0766 FuMain               Called VerifyUpdate(08d460be0f1f9f128413f816022a6439e0078018)
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0766 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuMain               Emitting PropertyChanged('Status'='idle')
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuPlugin             running superclassed update_detach() on test
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuPlugin             performing verify() on test
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuPlugin             running superclassed update_attach() on test
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuPlugin             already in runtime mode, skipping
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0768 FuCommon             creating path /var/lib/fwupd/verify
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0786 FuMain               Called GetDevices()
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0804 FuMain               Called GetDevices()
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0805 FuMain               Called Verify(08d460be0f1f9f128413f816022a6439e0078018)
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0806 XbSilo               compiling /var/lib/fwupd/verify/08d460be0f1f9f128413f816022a6439e0078018.xml:ctime=1578478257.765804…
Jan 08 10:10:57 autopkgtest fwupd[1422]: 10:10:57:0823 FuMain               Called GetDevices()
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
SUMMARY: total=1; passed=0; skipped=0; failed=1; user=0.1s; system=0.0s; maxrss=12184
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
```

Full log: https://objectstorage.prodstack4-5.canonical.com/v1/AUTH_77e2ada1e7a84929a74ba3b87153c0ac/autopkgtest-focal/focal/amd64/f/fwupd/20200108_101108_20e49@/log.gz

superm1, 2020-01-08:
Another  time it got this result from the daemon logs:
```
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0946 FuMain               FuMain: acquired name: org.freedesktop.fwupd
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0952 FuMain               Called GetRemotes()
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0971 FuMain               Called GetRemotes()
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0972 FuMain               Called ModifyRemote(fwupd-tests,Enabled=true)
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0972 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0974 FuMain               Emitting PropertyChanged('Status'='idle')
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0993 FuMain               Called GetDevices()
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0995 FuMain               Called VerifyUpdate(08d460be0f1f9f128413f816022a6439e0078018)
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0995 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuMain               Emitting PropertyChanged('Status'='idle')
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuPlugin             running superclassed update_detach() on test
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuPlugin             performing verify() on test
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuPlugin             running superclassed update_attach() on test
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuPlugin             already in runtime mode, skipping
Jan 08 14:06:49 autopkgtest fwupd[1423]: 14:06:49:0997 FuCommon             creating path /var/lib/fwupd/verify
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0016 FuMain               Called GetDevices()
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0035 FuMain               Called GetDevices()
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0036 FuMain               Called Verify(08d460be0f1f9f128413f816022a6439e0078018)
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0036 XbSilo               compiling /var/lib/fwupd/verify/08d460be0f1f9f128413f816022a6439e0078018.xml:ctime=1578492409.993861…
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0054 FuMain               Called GetDevices()
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0060 FuRemoteList         /etc/fwupd/remotes.d/fwupd-tests.conf.FV19D0 changed, reloading all remotes
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0060 Fwupd                looking for os-release at /etc/os-release
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0060 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0060 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/fwupd-tests.conf
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         ordering lvfs-testing=lvfs+1
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         ignoring unfound remote fwupd
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuRemoteList         ::remote_list changed
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuEngine             remote lvfs-testing not enabled, so skipping
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuEngine             no /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuEngine             remote vendor not enabled, so skipping
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 FuEngine             remote vendor-directory not enabled, so skipping
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               0.01ms      parse blob
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               0.04ms      loaded file
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               file: f04b0cb1-5986-194d-dabb-f55a68f4cae3, current:ba158754-5063-876a-ada5-9eec9e2f13ec, cached: (null)
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               compiling /usr/share/fwupd/remotes.d/dell-esrt/metadata.xml:ctime=1578492380.408000:func-id=AppStreamUpgrade@3:value=/usr/share/fwupd/remotes.d/dell-esrt/metadata.xml:value=dell-esrt…
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0061 XbSilo               0.06ms      compile /usr/share/fwupd/remotes.d/dell-esrt/metadata.xml:ctime=1578492380.408000:func-id=AppStreamUpgrade@3:value=/usr/share/fwupd/remotes.d/dell-esrt/metadata.xml:value=dell-esrt
[... cut ...]
Jan 08 14:06:50 autopkgtest fwupd[1423]: 14:06:50:0065 XbSilo               0.03ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='b585990a-003e-5270-89d5-3705a17f9a43']/../..` limit=1 -> 1 results
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
SUMMARY: total=1; passed=0; skipped=0; failed=1; user=0.1s; system=0.0s; maxrss=12036
FAIL: fwupd/fwupdmgr.test (Child process exited with code 2)
```

So I guess it seems like the reloading of the remote is still racing?

PR #1695 (2020-01-08): fu-remote-list: emit a changed signal when modifying a remote
This should notify the daemon of changes before the inotify letting
the data store get reloaded and fix autopkgtest race conditions.
Fixes: #1648

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 427e01fc (2020-01-08): fu-remote-list: emit a changed signal when modifying a remote

This should notify the daemon of changes before the inotify letting
the data store get reloaded and fix autopkgtest race conditions.
Fixes: #1648
closed 2020-01-08 by superm1