fwupd/fwupd #1653: WD19 is missing some vendor IDs

superm1, 2019-12-11:
As mentioned in https://github.com/fwupd/lvfs-website/pull/449 I think that the WD19 vendor ID's are not complete for what you are expecting.
Here's a snapshot from a production dock:

```
├─WD19TB:
│ │   Device ID:           a455a3a21752e5d31c228f1f5b51761dad88a432
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.02
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc
│ │   Install Duration:    1 minute
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │ 
│ ├─Thunderbolt controller in Dell dock:
│ │     Device ID:         8357922f7f208e1f3b7537141ce17db04bc219d9
│ │     Summary:           Thunderbolt controller
│ │     Current version:   43.00
│ │     Vendor:            Dell Inc (TBT:0x00D4)
│ │     Install Duration:  22 seconds
│ │     GUID:              c94770ca-1773-592c-b20a-e87243bc7cd0
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │   
│ ├─Package level of Dell dock:
│ │     Device ID:         1eca9eabb0c992c136e1deb1f89e3f70c465aa1c
│ │     Summary:           A representation of dock update status
│ │     Current version:   01.00.08.01
│ │     Vendor:            Dell Inc
│ │     Install Duration:  5 seconds
│ │     GUID:              8ceeeffd-51b6-580c-9b75-69143227aff8
│ │     Device Flags:      • Updatable
│ │                        • Supported on remote server
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:         ca36279eeac13463fd5d974c88794859294fbfb1
│ │     Summary:           USB 3.1 Generation 1 Hub
│ │     Current version:   01.21
│ │     Vendor:            Dell Inc (USB:0x413C)
│ │     Install Duration:  14 seconds
│ │     GUIDs:             86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4
│ │                        b27d25f1-019d-5718-b41a-02ddaefe5577
│ │                        ac5b774c-b49d-566b-9255-85f0f7f8a4ed
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:         acb404019656654d44f80922d94735e831d9bb40
│ │     Summary:           USB 3.1 Generation 2 Hub
│ │     Current version:   01.47
│ │     Vendor:            Dell Inc (USB:0x413C)
│ │     Install Duration:  3 seconds
│ │     GUIDs:             707c63d2-e597-5c40-84db-9b1bb4c48d96
│ │                        acfcd89b-105d-55b9-b85b-08bf8508f38c
│ │                        568ffa1e-a0db-5287-9ea3-872b60f7730b
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ └─VMM5331 in Dell dock:
│       Device ID:         58931e8c9ef5eead9c007563814f96fcadb1b993
│       Summary:           Multi Stream Transport controller
│       Current version:   05.04.00
│       Vendor:            Dell Inc
│       Install Duration:  6 minutes
│       GUID:              89fec0b6-6b76-5008-b82c-5e5c6c164007
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│                          • Device is usable for the duration of the update

```

Vendor strings are set for everything (IIRC from the quirk file),  but vendor ID is not.

hughsie, 2019-12-12:
>  but vendor ID is not

We can include quirks for the Vendor ID too, right?

hughsie, 2019-12-12:
As an aside, I currently have no Dell dock hardware 3 months after I requested it. If you could escalate this with your RH counterpart I'd be grateful.

superm1, 2019-12-12:
> We can include quirks for the Vendor ID too, right?

Yes, I'm not sure what to use for some of them though.

hughsie, 2019-12-12:
> Yes, I'm not sure what to use for some of them though.

From a 30,000ft view, a hub is just a USB device, no? I.e. the `USB:0x{dell}` id.

superm1, 2019-12-12:
Yeah I guess from a 30k ft view that makes sense.

commit e2db5684 (2019-12-12): trivial: Set vendor ID for Dell WD19 to USB:0x413C (Fixes: #1653)

View after this update:
```
├─WD19TB:
│ │   Device ID:           a455a3a21752e5d31c228f1f5b51761dad88a432
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.02
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       J4T6SV2/3169046018716226
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                    [... cut]
closed 2019-12-12 by superm1