fwupd/fwupd #1654: 'Stop job' for fwupd on most shutdowns and reboots

LinuxOnTheDesktop, 2019-12-11:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.5
```

```shell
$ fwupdmgr get-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           eded407096031a81370bc25ae01650697a7c72b6
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          09bc8035-c26c-56c8-af2d-5b0fd7e1efe1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.8.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               963522d7-bc73-51fa-a70c-cf615f42bcf6
│                          b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.43
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.18
│     Minimum Version:     0.1.18
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.10_B0036
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

Please answer the following questions:

- Operating system and version: Mint 19.2 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

Screenshot: 
[image]

**The fwupd systemd journal** contains only the following, perhaps because the journal is getting cleared at each boot.

```
-- Logs begin at Wed 2019-12-11 20:14:25 GMT, end at Wed 2019-12-11 20:21:14 GMT. --
Dec 11 20:14:27 x1 systemd[1]: Started Activate fwupd updates.
```

superm1, 2019-12-11:
Can you modify that job and add `--verbose` to the command?

LinuxOnTheDesktop, 2019-12-11:
I am afraid I do not understand. What command should I add `--verbose` to? If it is relevant: I got the journal output from using a GUI program called, I think (I am not on my Mint computer at the comment) Systemd Manager.

superm1, 2019-12-11:
IIRC it's `/etc/systemd/system/fwupd-activate.service`.
There should be a command in there that is calling `fwupdtool activate`.

Afterwards run `systemctl daemon-reload`.  Then on your next shutdown you should get a more verbose log in the journal for `fwupd-activate` that might indicate why it's taking a while.

LinuxOnTheDesktop, 2019-12-11:
I'm sorry, but I still do not understand. Are you saying I need to run

`/etc/systemd/system/fwupd-activate.service --verbose`

(or do that with the necessary adjustment if the path to `fwupd-activate.service` is different)? That seems strange. Perhaps instead you mean that I should do the following.

`sudo systemctl restart fwupd-activate --verbose`

Please clarify.

superm1, 2019-12-11:
Open up `/etc/systemd/system/fwupd-activate.service` in a text editor and modify the command that is in there that calls `fwupdtool activate` to make it call `fwupdtool activate --verbose`

superm1, 2019-12-11:
`systemctl daemon-reload` reloads the unit files for systemd after you made that change (so it's picked up)

LinuxOnTheDesktop, 2019-12-13:
I modified `etc/systemd/system/fwupd-activate.service` as advised. Some time later - after multiple reboots - the shutdown problem did reocur. Yet, `journalctl -u fwupd-activate.service` still shows only this:

-- Logs begin at Fri 2019-12-13 18:02:48 GMT, end at Fri 2019-12-13 18:14:30 GMT. --
Dec 13 18:02:50 x1 systemd[1]: Started Activate fwupd updates.

The problem with the logging seems to be: the log is showing only events from after the restart, i.e. from after the time the problem occured. Please advise.

superm1, 2019-12-13:
I don't really know, I think you should just mask the service. It's only used to ATA updates which aren't actually on LVFS at all. it

superm1, 2019-12-13:
Aren't on LVFS for your notebook I meant to say

LinuxOnTheDesktop, 2019-12-13:
Can't we somehow get logs from before the last boot?

Otherwise, I'll mask the service of give it a shorter timeout.

LinuxOnTheDesktop, 2020-01-12:
@stalebot: bump.

LinuxOnTheDesktop, 2020-02-07:
> I think you should just mask the service. It's only used to ATA updates which aren't actually on LVFS at all.

I should like elaboration. I do not understand the second sentence. Is fwup installing a service that . . it does not need?

superm1, 2020-02-07:
Correct, that service is only needed for hardware that needs activation on shutdown which is only ATA devices.

LinuxOnTheDesktop, 2020-02-07:
Thanks. I continue to require a bit of clarification. Here is why. Of my two laptops, one (the one for which I gave details above) has an NVME SSD and one has a SATA SSD. Would it be alright to mask the service not only on the NVME laptop but also on the SATA laptop? I think the following is the case (please tell whether I am right):

1.  my _NVME laptop_ has no ATA device and so I can disable the service on that machine;
2. my _SATA laptop_ has an ATA device (the SATA drive!) but there are no updates for that device on LVFS and so I can disable the service on that machine too.

Note also that while I am grateful for any workaround it _is_ a workaround.

superm1, 2020-02-07:
Sounds right to me.

LinuxOnTheDesktop, 2020-02-07:
It seems that masking the service is problematic:

```
sudo systemctl mask fwupd-activate.service --now
Failed to mask unit: File /etc/systemd/system/fwupd-activate.service already exists.
```

See also [this webpage](https://unix.stackexchange.com/questions/351692/how-can-we-mask-service-whose-unit-file-is-located-under-etc-systemd-system). So I've just disabled it. But that means after every update I will to disable it again, I think.

LinuxOnTheDesktop, 2020-03-08:
The problem persists.

superm1, 2020-03-09:
I don't know anything that can be done from fwupd's snap to help remove this this delay on shutdown.

That activate service is needed on some systems but not all.  If someone has a solution, happy to review a PR, but otherwise this will sit idle.

estan, 2020-03-14:
When I looked at my shutdown messages (see https://github.com/fwupd/fwupd/files/4332576/shutdown-log.txt), it seems the snap squashfs mount units are stopped before the `fwupd-activate.service`.

I wonder if the `RequiresMountsFor=/snap/fwupd/current` in `fwupd-activate.service` is really working as intended? Will that work if the path is a symlink like `current` is?

The systemd docs says

> RequiresMountsFor=
> 
>    Takes a space-separated list of absolute paths. Automatically adds dependencies of type Requires= and After= for all mount units required to access the specified path.
>
>    Mount points marked with noauto are not mounted automatically through local-fs.target, but are still honored for the purposes of this option, i.e. they will be pulled in by this unit.

which could be interpreted as saying it should work even if a path is a symlink to the actual mount point... but I don't know.

If it had worked, then I would have expected the stopping of `fwupd-activate.service` to run before the stopping of the mount units has finished during shutdown, because `RequiresMountsFor` is supposed to add a `Requires` and `After` dependency on the mount units, and so during shutdown, the dependee should be stopped before the dependencies if I understand correctly.

Maybe I'm misinterpreting the log though. Feel free to have a look at it @superm1.

For reference, on my system I have two mounts for fwupd:

```shell
estan@edison:~$ mount | grep fwupd
/var/lib/snapd/snaps/fwupd_1704.snap on /snap/fwupd/1704 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/fwupd_1625.snap on /snap/fwupd/1625 type squashfs (ro,nodev,relatime,x-gdu.hide)
estan@edison:~$
```

And here's some selected messages from the shutdown log:

```shell
estan@edison:~$ egrep "(1704|1625).*stop finished|fwupd-activate" /shutdown-log.txt
[  212.777431] systemd[1]: snap-fwupd-1625.mount: Job snap-fwupd-1625.mount/stop finished, result=done
[  212.833228] systemd[1]: snap-fwupd-1704.mount: Job snap-fwupd-1704.mount/stop finished, result=done
[  302.531276] systemd[1]: fwupd-activate.service: Stopping timed out. Terminating.
[  302.531513] systemd[1]: fwupd-activate.service: Changed stop -> stop-sigterm
[  302.558957] systemd[1]: fwupd-activate.service: Child 2669 belongs to fwupd-activate.service.
[  302.558967] systemd[1]: fwupd-activate.service: Control process exited, code=killed status=15
[  302.558970] systemd[1]: fwupd-activate.service: Got final SIGCHLD for state stop-sigterm.
[  302.559031] systemd[1]: fwupd-activate.service: Failed with result 'timeout'.
[  302.559037] systemd[1]: fwupd-activate.service: Changed stop-sigterm -> failed
[  302.559593] systemd[1]: fwupd-activate.service: Job fwupd-activate.service/stop finished, result=done
[  302.559667] systemd[1]: fwupd-activate.service: Unit entered failed state.
[  302.643341] systemd[1]: fwupd-activate.service: Failed to send unit remove signal for fwupd-activate.service: Transport endpoint is not connected
estan@edison:~$
```

Notice how the `snap-fwupd-1625.mount` and `snap-fwupd-1704.mount` units are stopped, before the stopping of `fwupd-activate.service` times out.

There are quite a bit of more messages in the log though, so would be good if someone could have a look at it.

estan, 2020-03-14:
Looking at the `After=` of the service:

```shell
estan@edison:~$ sudo systemctl show fwupd-activate.service | grep After=
After=systemd-journald.socket sysinit.target basic.target system.slice -.mount
estan@edison:~$
```

I'm not sure if the `-.mount` is the expected output whenever `RequiresMountsFor` is used, or if I should expect it to say `snap-fwupd-1625.mount snap-fwupd-1704.mount` there (i.e. the exact mount units it has identified and added as dependencies).

estan, 2020-03-14:
@superm1 I found this systemd bug about `RequiresMountsFor` not working with symlinks to directories: https://github.com/systemd/systemd/issues/8907. So it seems the `RequiresMountsFor=/snap/fwupd/current` in `fwupd-activate.service` (`current` being a symlink) is not going to work, and it seems Poettering is not to keen on fixing it since it would be complex.

I'm not sure how to fix this..

estan, 2020-03-14:
@superm1 Hm, since the mount units generated by `snapd` all seem to have `Before=snapd.service`, so the mounts are set up before the snapd daemon starts, could not `fwupd-activate.service` simply have `After=snapd.service`?

superm1, 2020-03-15:
Reading through that bug I think I finally understand why it was working in all my bench tests. The symlink is resolved when it's installed but that's it. So as soon as it changes it breaks again.

I think that's worth a shot to try switching like you proposed. Let me know if it helps :)

estan, 2020-03-15:
> I think that's worth a shot to try switching like you proposed. Let me know if it helps :)

It seems to work fine, the shutdown delay is gone. I made a pull request: https://github.com/fwupd/fwupd/pull/1865

However, related to this, my colleague @kalvdans made a good point: It's probably not safe to run the activation with file systems on the ATA devices still mounted read-write. The activation during shutdown should probably only occur once everything is mounted read-only, since no destructive operations should run on the ATA device between the activation and reboot.

This can be done by putting a script in `/usr/lib/systemd/system-shutdown` (or `/lib/systemd/system-shutdown` on Ubuntu it seems) instead. However, that will not work of course, since by that time, the snap squashfs mounts will probably have been long since unmounted. Perhaps if you temporarily re-mount the snap squashfs mount at that point?

snap is not really a good fit for a sysadmin tool like fwupd, which in addition needs to run things very late during shutdown :p

closed 2020-03-15 by superm1

superm1, 2020-03-16:
> It seems to work fine, the shutdown delay is gone. I made a pull request: #1865
>
> However, related to this, my colleague @kalvdans made a good point: It's probably not safe to run the activation with [...]

I've merged your proposal for now. However the point raised is a valid hypothetical concern for ATA depending upon the manufacturer implementation whether it's destructive or not. I don't believe we have enough evidence either way to support a conclusion.

That's actually how the non snap install works, putting a script in the directory mentioned above.

So I see a few options now:

1. Leave as is. The intersection of people using snap, using an old ATA disk, having updates on lvfs for an old ATA disk, and rebooting their machines instead of shutdown on update is super low.

2. Remove activate feature for ATA when building snap. Tell users to shut down. Let other plugins with activation and no data concern proceed as normal.

3. Create a statically linked binary for activate to put in that directory on install instead of the dynamically linked one we have now. The unfortunate thing here is that the architecture of fwupd having the activation code for plugins loaded at runtime will make this a lot more difficult.

The disks I have tested using the snap with ATA updates I personally didn't find any destructive behavior when activate was called. I'm currently leaning on option 1.

hughsie, 2020-03-16:
> I'm currently leaning on option 1.

I'm fine with that, but option 2 seems pretty easy to implement too.

estan, 2020-03-18:
Alright, sounds good to me. Understand there are technical difficulties and also think the likelihood is probably too small to even bother. Just wanted to raise the question.

PR #1865 (2020-03-15): Start fwupd-activate.service after snapd.service
Instead of using `RequiresMountsFor=/snap/fwupd/current`, which will not work since `/snap/fwupd/current` is a symlink [1].

This will work since the mount units generated by snapd all have `Before=snapd.service`, so will be stopped after `snapd.service` during shutdown.

With `After=snapd.service`, `fwupd-activate.service` will then stop before `snapd.service`, at a point when all snap mount units are still alive.

Fixes the issue where `fwupd-activate.service` hangs when stopped, causing a stop job timeout during shutdown.

[1] See https://git [... cut]
commit efc2e47e (2020-03-15): Start fwupd-activate.service after snapd.service

Instead of using RequiresMountsFor=/snap/fwupd/current, which will not
work since /snap/fwupd/current is a symlink [1].

This will work since the mount units generated by snapd all have
Before=snapd.service, so will be stopped after snapd.service during
shutdown.

With After=snapd.service, fwupd-activate.service will then stop before
snapd.service, at a point when all snap mount units are still running.

Fixes the issue where fwupd-activate.service hangs when stopped, causing
a stop job timeout during shutdown.

[1] See https://github.com/syste [... cut]
commit 56f6c06e (2020-04-15): Start fwupd-activate.service after snapd.service

Instead of using RequiresMountsFor=/snap/fwupd/current, which will not
work since /snap/fwupd/current is a symlink [1].

This will work since the mount units generated by snapd all have
Before=snapd.service, so will be stopped after snapd.service during
shutdown.

With After=snapd.service, fwupd-activate.service will then stop before
snapd.service, at a point when all snap mount units are still running.

Fixes the issue where fwupd-activate.service hangs when stopped, causing
a stop job timeout during shutdown.

[1] See https://github.com/syste [... cut]
commit 5b26d317 (2020-06-23): Start fwupd-activate.service after snapd.service

Instead of using RequiresMountsFor=/snap/fwupd/current, which will not
work since /snap/fwupd/current is a symlink [1].

This will work since the mount units generated by snapd all have
Before=snapd.service, so will be stopped after snapd.service during
shutdown.

With After=snapd.service, fwupd-activate.service will then stop before
snapd.service, at a point when all snap mount units are still running.

Fixes the issue where fwupd-activate.service hangs when stopped, causing
a stop job timeout during shutdown.

[1] See https://github.com/syste [... cut]