fwupd/fwupd #1659: Unable to update the firmware of an logitech unify reciver modell  U8000

tuxmaster5000, 2019-12-12:
```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

```shell
$ fwupdmgr get-devices
System Firmware
  DeviceId:             123fd4143619569d8ddb6ea47d1d3911eb5ef07a
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|require-ac|registered|needs-reboot
  Vendor:               
  Version:              MKQ7710H.86A.0074.2018.1025.1727
  VersionFormat:        plain
  Icon:                 computer
  Created:              2019-12-12
  UpdateError:          Firmware can not be updated in legacy mode, switch to UEFI mode.

Hitachi HDS723020BLA642
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 c5e0df58-c081-5c67-a67b-69cfc5a481f6
  Guid:                 461ac46c-5439-5b77-b1f2-a75d024ea620
  Guid:                 55243e86-0ca6-5247-8525-00e93fa160b9
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              MN6OAA10
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-12-12

SAMSUNG MZ7TD256HAFV-000L9
  DeviceId:             0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
  Guid:                 543ddeb8-1d9f-5b7e-80d3-5aafdbe24534
  Guid:                 dabb7316-b923-527b-9c4c-9855198795f4
  Guid:                 ef0f5dcf-3f29-5580-9961-dabce82231fe
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              DXT02L5Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-12-12

HDT722516DLA380
  DeviceId:             7466579651466dd8b56d7a19479bed768440026d
  Guid:                 4f9b4da4-014f-5ac3-838c-2e7dccf2cdd7
  Guid:                 25f3635d-7b27-589e-87ec-fd9bbccc7827
  Guid:                 c9bc3512-f59a-51f2-8812-6895e7fcba4a
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              V43OA91A
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-12-12

Unifying Receiver
  DeviceId:             4d102750edb193c8986ff25e9f3955ec9fc9607e
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.02_B0009
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-12-12
```

Please answer the following questions:

- Operating system and version: Fedora 30 x86_64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? no efi mode

Error:
fwupdmgr get-updates
No releases found for device: Not compatible with org.freedesktop.fwupd version 1.2.11, requires >= 1.3.1, Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

superm1, 2019-12-12:
`No releases found for device: Not compatible with org.freedesktop.fwupd version 1.2.11, requires >= 1.3.1, Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]`

As mentioned in the final error; can you please try a newer version of fwupd than 1.2.11?

tuxmaster5000, 2019-12-12:
I have open an ticket at bugzilla of Fedora for it.

superm1, 2019-12-12:
For what?

FFY00, 2019-12-12:
This just means that your device doesn't have any available firmware updates, I know the error can seem scary but that's all it means.

superm1, 2019-12-12:
Well no firmware updates that can apply to 1.2.111, it shows a requirement on fwupd 1.3.1+ though.

FFY00, 2019-12-12:
Yes but that shouldn't really matter. It's not relevant here.

The issue is that the same PID has two hardware variants, hence different firmwares. The way we figure out which firmware to install is by looking at the bootloader version. The bootloader version goes as follows: `BOT[bootloader identifier (different between projects)].[booloader version]`. Each bootloader then has two variants when it comes to firmware, unsigned firmware (used in older bootloader versions) and signed firmware (used in newer bootloader versions).

What is happening here is that fwupd sees there is an available firmware for this PID and then tries to figure out if it's compatible. Here the firmware fwupd was checking is for `BOT03.0[0-1]_*`, so unsgined receivers with texas chips but the receiver plugged is not compatible. Even though it is the texas hardware variant, it has a newer bootloader, so it needs the signed variant of the firmware.

TLDR: We have 4 variants of firmwares for unifying receiver but unifying receivers all share one USB PID, and so this regex(s) are used to figure out the if the plugged receiver is the correct variant.

FFY00, 2019-12-12:
fwupd should handle failing to match the regex a bit differently.

superm1, 2019-12-12:
@FFY00 basically a duplicate of https://github.com/fwupd/fwupd/issues/1261 then?

FFY00, 2019-12-12:
Yes, every time you get errors about the bootloader regex is very very very probably this.

commit 50903fff (2019-12-12): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit a712448a (2019-12-13): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit c361b675 (2019-12-13): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit 1681058a (2019-12-13): fu-engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
closed 2019-12-13 by superm1