fwupd/fwupd #1661: Problems detecting and updating 8Bitdo wireless usb adapter

Lahvuun, 2019-12-13:
```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.2.11

daemon version: 1.3.5
```

- Operating system and version: Gentoo Linux (rolling release) ~amd64
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `portage`
- Have you tried rebooting? Yes

---

Product page: https://www.8bitdo.com/wireless-usb-adapter/

When plugged in firmware update mode, lsusb shows:

```Bus 001 Device 005: ID 0483:5760 STMicroelectronics 8Bitdo```

This seems to correspond to the following entry in `/usr/share/fwupd/quirks.d/ebitdo.quirk`:
```
[DeviceInstanceId=USB\VID_0483&PID_5760]
Plugin = ebitdo
Flags = is-bootloader,will-disappear
InstallDuration = 120
```

However `fwupdmgr get-devices` doesn't list the adapter.

Trying to update manually using `8BitDo_USB_RR_Firmware_V1.30.dat` which I got from [8Bitdo's support page](https://support.8bitdo.com/) ([direct link](https://download.8bitdo.com/Firmware/Receiver/usb_adapter/8BitDo_USB_RR_Firmware_V1.30.zip)):

`sudo /usr/libexec/fwupd/fwupdtool install-blob 8BitDo_USB_RR_Firmware_V1.30.dat`

I get the following message:

```
08:44:16:0493 FuEngine             failed to add USB device 0483:5760: outbuf size wrong, expected 4 got 8
```

The error message is from [plugins/ebitdo/fu-ebitdo-device.c:146](https://github.com/fwupd/fwupd/blob/master/plugins/ebitdo/fu-ebitdo-device.c#L146), with the call originating from [plugins/ebitdo/fu-ebitdo-device.c:325](https://github.com/fwupd/fwupd/blob/master/plugins/ebitdo/fu-ebitdo-device.c#L325)

So, I tried the following change:

```diff
diff --git a/plugins/ebitdo/fu-ebitdo-device.c b/plugins/ebitdo/fu-ebitdo-device.c
index 309e5da3..b9f1f100 100644
--- a/plugins/ebitdo/fu-ebitdo-device.c
+++ b/plugins/ebitdo/fu-ebitdo-device.c
@@ -289,7 +289,7 @@ fu_ebitdo_device_setup (FuDevice *device, GError **error)
 {
        FuEbitdoDevice *self = FU_EBITDO_DEVICE (device);
        gdouble tmp;
-       guint32 version_tmp = 0;
+       guint64 version_tmp = 0;
        guint32 serial_tmp[9] = {0};

        /* in firmware mode */
```

And, to my surprise, the adapter gets recognized:

```
lahvuun@lahvuun ~/git/fwupd fwupdmgr get-devices
System Product Name
│
├─8Bitdo:
│     Device ID:           9037d31b0ec04c23441d60a55c29f07b6869eb59
│     Summary:             A redesigned classic game controller
│     Current version:     1.30
│     Vendor:              8Bitdo (USB:0x0483)
│     Install Duration:    2 minutes
│     GUIDs:               6c1f10fc-1878-500a-939f-7db26c827953
│                          bc1739b5-313d-5a82-a123-95e5dc4a5702
│     Device Flags:        • Updatable
│                          • Is in bootloader mode
│                          • Device will not re-appear after update completes
```

I was also able to update from v1.22 to v1.30 with `sudo /usr/libexec/fwupd/fwupdtool install-blob 8BitDo_USB_RR_Firmware_V1.30.dat`

---

I'd submit this as a pull request, but it seems that people have been able to update the same adapter without the change: https://github.com/fwupd/fwupd/issues/994#issuecomment-515565646 so I'm not quite sure how to properly fix this and not break it for others.

Please tell me if you need any more info.

superm1, 2019-12-13:
@Steltek, can you upgrade to the current FW that @Lahvuun has and see if the behavior changed for you?

hughsie, 2019-12-13:
My guess would be the firmware either deliberately or accidentally started returning a uint64_t rather than the previous uint32_t. Why it would do that I have no idea and we'd have to ask 8bitdo, which I'll do on twitter now.

Steltek, 2019-12-14:
At 1.27, it shows when doing get-devices (provided it is in update mode):

```
root@laptop:/# fwupdmgr get-devices

HP EliteBook 8470p
│
├─8Bitdo:
│     Device ID:           54c916aa848583256a9b6959446d0890818cbe18
│     Summary:             A redesigned classic game controller
│     Current version:     1.27
│     Vendor:              8Bitdo (USB:0x0483)
│     Install Duration:    2 minutes
│     Serial Number:       BootMod
│     GUIDs:               6c1f10fc-1878-500a-939f-7db26c827953 ← USB\VID_0483&PID_5760&REV_0200
│                          bc1739b5-313d-5a82-a123-95e5dc4a5702 ← USB\VID_0483&PID_5760
│     Device Flags:        • Updatable
│                          • Is in bootloader mode
│                          • Device will not re-appear after update completes
│   

```
Installing the 1.30 update with fwupdtool also works:
```
root@laptop:/# /usr/lib/fwupd/fwupdtool install-blob 8BitDo_USB_RR_Firmware_V1.30.dat
Loading…                 [-                                      ]22:11:08:0613 FuPluginUefi         Error opening directory “/sys/firmware/efi/esrt/entries”: No such file or directory
Loading…                 [***************************************]
Choose a device:
0.      Cancel
1.      bbbf1ce3d1cf15550c3760b354592040292415bb (3rd Gen Core processor Graphics Controller)
2.      54c916aa848583256a9b6959446d0890818cbe18 (8Bitdo)
3.      c3857357b6f271e9254918e2d524d5a5b6f6b747 (BCM20702A0)
4.      e2623122c99d58220498aacbfcfdb1baebbae3c5 (Intel AMT [unprovisioned])
5.      1869f5eae449e4273b74ff6cea32a9bb935430b2 (SD/MMC Host Controller)
6.      602b0a6cc821d155208724f0e22f8d111542b74c (Samsung SSD 840 Series)
7.      123fd4143619569d8ddb6ea47d1d3911eb5ef07a (System Firmware)
2
Installing on 8Bitdo…    [  -                                    ]
Writing…                 [***************************************]
```
Doing get-devices (with the device in Update mode), still works with 1.30:
```
root@laptop:/# fwupdmgr get-devices
HP EliteBook 8470p
│
├─8Bitdo:
│     Device ID:           54c916aa848583256a9b6959446d0890818cbe18
│     Summary:             A redesigned classic game controller
│     Current version:     1.30
│     Vendor:              8Bitdo (USB:0x0483)
│     Install Duration:    2 minutes
│     Serial Number:       BootMod
│     GUIDs:               6c1f10fc-1878-500a-939f-7db26c827953 ← USB\VID_0483&PID_5760&REV_0200
│                          bc1739b5-313d-5a82-a123-95e5dc4a5702 ← USB\VID_0483&PID_5760
│     Device Flags:        • Updatable
│                          • Is in bootloader mode
│                          • Device will not re-appear after update completes
```

This is on a Debian amd64 system (Kernel 4.19.0-6-amd64) with the following fwupdmgr version:
```
root@laptop:/# fwupdmgr --version
client version: 1.3.4
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.4
```

closed 2020-01-20 by stale[bot]

javanthropus, 2020-05-23:
I'm experiencing exactly the same issue on Ubuntu 20.04:
```sh
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

I tried it with the device fresh out of the box and after upgrading it from version 1.27 to version 1.30 using a Mac and the official upgrade software.  `lsusb` reports the device when it's in upgrade mode, but `fwupdmgr` does not.

Lahvuun, 2020-05-23:
A-ha! So I'm not the only one who experienced this.

Back when I was first dealing with this issue my theory was that there might actually be multiple revisions of the receiver which for some reason behave differently, yet are indistinguishable in bootloader mode.

For example, there's actually two separate but very similar receivers:

[image]

I have the red one. Supposedly the only difference between them is that the grey one supports PS Classic. This is even reflected in the manuals (left is the manual for the grey receiver, right is for the red one):

[image]

However, if I connect my M30 through my red receiver and hold Select + Down the receiver reconnects as: `Bus 001 Device 008: ID 054c:0cda Sony Corp. PlayStation Classic controller`. 

Maybe once 8Bitdo released the grey receiver, they stopped producing the original red one, and simply started selling the same base receiver with PS Classic support as two separate products: red and grey? If @Steltek has the original red one, or the new grey one, it might explain the discrepancy between returned data types.

---

@hughsie I checked your twitter and it looks like 8Bitdo never replied you, at least on twitter. If the reason you never pursued this further was 8Bitdo not contacting you back, would you be interested in a pull request trying to work around this? I was thinking maybe it'd be possible  to simply try uint32_t and if it fails, then try uint64_t before giving up. I don't see how this could break currently working devices, while it'd fix this issue.

I actually tried to find how the windows firmware updater does it, but either ILSpy is generating bad code, or 8Bitdo is doing something really weird to get the version number from the payload:
```C#
		while (num != 0)
		{
			num--;
			if (SHid.writeHid(version_CMD, length) != 1)
			{
				continue;
			}
			byte[] array = new byte[64];
			if (SHid.readHid(array, (uint)array.Length) != 0
                            || (gamePad.pid() == 12544 && array[0] != 34)
                            || (array[0] == 0 && array[1] == 0 && gamePad.pid() == Const.GBAdapter_PID)
                            || ((gamePad.pid() == 12294 || gamePad.pid() == 12296) && array[0] != 33))
			{
				continue;
			}
			getGamePadDetails(gamePad, array);
```
```C#
private static GamePadDetailsBase getGamePadDetails(GamePadDetailsBase gamePad, byte[] readData)
{
	if (gamePad.vid() == 121 && gamePad.pid() == 17)
	{
		return gamePad;
	}
	gamePad.Version = (float)(double)(uint)(readData[1] | (readData[2] << 8)) / 100f;
	if (gamePad.pid() == Const.USB24G_PID || gamePad.pid() == 12296)
	{
		string tmp = Convert.ToString(readData[1] | (readData[2] << 8), 16);
		gamePad.Version = Tools.getVersionIntWithString(tmp) / 100f;
	}
	gamePad.Beta = readData[3];
	return gamePad;
}
```
Although this is code for getting versions from devices in normal mode, from what I can tell, 8Bitdo do something pretty similar for those in bootloader mode.

So, I figured, it might be easier to just try uint32_t and then uint64_t.

Steltek, 2020-05-25:
I have the red/orange one, shown on the right above, called '8BitDo Wireless USB Adapter' on their site (https://www.8bitdo.com/wireless-usb-adapter/). The grey one is apparently called 'USB Wireless Adapter for PS classic edition' (https://www.8bitdo.com/usb-adapter-ps1-classic/)

javanthropus, 2020-05-25:
I also have the red/orange one.  If there is any other information I can dig up about it, just tell me how to gather it.

graduatio, 2021-08-01:
This issue seems to persist. I receive the same „outbuf size wrong, expected 4 got 8“ error message when trying to detect and update my adapters.

Ubuntu 20.04
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11

graduatio, 2021-08-03:
Also tested with client version 1.6.2 with the same result. Is there any chance to fix this issue the way @Lahvuun proposed? Otherwise fwupd seems to be incompatible with some 8BitDo adapters.

hughsie, 2021-08-03:
> it might be easier to just try uint32_t and then uint64_t.

If that works, it's fine for me. Can someone with the affected hardware spin up a pull requests please?

PR #3615 (2021-08-08): Fix detection of 8Bitdo wireless usb adapter
Limits copying the get version response payload to the size of the out buffer.
Devices that return a larger payload pack the necessary version information at the beginning of the payload.

Fixes: #1661

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 3604b897 (2021-08-08): Fix detection of 8Bitdo wireless usb adapter

Limits copying the get version response payload to the size of the out buffer.
Devices that return a larger payload pack the necessary version information at
the beginning of the payload.

Fixes: #1661
commit ceaa36ce (2021-08-08): Fix detection of 8Bitdo wireless usb adapter

Limits copying the get version response payload to the size of the out buffer.
Devices that return a larger payload pack the necessary version information at
the beginning of the payload.

Fixes: #1661
commit b9b38fb5 (2021-08-08): Fix detection of 8Bitdo wireless usb adapter

Limits copying the get version response payload to the size of the out buffer.
Devices that return a larger payload pack the necessary version information at
the beginning of the payload.

Fixes: #1661