fwupd/fwupd #1662: Clarify install command parameters

zabbal, 2019-12-15:
Both man page and --help output show DEVICE-ID or REMOTE-ID with the exception of install command which is documented as "install FILE [ID] ". Please clarify what kind of ID is expected.

While at it, would be nice to unify *E-ID vs *E_ID - it looks odd that half of the commands use dash while other half uses underscore.

```shell
$ fwupdmgr --version
client version: 1.2.10
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.10
```

- Operating system and version: Ubuntu 19.10
- How did you install fwupd: apt

hughsie, 2019-12-18:
> install FILE [ID]

This is supposed to be `DEVICE_ID` or `DEVICE-ID` -- I'd would be awesome if you could do a PR to fix this for us -- I'm happy to help you get started if you're stuck. I'm also not sure what's convention in the --help output, either dashes or underscores is fine but you're correct we should be consistent. If you want, you get to choose and fix it in the PR :)

Thanks, and have a great Christmas.

closed 2020-01-09 by superm1

zabbal, 2020-01-14:
Thanks for fixing this - sadly I couldn't allocate time during vacation to work on it.

commit 234c864e (2020-01-09): trivial: fu-util/fu-tool: Unify usage of - and _ in help output (Fixes: #1662)