fwupd/fwupd #1665: Information about MST Hub firmware update using eeprom file from official Windows executable

florentc, 2019-12-18:
Hello, please forgive me and remove the issue if the following has no place in the issue tracker.

I own an HP laptop (Folio 1040 g3) along with its docking station (HP 2013 UltraSlim Docking Station) which causes serious external display artifacts and unstability under the current firmware of the embedded Synaptics VMM2230 DisplayPort MST Hub. The current version (2.30) is listed in the output of `fwupdmgr get-devices` and there is a flag "Updatable".

HP has released a fix (version 2.33) as a Windows executable (encompassing a `.eeprom` file) but there are no available updates through `fwupd`. Since I fail to perform the update using wine and cannot consider setting up a Windows partition for the moment, I would like to know if there is some way to perform it using the eeprom file from HP's official fix.

I noticed there is a `synaptics-mst` plugin in `fwupd` and official support for Dell docking stations with MST hubs of the same family of products (VMM****). The Dell equivalents of HP's Windows executable to update the firmware also seem to share common, if not identical, components.

Do you think there is some way to perform the update relying on the eeprom file provided by HP and existing fwupd support for similar products?

hughsie, 2019-12-18:
I guess it's possible, but it will have had zero testing by HP on Linux. The stock answer is talk to the OEM, in this case HP and convince them to upload it to the LVFS. In the meantime, if you're willing to give it a try, and wont hold me responsible if it breaks, `sudo /usr/libexec/fwupd/fwupdtool install-blob` is probably what you're looking for.

florentc, 2019-12-27:
Thank you very much for the information. I will try this method eventually, the device is barely usable anyway.

closed 2020-01-06 by superm1

Rondom, 2020-07-21:
For those finding this via Google (or in the unlikely event that HP puts the same priority on FW updates on Linux like Dell and now Lenovo do)...

I have successfully updated the Synaptics MST HUB VMM2230 in the "HP 2013 UltraSlim Docking Station" as follows:
Current firmware version as of this writing is 2.33.004 delivered via [SoftPaq sp100099](http://ftp.hp.com/pub/softpaq/sp100001-100500/sp100099.exe). Later versions can be found on the corresponding drivers page for compatible laptops. The file can be extracted using `7za x sp100099.exe`.

```
❯ sha256sum tesla_firmware_hp_Ver2_33_004.eeprom 
a81b9108b866c6b1ca7e5048b6ccf6a66f187aedb484a4da0c9c4c252030b6ad  tesla_firmware_hp_Ver2_33_004.eeprom

❯ fwupdmgr get-devices                          
HP EliteBook 840 G3
│
<snip>
│   
└─VMM2230:
      Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
      Summary:             Multi-Stream Transport Device
      Current version:     2.30.00
      Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
                           c0320ff3-dab2-5339-8af1-3a9e845bc875
                           c4c6413a-5ebc-5eb2-931d-22e42123db5e
      Device Flags:        • Updatable

❯ sudo fwupdtool install-blob ./sp100099/tesla_firmware_hp_Ver2_33_004.eeprom d762543f8c20f636e6fff031a000078d3e10c600
Loading…                 [***************************************]
Installing on VMM2230…   [ -                                     ]
Restarting device…       [***************************************]

❯ fwupdmgr get-devices                          
HP EliteBook 840 G3
│
<snip>
│   
└─VMM2230:
      Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
      Summary:             Multi-Stream Transport Device
      Current version:     2.33.04
      Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
                           c0320ff3-dab2-5339-8af1-3a9e845bc875
                           c4c6413a-5ebc-5eb2-931d-22e42123db5e
      Device Flags:        • Updatable
```

joevt, 2021-02-21:
I have 3 DisplayPort 1.4 MST Hubs:
- CalDigit SOHO (Synaptics VMM5210 - supports DSC)
- HP Thunderbolt Dock G2 (Synaptics VMM5330 - supports DSC)
- Delock 87737 (Synaptics VMM5330 - doesn't supports DSC)

fwupd doesn't list those devices. Is there a way to make fwupd list them? It seems to me that fwupd should be able to find these devices by scanning `/dev/drm_dp_aux*` and show some info about them. They all have the same Synaptics three-byte OUI in the DPCD registers.

There's a firmware update for the HP Thunderbolt Dock G2 which updated the firmware of the DisplayPort Hub from 5.3.0.1 to 5.4.0.5 and now supports Display Stream Compression (DSC). My Delock 87737 has firmware 5.3.x.x and doesn't support DSC. Can I apply the firmware from the HP to the Delock?

I have a 512KiB (exactly) firmware file from the HP firmware updater. Is there a way to tell if it's the correct format for fwupd?

hughsie, 2021-02-21:
> Is there a way to make fwupd list them

https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/synaptics-mst.quirk has the GUIDs to match. We don't match all VMMs as we don't know what will happen if we probe them, so only list tested devices.

> Is there a way to tell if it's the correct format for fwupd

We do have a sanity check, but I'm not going to tell you to try it in case it bricks the hardware in some way: https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/fu-synaptics-mst-device.c#L787

joevt, 2021-02-22:
After using `-v`, I found that the mst plugin doesn't run when there's an AMD gpu installed (I have an AMD W5700). There's a message `amdgpu has known issues with synaptics_mst`. According to https://github.com/fwupd/fwupd/issues/1121 , the problem is that accessing `/dev/drm_dp_aux*` causes a hang. The plugin doesn't run even if I connect the MST hubs only to Intel graphics. I have to remove my AMD gpu. Wouldn't it be better for the code to skip only the `drm_dp_aux*` devices that are from the AMD GPU? Shouldn't there be an override to allow working with AMD GPU ports in case the AMD driver gets fixed one day?

> We do have a sanity check, but I'm not going to tell you to try it in case it bricks the hardware in some way: https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/fu-synaptics-mst-device.c#L787

There's a command `fwupdtool firmware-parse` but I don't think it has an option for synaptics-mst firmware type. Maybe this could be updated so it can be used as a sanity check. `fwupdtool firmware-read` doesn't work for MST firmware, so I can't compare the read result with the file from the firmware updater to verify that they are identical or at least have the same format.

> https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/synaptics-mst.quirk has the GUIDs to match. We don't match all VMMs as we don't know what will happen if we probe them, so only list tested devices.

I think I need to know the board id to make the GUID. Is there a method to get that info?
https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/synaptics-mst-evb.quirk  has an example to ignore board-id. It seems to work. I did `sudo gedit /usr/share/fwupd/quirks.d/synaptics-mst.quirk` and added:
```
[DeviceInstanceId=MST-panamera-vmm5330-2]
Flags = ignore-board-id
```
Actually, maybe this works because I removed the AMD GPU. Commenting out those lines I added doesn't change the fact that fwupd is now listing my MST hubs. That means I don't need those lines, or the changes I made were cached somewhere?

fwupdtool and fwupdmgr shows the following GUID and version info:
- HP Thunderbolt Dock G2: GUID:MST-panamera-vmm5333-849 version:5.04.05
- Delock 87737: GUID:MST-panamera-vmm5330-2 version:5.03.08
- CalDigit SOHO: GUID:MST-panamera-vmm5210-2 version:5.05.202

I'm not sure what a VMM5333 is. Is it possible this was VMM5330 before I did the HP Thunderbolt Dock G2 firmware update recently? I don't think I want to try updating the Delock with the HP firmware without knowing that the chips are the same.

The MST plugin seems to works even when I chain all the MST hubs together (Intel -> HP -> CalDigit -> Delock) so that's cool. Linux doesn't show the MST topology anywhere though. Sometimes the DP ports of a MST Hub don't appear unless something is connected. The CalDigit shows three ports in Linux even though it physically only has two. So in total, I see 12 DP aux ports numbered 0 to 11. I need to manually switch the display connected at the end of the chain from 4K60 to 4K30 to get an image. I thought that Linux or Intel drivers would be smart enough to automatically choose the max refresh rate for the available bandwidth. I am using Ubuntu 20.04.2 LTS.

superm1, 2021-02-22:
> Wouldn't it be better for the code to skip only the drm_dp_aux* devices that are from the AMD GPU?

Patches certainly welcome here if you are able to come up with a way to do this.  At a first glance walking up the change in the udev database, I find that you have to go all the way up from drm_dp_aux to the connector, to the card and then to the PCI subsystem to be able to tell what vendor supports it.

However I would still worry about getting caught with hybrid solutions where they use shared memory to display something.

>Shouldn't there be an override to allow working with AMD GPU ports in case the AMD driver gets fixed one day?

Code changes are cheap.  If it's fixed some day we can set a minimum kernel version to use AMD.

>I'm not sure what a VMM5333 is.

That's a model number.

>Sometimes the DP ports of a MST Hub don't appear unless something is connected. 

Right, that's a limitation of how DP aux channel works.

>HP Thunderbolt Dock G2: GUID:MST-panamera-vmm5333-849 version:5.04.05
Delock 87737: GUID:MST-panamera-vmm5330-2 version:5.03.08
CalDigit SOHO: GUID:MST-panamera-vmm5210-2 version:5.05.202

I guess you can submit a PR to add these 3 devices into the quirk list to probe.

superm1, 2021-02-22:
>I guess you can submit a PR to add these 3 devices into the quirk list to probe.

Actually the last 2 I would be worried about are reference ID's not proper ones with a customer ID.

hughsie, 2021-02-22:
> are reference ID

Well spotted. I don't think we could support those in fwupd in any meaningful way, right?

superm1, 2021-02-22:
>Do you know if there's an open upstream issue in the kernel bugtracker?

Yeah there's a line of bugs that link from comments in our codebase.  They haven't been touched in _ages_.

>Well spotted. I don't think we could support those in fwupd in any meaningful way, right?

I think maybe add them to the quirk list that's not installed, but in my mind if you can't distinguish an EVB from a retail product you shouldn't try to update it.

hughsie, 2021-02-22:
> if you can't distinguish an EVB from a retail product you shouldn't try to update it

In your experience, whats the chance the OEM is just using the reference firmware too? It would explain the _shows three ports in Linux even though it physically only has two_ if the vendors are just shipping the provided blob.

superm1, 2021-02-22:
>In your experience, whats the chance the OEM is just using the reference firmware too?

Yes - I think it's very likely to be the reference firmware if the customer ID hasn't changed.  I don't think that changes things though in terms of whether we should host and enable firmware updates for the reference ID.  What if there is a new bug introduced in the reference firmware that only manifests on one OEM's product?  It's up to the OEM to do the testing and validate.  We really need something to tie it back to a physical device model if it's using a reference firmware and we want to update it.

> It would explain the shows three ports in Linux even though it physically only has two if the vendors are just shipping the provided blob.

Well another explanation is that a 3rd port is just not populated in the device.  You might open it up and find some empty pads somewhere :)

hughsie, 2021-02-22:
> if the customer ID hasn't changed

Is there a hard and fast rule we can use for "unchanged customer ID"? I think we should probably put a g_warning() in the code somewhere. Or maybe unset updatable or something like that?

superm1, 2021-02-22:
>Is there a hard and fast rule we can use for "unchanged customer ID"? I think we should probably put a g_warning() in the code somewhere. Or maybe unset updatable or something like that?

Yeah, unsetting updatable and setting update_error sounds like a pretty good consolation.  At least people can use `fwupdtool install-blob` to update then without the concern of breaking the ecosystem.

hughsie, 2021-02-22:
So something like `board_id <= 0x2 -> !UPDATABLE`?

superm1, 2021-02-22:
I _think_ the customer ID is encoded in the second byte.  So basically if it's got a 0 second byte then it's a reference firmware.

hughsie, 2021-02-22:
Indeed! https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-mst/fu-synaptics-mst-connection.h#L17 -- fixed in https://github.com/fwupd/fwupd/pull/2923

joevt, 2021-02-23:
> > I'm not sure what a VMM5333 is.
> 
> That's a model number.

I mentioned it because the Synaptics website has a VMM5330 but not a VM5333 and all my searches indicated that the HP used the VMM5330.

> > HP Thunderbolt Dock G2: GUID:MST-panamera-vmm5333-849 version:5.04.05
> > Delock 87737: GUID:MST-panamera-vmm5330-2 version:5.03.08
> > CalDigit SOHO: GUID:MST-panamera-vmm5210-2 version:5.05.202
> 
> I guess you can submit a PR to add these 3 devices into the quirk list to probe.

They seem to be appearing now without modifying the quirk since I removed the AMD GPU. So I don't think there's anything to do there except for testing on a fresh install of Linux.

My comments about customer ID are in the new thread at https://github.com/fwupd/fwupd/pull/2923

superm1, 2021-02-23:
>I mentioned it because the Synaptics website has a VMM5330 but not a VM5333 and all my searches indicated that the HP used the VMM5330.

The only definitive way to know about it would be to open and look at the silkscreen on the chip.  But I would say 🤷 
 
>They seem to be appearing now without modifying the quirk since I removed the AMD GPU. So I don't think there's anything to do there except for testing on a fresh install of Linux.
>My comments about customer ID are in the new thread at #2923

👍

joevt, 2021-02-24:
> The only definitive way to know about it would be to open and look at the silkscreen on the chip. But I would say 🤷

I just remembered I took pictures of the inside of the HP Thunderbolt Dock G2 a while ago. At the time I was looking for confirmation of the JHL7440 Thunderbolt controller. The MST chip is definitely a Synaptics VMM5333BJG. The Delock and CalDigit don't have screws so I won't be looking inside those.

vvmruder, 2021-03-11:
> For those finding this via Google (or in the unlikely event that HP puts the same priority on FW updates on Linux like Dell and now Lenovo do)...
> 
> I have successfully updated the Synaptics MST HUB VMM2230 in the "HP 2013 UltraSlim Docking Station" as follows:
> Current firmware version as of this writing is 2.33.004 delivered via [SoftPaq sp100099](http://ftp.hp.com/pub/softpaq/sp100001-100500/sp100099.exe) ([info](http://ftp.hp.com/pub/softpaq/sp100001-100500/sp100099.html)). Later versions can be found on the corresponding drivers page for compatible laptops. The file can be extracted using `7za x sp100099.exe`.
> 
> ```
> ❯ sha256sum tesla_firmware_hp_Ver2_33_004.eeprom 
> a81b9108b866c6b1ca7e5048b6ccf6a66f187aedb484a4da0c9c4c252030b6ad  tesla_firmware_hp_Ver2_33_004.eeprom
> 
> ❯ fwupdmgr get-devices                          
> HP EliteBook 840 G3
> │
> <snip>
> │   
> └─VMM2230:
>       Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
>       Summary:             Multi-Stream Transport Device
>       Current version:     2.30.00
>       Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
>       GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
>                            c0320ff3-dab2-5339-8af1-3a9e845bc875
>                            c4c6413a-5ebc-5eb2-931d-22e42123db5e
>       Device Flags:        • Updatable
> 
> ❯ sudo fwupdtool install-blob ./sp100099/tesla_firmware_hp_Ver2_33_004.eeprom d762543f8c20f636e6fff031a000078d3e10c600
> Loading…                 [***************************************]
> Installing on VMM2230…   [ -                                     ]
> Restarting device…       [***************************************]
> 
> ❯ fwupdmgr get-devices                          
> HP EliteBook 840 G3
> │
> <snip>
> │   
> └─VMM2230:
>       Device ID:           d762543f8c20f636e6fff031a000078d3e10c600
>       Summary:             Multi-Stream Transport Device
>       Current version:     2.33.04
>       Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
>       GUIDs:               49ec4eb4-c02b-58fc-8935-b1ee182405c7
>                            c0320ff3-dab2-5339-8af1-3a9e845bc875
>                            c4c6413a-5ebc-5eb2-931d-22e42123db5e
>       Device Flags:        • Updatable
> ```

I used this hint and fixed my problem with the ultraslim dock D9Y32AA. It now works as expected in all modes with external monitor via DisplayPort. Before This setup was not working. Monitor settings showed monitor correctly (resolution, name, scale factor, etc.). I was also able to move windows on the display. But monitor always stayed blank and announced "no signal".

After updating the firmware to version 2.33.04 as described everything worked! Great! I already thought I will give up. You can find links to the latest versions of firmware here:
https://www.admin-enclave.com/en/articles/windows/477-windows-10-driver-for-hp-elitebook-850-g1-notebook.html

Iam on:
* HP Elitebook 840 G6
* Intel UHD
* Ubuntu 20.04
* BenQ PD2700U Monitor

joevt, 2021-03-21:
I received from Delock a new version (5.04.135) of the firmware for my Delock 87737 MST Hub. I was able to apply it using `fwpdtool`. Now the Delock 87737 supports DSC like the MST hubs in the HP Thunderbolt Dock G2 and CalDigit SOHO.  DSC allows connecting three 4K 60Hz 8bpc RGB displays to the hub.

hughsie, 2021-03-22:
Can you share your delock vendor contact? It sounds like we need firmware on the LVFS!

joevt, 2021-03-22:
@hughsie, I did mention to the guy the website https://fwupd.org where a vendor can post firmware updates.

@superm1 said in a previous post above that the CalDigit and Delock are using a Reference ID (the number 2) instead of a Customer ID for the Board ID. Their firmware wouldn't be suitable (at least for automatic updates) since their products can't be distinguished from other products that use the Reference ID.

As I showed in a previous post, only my HP has a valid Customer ID for the Board ID (the number 849) and HP already has their own firmware updater for Windows (which the firmware can be extracted from). The CalDigit is new enough that it doesn't have a firmware update.

HP Thunderbolt Dock G2: GUID:MST-panamera-vmm5333-849 version:5.04.05
Delock 87737: GUID:MST-panamera-vmm5330-2 version:5.04.135
CalDigit SOHO: GUID:MST-panamera-vmm5210-2 version:5.05.202

hughsie, 2021-03-22:
> since their products can't be distinguished from other products that use the Reference ID

If it's literally reference firmware from synaptics directly, that cleanly installs onto a reference platforms, maybe it should be uploaded *by* synaptics?

joevt, 2021-03-23:
> If it's literally reference firmware from synaptics directly, that cleanly installs onto a reference platforms, maybe it should be uploaded _by_ synaptics?
Maybe it's like Thunderbolt firmware with different bits that can be customized by the vendor, so Intel doesn't provide the firmware updates. They go one step further and don't provide the driver updates either. It's a pain because the vendors all have different versions of the firmware and drivers. Maybe Delock customized some stuff but didn't customize the customer ID.

hughsie, 2021-03-23:
I think that's a question for Delock -- I'd be happy to be included in an email cc if that might help.

superm1, 2021-03-23:
I still stand by the only safe way to do this is to have a customer ID in the firmware.  Even if they aren't modifying it in any way, there needs to be a way to identify this specific hardware takes this specific reference firmware.  I'm not comfortable with the precedent in any other way.  Delock can absolutely work with Synaptics to get a customer ID added into the firmware.