fwupd/fwupd #1667: Failed to update Logitech Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036

da2x, 2019-12-20:
```shell
sudo fwupdmgr --verbose update
Upgrade available for Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR24.11_B0036 for Unifying Receiver...
(fwupdmgr:28782): FuCommon-DEBUG: 13:21:52.133: creating path /root/.cache/fwupd
(fwupdmgr:28782): FuMain-DEBUG: 13:21:52.134: skpping download as file already exists
(fwupdmgr:28782): Fwupd-DEBUG: 13:21:52.135: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:28782): Fwupd-DEBUG: 13:21:52.136: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:28782): Fwupd-DEBUG: 13:21:52.175: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:28782): Fwupd-DEBUG: 13:21:52.177: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
└─Unifying Receiver:
      Device ID:           ef5791f75bea792344af5e07f52d4bb54563221c
      Summary:             A miniaturised USB wireless receiver (bootloader)
      Current version:     RQR24.00_B0000
      Bootloader Version:  BOT03.03_B0009
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    18 seconds
      GUIDs:               279ed287-3607-549e-bacc-f873bb9838c4
                           6157c022-3ead-577a-9666-1c22f32a2648
                           87fd7145-3913-50c8-bfcb-86f85006d7d1
                           cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
      Device Flags:        • Updatable
                           • Supported on remote server
                           • Is in bootloader mode
```
Please answer the following questions:

- Operating system and version: Fedora 31 Workstation
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Preinstalled
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

FFY00, 2019-12-20:
I am not sure but I think we are wrongly detecting the device being in bootloader mode, but I could be wrong. IIRC when we are in bootloader mode fwupd doesn't report the firmware version.

Can you try again but also provide the output of `lsusb` together with the other logs?

da2x, 2019-12-20:
It still fails in the same way.

`lsusb` before retrying:

    Bus 005 Device 017: ID 046d:c52b Logitech, Inc. Unifying Receiver

`lsusb` after retrying:

   Bus 005 Device 017: ID 046d:aaad Logitech, Inc. USB BootLoader

FFY00, 2019-12-20:
After it failing one time (usb pid = `aaad`) try updating again and see f it works.

da2x, 2019-12-20:
> After it failing one time (usb pid = aaad) try updating again and see f it works.

I then get a different error:

    ignoring Unifying Receiver [41407450b96ca940bd7325682a1c741969c48047] as not updatable

FFY00, 2019-12-20:
@superm1 any ideas? I am not sure why the first try is exiting with `failed to send: failed to write: wrote -1 of 7`.

rismakki, 2020-01-05:
I had similar failure, I've had it all the time while fwupdmgr has been in Fedora.

Once fwupdmgr reported today that this was actually a known issue from last December:
**https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Logitech-Restart-Failure**

With me there was also failure 
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7

My failing firmware update took 6.3 seconds (pressed Y by hand),
while doing "time fwupdmgr update" by hand.

**After "wrote -1 of 7", there was less than three seconds until USB device introduced
itself again (old M215 mouse)**
Then I removed the USB device and plugged it again.
Then I tried the USB receiver update again. No success, same problem.

**According the known issue I understood finally how to do the update successfully**:
- Do fwupdmgr update, press Y
- Wait 30 seconds (maybe with very old M215 mouse just the total 10 second wait would have been fine).
- Do fwupdmgr update, press Y again, while mouse is not responsive.

"Successfully installed firmware"

rismakki, 2020-01-05:
Some strace snippet at the failure point:
There were 135 repeated Timeout poll() messages, according to strace log,
total strace logging happened between (stat command for strace output):
Birth: 2020-01-05 18:43:08.974502357 +0200
Modify: 2020-01-05 18:43:16.488453491 +0200

poll([{fd=3, events=POLLIN}], 1, 40)    = 0 (Timeout)
poll([{fd=3, events=POLLIN}], 1, 40)    = 0 (Timeout)
poll([{fd=3, events=POLLIN}], 1, 40)    = 0 (Timeout)
poll([{fd=3, events=POLLIN}], 1, 40)    = 0 (Timeout)
poll([{fd=3, events=POLLIN}], 1, 40)    = 0 (Timeout)
poll([{fd=3, events=POLLIN}], 1, 40)    = 1 ([{fd=3, revents=POLLIN}])
read(3, "\1\0\0\0\0\0\0\0", 16)         = 8
write(3, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x562cadcf5af0, FUTEX_WAKE_PRIVATE, 2147483647) = 0
close(8)                                = 0
write(2, "failed to detach to bootloader: "..., 79) = 79

While I stored the strace output,
fwupd daemon reported:
journalctl -u fwupd -o short-precise | grep 18:43

Jan 05 18:43:16.501496 localhost.localdomain fwupd[2816]: 16:43:16:0501 FuDevice             disabling polling: failed to open /dev/hidraw2

So for me the firmware update would have succeeded, if polling would
not have been disabled after roughly 7.5 seconds (since fwupdmgr program start with strace). 10 second patience would have been enough for me (not a new encrypted device that requires 30 seconds polling time).

Unifying Receiver (after successful firmware update though):
      Device ID:           9db5efdfd761a9a78ce7a8bf43793d1b34a2a919
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.10_B0036
      Bootloader Version:  BOT03.01_B0008
      Vendor:              Logitech, Inc. (USB:0x046D)
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                           de81631f-2117-51f8-bcb1-4f874a293b29
      Device Flags:        • Updatable
                           • Supported on remote server

rpm -q fwupd
fwupd-1.3.5-1.fc31.x86_64

fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5

Before failed update, lsusb:
Bus 001 Device 029: ID 046d:c52b Logitech, Inc. Unifying Receiver
After failed update, lsusb:
Bus 001 Device 030: ID 046d:aaac Logitech, Inc. USB BootLoader

                              "Plugin" : "logitech_hidpp", (within Git master branch "unifying" was renamed into "logitech_hidpp", see #1329)

**So this "Logitech Restart Failure" known issue refers still into the old name "unifying", instead of "logitech_hidpp"**

Here was my bug report Hash ID uploaded by fwupdmgr:
Proceed with upload? [Y|n]: y
Upload message: 17a058f1703a159108c5c2814c2a3a72ac103969 replaces old report
Successfully uploaded 1 report

[root@localhost ~]# time fwupdmgr update
Upgrade available for Unifying Receiver from RQR24.05_B0029 to RQR24.10_B0036
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR24.10_B0036 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7

real	0m6.754s (no strace here, maybe total time varied)

hughsie, 2020-01-06:
> So this "Logitech Restart Failure" known issue refers still into the old name "unifying", instead of "logitech_hidpp"

I've changed the known issue into a regex that supports both the old and new plugin name.

superm1, 2020-01-06:
Looking at the relevant code: `fu_logitech_hidpp_runtime_detach` sends an argument of `FU_UNIFYING_DEVICE_TIMEOUT_MS` to `fu_logitech_hidpp_send` but that's not used by the function.

I wonder if a sufficient fix would be wrap that IO channel write call in a loop that retries once a second until the timeout is actually hit and only fail after that.

FFY00, 2020-01-06:
Did we end up fixing not notifying the user a replug is needed?

superm1, 2020-01-06:
Users should be notified about needing to replug when those instances come up now.

superm1, 2020-01-06:
@rismakki would you be able to see if PR https://github.com/fwupd/fwupd/pull/1691 fixes the issue for you?  Completely untested by me - I don't have any Logitech devices that operate this way.

superm1, 2020-01-08:
OK so I want to bring all the people with these issues into a single thread.  I think we actually have 3 duplicate issues all of this same item.

Duplicate of https://github.com/fwupd/fwupd/issues/1183

commit 57f35893 (2020-01-07): logitech_hidpp: When detaching to bootloader use non-blocking IO and wait for timeout

Should hopefully resolve issues that take a long time for devices to reboot.
Fixes: #1667
commit 38d70742 (2020-02-11): logitech_hidpp: When detaching to bootloader use non-blocking IO and wait for timeout

Should hopefully resolve issues that take a long time for devices to reboot.
Fixes: #1667
closed 2020-01-08 by superm1