fwupd/fwupd #1669: How to upgrade fwupd to latest using source

kopax, 2019-12-23:
I need to compile `fwupd` from source, I did: 

    git clone git@github.com:fwupd/fwupd.git \
       && cd fwupd \
       && git submodule update --init --recursive \
       && meson build \
       && ninja -C build

`ninja -C build` produce errors such as:

- `undefined reference to `xb_builder_source_ctx_get_bytes'`
- `"argument builder: Unresolved type: 'JsonBuilder*'`

This is the result of the logs:

```
Signing certificate...
[50/446] Linking target libfwupdplugin/libfwupdplugin.so.1.0.0.
FAILED: libfwupdplugin/libfwupdplugin.so.1.0.0 
cc  -o libfwupdplugin/libfwupdplugin.so.1.0.0 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-archive.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-chunk.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-cab.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-guid.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-version.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-device-locker.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-device.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-dfu-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware-common.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware-image.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-hwids.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-ihex-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-io-channel.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-plugin.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-smbios.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-srec-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-udev-device.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-usb-device.c.o' -Wl,--no-undefined -Wl,--as-needed -shared -fPIC -Wl,--start-group -Wl,-soname,libfwupdplugin.so.1 -Wl,-z,relro -Wl,-z,defs -Wl,-z,now libfwupd/libfwupd.so.2.0.0 -Wl,--version-script,/home/dka/workspace/github.com/fwupd/fwupd/libfwupdplugin/fwupdplugin.map /usr/lib/x86_64-linux-gnu/libxmlb.so /usr/lib/x86_64-linux-gnu/libgio-2.0.so /usr/lib/x86_64-linux-gnu/libgobject-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so /usr/lib/x86_64-linux-gnu/libgusb.so /usr/lib/x86_64-linux-gnu/libusb-1.0.so /usr/lib/x86_64-linux-gnu/libgudev-1.0.so -Wl,--export-dynamic /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so -pthread /usr/lib/x86_64-linux-gnu/libarchive.so /usr/lib/x86_64-linux-gnu/libjson-glib-1.0.so /usr/lib/x86_64-linux-gnu/libgcab-1.0.so -Wl,--end-group '-Wl,-rpath,$ORIGIN/../libfwupd' -Wl,-rpath-link,/home/dka/workspace/github.com/fwupd/fwupd/build/libfwupd  
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_convert_quirk_to_xml_cb':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:93: undefined reference to `xb_builder_source_ctx_get_bytes'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_add_quirks_for_path':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:177: undefined reference to `xb_builder_source_add_adapter'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_lookup_by_id':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:289: undefined reference to `xb_silo_query_first_full'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_lookup_by_id_iter':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:351: undefined reference to `xb_silo_query_full'
collect2: error: ld returned 1 exit status
[59/446] Generating Fwupd-2.0.gir with a custom command.
g-ir-scanner: link: cc -o /home/dka/workspace/github.com/fwupd/fwupd/build/tmp-introspectl9xmlhz4/Fwupd-2.0 /home/dka/workspace/github.com/fwupd/fwupd/build/tmp-introspectl9xmlhz4/Fwupd-2.0.o -L. -Wl,-rpath,. -Wl,--no-as-needed -L/home/dka/workspace/github.com/fwupd/fwupd/build/libfwupd -Wl,-rpath,/home/dka/workspace/github.com/fwupd/fwupd/build/libfwupd -lfwupd -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsoup-2.4 -ljson-glib-1.0 -lgio-2.0 -lgobject-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0
libfwupd/fwupd-release.c:1537: Warning: Fwupd: fwupd_release_to_json: argument builder: Unresolved type: 'JsonBuilder*'
libfwupd/fwupd-device.c:1796: Warning: Fwupd: fwupd_device_to_json: argument builder: Unresolved type: 'JsonBuilder*'
ninja: build stopped: subcommand failed.
```

How can I build `fwupd` from source?

```shell
$ fwupdmgr --version # curent version installed 
fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.5
```

- Operating system and version: Debian 10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): from source
- Have you tried rebooting? yes
- Are you using an NVMe disk? no
- Is secure boot enabled (only for the UEFI plugin)? yes

closed 2019-12-24 by kopax

superm1, 2020-01-06:
I guess you got this sorted out?  Would you mind updating the wiki with any insight that might help others?

https://github.com/fwupd/fwupd/wiki/Compilation

kopax, 2020-01-06:
I should have, but now it's a bit too late I haven't kept trace of the change. Sorry about this.