fwupd/fwupd #1670: Problem detecting the Prometheus Fingerprintreader

Z4nterox, 2019-12-23:
Fwupd won't detect the above mentioned fingerprint reader, at least not most of the time.
It did detect it once, after a reboot it was gone, one or two weeks later it showed up again and again, after a reboot, it was gone. I've done nothing out of the ordinary both times, at least not that I know of.

**Lsubs detects it:**
```shell
$ lsusb | grep "Synaptics"
Bus 001 Device 004: ID 06cb:00bd Synaptics, Inc. 
```

**My fwupd version:**
```shell
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5
```

**Here are the devices that fwupd detects:**
```shell
$ fwupdmgr get-devices
20N3S02J00
│
├─Thunderbolt Controller:
│     Device ID:           9612645e5dbf7a4582aefe4cb3911dc6114495e8
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               52265728-359a-574c-9a6a-a23d3b1f952d
│                          0e47d34d-7528-5f66-8e24-a58743b8db61
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─SAMSUNG MZVLB512HBJQ-000L7:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     3L2QEXF7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          f4ceb2a8-09c4-55c0-af2a-13ccd63925b1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           6150dd1f7291b0709289ab8a53cc85a17e117ef2
│     Current version:     0.1.58
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                603baf73-b997-45b5-86b4-2f981a008e18
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:     192.47.1524
│     Minimum Version:     0.0.1
│     GUID:                bf2d0f8b-f9a9-400c-8914-36c225d16eb4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:     0.1.16
│     Minimum Version:     0.1.16
│     GUID:                38ea6335-29ca-417b-8cd4-6b4e5e866f92
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           98a394c65d64ece8aae3126fb0f85364bb915b49
│     Current version:     1.1.16
│     Minimum Version:     0.0.1
│     GUID:                25dd5a7f-4151-401f-ae35-1440807c7bba
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           720296b654bf23d974c282ad9cd59d0b9d33c971
      Current version:     0.0.26406
      Minimum Version:     0.0.1
      GUID:                26aac102-e03d-4eb0-a663-9372614a765a
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**And here is my Arch Linux:**
```shell
Operating System: Arch Linux
            Kernel: Linux 5.4.6-arch1-1
      Architecture: x86-64
```

### And just in case:

Please answer the following questions:

- I've installed fwupd via pacman 
- I've tried rebooting numerous times
- Yes, I'm using an NVMe disk

Help would be greatly appreciated.

kopax, 2019-12-23:
I have the same issue (see https://github.com/fwupd/fwupd/issues/1669), except I could build fwupd from source using the master branch, it fails at:

```
dka@dka:[~/workspace/github.com/fwupd/fwupd (master)]: ninja -C build
ninja: Entering directory `build'
[1/388] Linking target libfwupdplugin/libfwupdplugin.so.1.0.0.
FAILED: libfwupdplugin/libfwupdplugin.so.1.0.0 
cc  -o libfwupdplugin/libfwupdplugin.so.1.0.0 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-archive.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-chunk.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-cab.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-guid.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-common-version.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-device-locker.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-device.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-dfu-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware-common.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-firmware-image.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-hwids.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-ihex-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-io-channel.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-plugin.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-smbios.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-srec-firmware.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-udev-device.c.o' 'libfwupdplugin/0205040@@fwupdplugin@sha/fu-usb-device.c.o' -Wl,--no-undefined -Wl,--as-needed -shared -fPIC -Wl,--start-group -Wl,-soname,libfwupdplugin.so.1 -Wl,-z,relro -Wl,-z,defs -Wl,-z,now libfwupd/libfwupd.so.2.0.0 -Wl,--version-script,/home/dka/workspace/github.com/fwupd/fwupd/libfwupdplugin/fwupdplugin.map /usr/lib/x86_64-linux-gnu/libxmlb.so /usr/lib/x86_64-linux-gnu/libgio-2.0.so /usr/lib/x86_64-linux-gnu/libgobject-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so /usr/lib/x86_64-linux-gnu/libgusb.so /usr/lib/x86_64-linux-gnu/libusb-1.0.so /usr/lib/x86_64-linux-gnu/libgudev-1.0.so -Wl,--export-dynamic /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so -pthread /usr/lib/x86_64-linux-gnu/libarchive.so /usr/lib/x86_64-linux-gnu/libjson-glib-1.0.so /usr/lib/x86_64-linux-gnu/libgcab-1.0.so -Wl,--end-group '-Wl,-rpath,$ORIGIN/../libfwupd' -Wl,-rpath-link,/home/dka/workspace/github.com/fwupd/fwupd/build/libfwupd  
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_convert_quirk_to_xml_cb':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:93: undefined reference to `xb_builder_source_ctx_get_bytes'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_add_quirks_for_path':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:177: undefined reference to `xb_builder_source_add_adapter'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_lookup_by_id':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:289: undefined reference to `xb_silo_query_first_full'
/usr/bin/ld: libfwupdplugin/0205040@@fwupdplugin@sha/fu-quirks.c.o: in function `fu_quirks_lookup_by_id_iter':
/home/dka/workspace/github.com/fwupd/fwupd/build/../libfwupdplugin/fu-quirks.c:351: undefined reference to `xb_silo_query_full'
collect2: error: ld returned 1 exit status
[9/388] Generating fwupdplugin.vapi with a custom command.
Soup-2.4.gir:1334.7-1334.21: warning: Virtual method `Soup.AuthDomain.accepts' conflicts with method of the same name
[10/388] Compiling C object 'libfwupdplugin/0205040@@fwupdplugin-self-test@exe/fu-plugin.c.o'.
ninja: build stopped: subcommand failed.
```

I explained all the step I did here and all were with success: https://forums.lenovo.com/t5/Other-Linux-Discussions/How-To-Configure-X1-Carbon-Gen-7-on-Debian-FingerPrint-4G-Modem/m-p/4609363/highlight/true#M14174

I suspect that I needed to build `fwupd` first to get 1.35 like your but your issue shown no difference with mine. :face_with_head_bandage:

Z4nterox, 2019-12-23:
> I have the same issue (see #1669), except I could build fwupd from source using the master branch, it fails at:
>
> ```
> dka@dka:[~/workspace/github.com/fwupd/fwupd (master)]: ninja -C build
> ninja: [...]

Luckily I never had any problem updating/building fwupd since Arch is rolling release and pacman has the newest versions most of the time.

kopax, 2019-12-23:
are you sure that having `fwupd` upgraded was necessary in order to get your Lenovo laptop having the finger print working on linux? Where have you find such info please ?

Z4nterox, 2019-12-23:
> are you sure that having `fwupd` upgraded was necessary in order to get your Lenovo laptop having the finger print working on linux? Where have you find such info please ?

Never said that, I just mentioned that updating the version was no problem for me.
And "necessary in order to get your Lenovo laptop having the finger print working on linux", well, I wouldn't know, since it's not working for me, which is why I opened this issue. 

I don't think I can help you there mate, sorry.

kopax, 2019-12-23:
No worries, after successfully upgrading I am able to use my device `06cb:00bd Synaptics, Inc`. Thanks for being helpful!

Z4nterox, 2019-12-23:
> No worries, after successfully upgrading I am able to use my device `06cb:00bd Synaptics, Inc`. By any chance, do you know any UI to use it for debian, I couldn't find one ? Thanks for being helpful!

Never used Debian before, so I don't really know. I'm using fprint on Arch, maybe there is something similar for Debian?

kopax, 2019-12-23:
I was not able to configure it with a gui but I was able with `fprintd-enroll` to register my finger, `fprintd-verify` to verify my registered fingerprint, and `libpam-printd-dev` and an edition in `/etc/pam.d/common-auth` and added at the very top:

```diff
+ auth	[success=3 default=ignore]	pam_fprintd.so max_tries=1 timeout=10
auth	[success=2 default=ignore]	pam_unix.so nullok_secure
auth	[success=1 default=ignore]	pam_ldap.so minimum_uid=10000 use_first_pass
```

This allowed me to use fingerprint for login, and bash login. 

I haven't found a way to get a fingerprint per user, only per system. I would gladly have a fingerprint for my **root** user.

So far, my fingerprint can login into anyone including root. Have you find a way to use fprintd for a per user fingerprint ?

Z4nterox, 2019-12-24:
> I was not able to configure it with a gui but I was able with `fprintd-enroll` to register my finger, `fprintd-verify` to verify my registered fingerprint, and `libpam-printd-dev` and an edition in [...]

https://jlk.fjfi.cvut.cz/arch/manpages/man/fprintd.1

Fprint is not working right now for me, because the firmware on the fingerprint reader is too old for fprint, but I can't update the firmware because fwupd can't detect the device. So, I can't really help you with fprint since I can't test it.

kopax, 2019-12-24:
That's a sick situation you're. Perhaps try to downgrade and try again if you can.

Z4nterox, 2019-12-24:
> That's a sick situation you're. Perhaps try to downgrade and try again if you can.

I've tried reinstalling fwupd a couple of times but, sure, I will give downgrading a try.

Z4nterox, 2019-12-26:
Ok, so....

Downgrading in itself doesn't seem to help. But...

After either upgrading or downgrading fwupd, the fingerprint reader does get detected. After another reboot the device is gone again, no matter what version of fwupd I'm currently on.

Output directly after downgrading (or downgrading and reupdating, both yield the same result) till the next reboot.

```bash
$ fwupdmgr get-devices
20N3S02J00
│
├─T490/T590/P53s/P43s Thunderbolt Controller:
│     Device ID:           9612645e5dbf7a4582aefe4cb3911dc6114495e8
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     Flags:               internal|updatable|require-ac|registered
│     GUID:                52265728-359a-574c-9a6a-a23d3b1f952d
│   
├─Prometheus:
│ │   Device ID:           51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
│ │   Summary:             Fingerprint reader
│ │   Current version:     10.01.3073367
│ │   Vendor:              Synaptics (USB:0x06CB)
│ │   Install Duration:    2 seconds
│ │   Flags:               updatable|registered
│ │   GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│ │                        8088f861-6318-5b1e-9ce4-fbddbedb09ac
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:         a717531e28f0288a4e7af638b8e6afc0a1c8743e
│       Current version:   0021
│       Minimum Version:   0021
│       Vendor:            Synaptics (USB:0x06CB)
│       Flags:             updatable|registered
│       GUID:              7c5a1e62-38fa-5859-9337-09dbac6377e4
│     
├─SAMSUNG MZVLB512HBJQ-000L7:
│     Device ID:           f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Summary:             NVM Express Solid State Drive
│     Current version:     3L2QEXF7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUIDs:               47335265-a509-51f7-841e-1c94911af66b
│                          f4ceb2a8-09c4-55c0-af2a-13ccd63925b1
│   
├─System Firmware:
│     Device ID:           6150dd1f7291b0709289ab8a53cc85a17e117ef2
│     Current version:     0.1.58
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                603baf73-b997-45b5-86b4-2f981a008e18
│   
├─UEFI Device Firmware:
│     Device ID:           b1c32d855d85a2124e865f0a03f8c2384dcca167
│     Current version:     192.47.1524
│     Minimum Version:     0.0.1
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                bf2d0f8b-f9a9-400c-8914-36c225d16eb4
│   
├─UEFI Device Firmware:
│     Device ID:           489f23b2ba9c1adf3e9f9f10598c98ba5c6bba39
│     Current version:     0.1.16
│     Minimum Version:     0.1.16
│     Flags:               internal|updatable|require-ac|supported|registered|needs-reboot
│     GUID:                38ea6335-29ca-417b-8cd4-6b4e5e866f92
│   
├─UEFI Device Firmware:
│     Device ID:           98a394c65d64ece8aae3126fb0f85364bb915b49
│     Current version:     1.1.16
│     Minimum Version:     0.0.1
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                25dd5a7f-4151-401f-ae35-1440807c7bba
│   
└─UEFI Device Firmware:
      Device ID:           720296b654bf23d974c282ad9cd59d0b9d33c971
      Current version:     0.0.26406
      Minimum Version:     0.0.1
      Flags:               internal|updatable|require-ac|registered|needs-reboot
      GUID:                26aac102-e03d-4eb0-a663-9372614a765
```

kopax, 2019-12-26:
Do you wait for the device to be up and running ? It take some time for mine to start up after reboot. But it comes up at the end without requiring further action. Have you tested it under windows ? Would be interesting to see if you have any hardware issue with the finger print reader.

Z4nterox, 2019-12-26:
> Do you wait for the device to be up and running ? It take some time for mine to start up after reboot. But it comes up at the end without requiring further action. Have you tested it under windows ? Would be interesting to see if you have any hardware issue with the finger print reader.

Waiting doesn't change anything sadly. And it works perfectly fine under windows.

kopax, 2019-12-27:
If you are perfectly sure it work with windows, I will start using `fwupd` v1.35 because that's the one that worked for me:

```
fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.5
```

Also, I would perhaps start from scratch on a fresh OS, I had a nonunderstandable issue after configuring wrongly my BIOS at some point, fixed it but the firmware was corrupt. Reinstalling was a pretty straight forward process and I got my WWAN card working directly.

This has nothing to do with your fingerprint reader, but I see no reason why you have such behavior.

Lastly, you should compile `fprintd` and `libfprint` from source using their `master` branch, and post the full logs and command used for building. You must have missed something, a build warning error, missing warning, etc...

Z4nterox, 2019-12-30:
Compiled libfprint and fprintd from source. Now fwupd can detect the fingerprint scanner and the fingerprint scanner itself works as well. 

Thank you so very much.

closed 2019-12-30 by Z4nterox

kopax, 2019-12-31:
I'm glad you solve it!

allmazz, 2021-01-27:
Hello. Fwupd does not see the Synaptics fingerprint sensor. Compiled fwupd 1.3.5. Tried to disable Windows UEFI Update. frpintd and libfprint are compiled by me. I am using Manjaro. Lenovo Thinkpad E14 Gen 2 laptop. The sensor works in Windows. BIOS updated. Secure Boot is also disabled. Please tell me what I can try.

hughsie, 2021-01-27:
Can you try fwupd 1.5.5 please. If that doesn't work, can you try a fedora 33 live CD? You can update fwupd 
to 1.5.5 even without installing.

allmazz, 2021-01-28:
Thanks for the answer. Initially, I had 1.5.5 installed, it did not work with it either. In Fedora the same thing, in the fwupdmgr get-devices there is no fingerprint sensor and an error when trying to install the driver.

whitefalk, 2023-03-29:
I have the same issue on a Lenovo Thinkpad T14 G2 AMD.
 fwupdmgr (v1.8.10) didn't list the Prometheus devices.

It shows up in lsusb, and fprintd works fine: