fwupd/fwupd #1671: No updates found for the Dell TB16 dock on XPS 13 9370

andrejpodzimek, 2019-12-25:
[This download page](https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverid=7tj1h) shows new firmware for the TB16 dock. Because `fwupdmgr` didn't show any updates for it, I tried to find out whether it could be updated in other ways.

The binary for download, `DellDockingStationFwUp_1.0.2_12052018_TB16.exe`, appears to be runnable under UEFI, but I'm not really sure how to make it run. I tried the boot menu and `BIOS Flash Update`, which did see the file currently stored on my `boot/efi` vfat partition, but it refused to flash it, because, IIUC, it expects only system BIOS updates, not other types of update files.

The system BIOS is at its latest version, which is great; `fwupd` has worked fine there. :+1: I've updated the internal NVME using [this approach](https://gist.github.com/klingtnet/22ab0b907e2d9d20f98c72c93ea5dd37#gistcomment-2830279) and it also worked just fine.

But I couldn't figure out how to update the dock and its "cable" (which is another computer with firmware in it), using `fwupdmgr` or otherwise. :sob:

- Operating system and version: **ArchLinux, kernel `5.4.6-arch3-1`**
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): **`pacman`**
- Have you tried rebooting? **:grinning: Sure.**
- Are you using an NVMe disk? **Yes, two of those, PCIe and Thunderbolt.**
- Is secure boot enabled (only for the UEFI plugin)? **No, I was too lazy to install `shim` or the like.**

```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

```shell
$ fwupdmgr get-devices
XPS 13 9370
│
├─Dell Thunderbolt Cable:
│     Device ID:           a4a18a4366372e8a0eed1347b1ca0185d3a23931
│     Current version:     16.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUID:                b4fd3cdf-4e3a-5090-a583-45367cfd6421 ← TBT-00d4b051
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Dell Thunderbolt Dock:
│     Device ID:           10f968084d300dbc2c93a92e57f59ad51692cf0d
│     Current version:     16.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUID:                76cc74d4-f062-5b93-a11c-8d2a58a25848 ← TBT-00d4b054
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Portable SSD X5:
│     Device ID:           f99d0eff1001b4c29940f5df8dcd903c778b723e
│     Current version:     35.00
│     Vendor:              SAMSUNG ELECTRONICS CO., LTD (TBT:0x0175)
│     GUID:                ed2454a2-0f06-5e01-9708-ff961fa07a92 ← TBT-01754000
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Thunderbolt Controller:
│     Device ID:           91cb4290ca585a69eb6c7155ca9f84ac633cb6f0
│     Souhrn:              Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a ← TBT-00d407e6
│                          31dedd69-1429-5224-87d4-a8fb8072a50e ← TBT-00d407e6-0000:05:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─KXG50ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:           f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Souhrn:              NVM Express Solid State Drive
│     Current version:     AADA4107
│     Vendor:              Toshiba Corporation (NVME:0x1179)
│     Serial Number:       88RS10J9T8KQ
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab ← NVME\VEN_1179&DEV_0116&REV_00
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9 ← NVME\VEN_1179&DEV_0116
│                          2ce5641c-516e-5c6b-908f-6279409807b5 ← KXG50ZNV1T02 NVMe TOSHIBA 1024GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung Portable SSD X5:
│     Device ID:           6ffc09023f3810cce192d24b6358877369bfc30b
│     Souhrn:              NVM Express Solid State Drive
│     Current version:     1P3QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S423NV0K800180V
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          b1913cf1-29b4-58a0-9079-893f307b2464 ← Samsung Portable SSD X5
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
├─System Firmware:
│     Device ID:           8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│     Current version:     0.1.11.1
│     Minimum Version:     0.1.11.1
│     Vendor:              Dell Inc.
│     GUID:                7ceaf7a8-0611-4480-9e30-64d8de420c7c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│     Device ID:           370e10407b1f04ade798a9f1d3e1fa57c67750c3
│     Souhrn:              Platform TPM device
│     Current version:     7.2.0.1
│     Vendor:              Dell Inc.
│     GUIDs:               cb8da68d-cd80-5f5b-8fef-038383adbb83 ← 07e6-2.0
│                          7d65b10b-bb24-552d-ade5-590b3b278188 ← DELL-TPM-2.0-NTC-NPCT
│                          6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d ← DELL-TPM-2.0-NTC-NPCT75x
│                          fe462d4a-e48f-5069-9172-47330fc5e838 ← DELL-TPM-2.0-NTC-NPCT75xrls
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
└─Touchpad:
      Device ID:           88861d85ec873ffdf484ac0b981c640781bd6937
      Current version:     1.3.2375007
      Bootloader Version:  54.0
      Vendor:              Dell Inc. (HIDRAW:0x06CB)
      GUIDs:               f4384034-9243-5334-8075-a534be913e46 ← HIDRAW\VEN_06CB&DEV_76AF&REV_00
                           424bd00e-9789-5cdf-a12a-3c81bc4676d6 ← HIDRAW\VEN_06CB&DEV_76AF
                           140f4458-951b-5bb9-85e2-879bd5b02615 ← SYNAPTICS_RMI\TM3038-003
                           b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d ← SYNAPTICS_RMI\TM3038
      Device Flags:        • Internal device
                           • Updatable
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,0002
Boot0000  Windows Boot Manager  HD(1,GPT,10c51780-d1a2-46d7-a9f1-4ef50d6b52da,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...AM...............
Boot0001* arch  HD(1,GPT,36696fe8-b87e-b742-a4dc-19a80959f9bd,0x800,0x100000)/File(\EFI\arch\grubx64.efi)
Boot0002* Linux Firmware Updater        HD(1,GPT,36696fe8-b87e-b742-a4dc-19a80959f9bd,0x800,0x100000)/File(\EFI\arch\fwupdx64.efi)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-7ceaf7a8-0611-4480-9e30-64d8de420c7c-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
/boot
├── efi
│   ├── DellDockingStationFwUp_1.0.2_12052018_TB16.exe
│   ├── EFI
│   │   ├── Dell
│   │   │   └── Bios
│   │   │       └── Recovery
│   │   │           ├── BIOS_CUR.RCV
│   │   │           └── BIOS_PRE.rcv
│   │   ├── arch
│   │   │   ├── fw
│   │   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   │   └── fwupd-7ceaf7a8-0611-4480-9e30-64d8de420c7c.cap
│   │   │   ├── fwupdx64.efi
│   │   │   └── grubx64.efi
│   │   └── archlinux
│   │       └── grubx64.efi
│   └── grub
│       ├── ...
├── grub -> efi/grub
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

20 directories, 433 files
```

hughsie, 2019-12-31:
@superm1?

wiktor-k, 2020-01-02:
@andrejpodzimek how did you get TB16 to appear on the list in `fwupdmgr get-devices`? I'm using TB16 on XPS 9350 and everything works well (including TB security through `boltctl`) but `get-devices` doesn't mention the dock or the cable (boltctl on the other hand lists them)... :thinking: maybe I should open a separate issue (I found this one wondering why I don't have updates for TB16 through fwupd, that otherwise works great).

andrejpodzimek, 2020-01-02:
@wiktor-k The TB-16 dock, IIRC, has been listed since the day I installed Arch on the machine. I didn’t have to add it explicitly or the like. It has always been shown by both `boltctl` and `fwupdmgr`. (Just the firmware updates for it have never been found.)

wiktor-k, 2020-01-02:
Ah, okay, it seems `get-devices` doesn't show updatable devices that have non-updatable parents (this seems not right...):

<details>
<summary>TB16 non-updatable hides updatable subdevices</summary>

```
$ fwupdmgr get-devices --show-all-devices
XPS 13 9350
│
├─Thunderbolt Controller:
│     Device ID:           948425057b80f0b719b9d5374aa17ed7ff2fc861
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     26.01
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               dbbe0ed9-df91-529c-ac86-8976b78f726d
│                          8e9c55fc-e381-57ae-8f74-a3e5df340bd0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Dell TB16:
│ │   Device ID:           6ed90a25ed5034d123a54684eaad5a5cd852b1f7
│ │   Summary:             A Thunderbolt™ 3 docking station
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Requires AC power

│ │   ^---- this is non-updatable and will make all leaf devices hidden when not using --show-all-devices

│ ├─Dell Thunderbolt Cable:
│ │     Device ID:         fbdcc600105a732e21b06f910e9ad78c88ee432a
│ │     Current version:   16.00
│ │     Vendor:            Dell (TBT:0x00D4)
│ │     GUID:              b4fd3cdf-4e3a-5090-a583-45367cfd6421
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Device stages updates
│ │   
│ ├─Dell Thunderbolt Dock:
│ │     Device ID:         a6d6fc68b117742d077081667731649ff05fe565
│ │     Current version:   27.00
│ │     Vendor:            Dell (TBT:0x00D4)
│ │     GUID:              76cc74d4-f062-5b93-a11c-8d2a58a25848
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Device stages updates
...
```

</details>

closed 2020-01-06 by hughsie

mightymos, 2024-08-30:
For those coming here by way of search, the TB16 (at least the MST firmware) can be updated to the latest version 3.12.002 using the CAB file located here using fwupdmgr install or possibly local-install option:
https://fwupd.org/lvfs/devices/com.dell.mst0a52c8c7.firmware

andrejpodzimek, 2024-09-12:
> For those coming here by way of search, the TB16 (at least the MST firmware) can be updated to the latest version 3.12.002 using the CAB file located here using fwupdmgr install or possibly local-install option: https://fwupd.org/lvfs/devices/com.dell.mst0a52c8c7.firmware

This↑ yields _No supported devices found_, so it would have to be brute-forced somehow. Probably not a good idea.

mightymos, 2024-09-12:
> > For those coming here by way of search, the TB16 (at least the MST firmware) can be updated to the latest version 3.12.002 using the CAB file located here using fwupdmgr install or possibly local-install option: https://fwupd.org/lvfs/devices/com.dell.mst0a52c8c7.firmware
> 
> This↑ yields _No supported devices found_, so it would have to be brute-forced somehow. Probably not a good idea.
> 
> EDIT: No, it cannot be forced. Even when you look up the dock’s device ID from `report-devices` and use `--force` on `local-install` with the device ID, it yields the same error message.
> 
> So this is **not** working, unresolved and should be reopened.

Oh no, okay, well here's my console output at least:
```
ubuntu@ubuntu:~/Downloads$ sudo fwupdmgr get-devices --show-all
Dell Inc. Latitude 5530
│
├─Unknown Device:
│     Device ID:          270cc74de90fd845bd7956e61af39c8188856abf
│     Serial Number:      3CM046171Z  
│     GUID:               69b84d5b-abb1-501e-9d80-8ff62c94dbd7 ← DRM\VEN_HPN&DEV_366C
│  
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│  
├─12th Gen Intel Core™ i7-1265U:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000434
│     Vendor:             Intel
│     GUIDs:              ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│                         d49f8285-de61-572f-ae6d-b83795576416 ← CPUID\PRO_0&FAM_06&MOD_9A&STP_4
│     Device Flags:       • Internal device
│  
├─Alder Lake-UP3 GT2 [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              7fc247b0-a7a8-55a7-b29c-ffb9725436dc ← PCI\VEN_8086&DEV_46A8
│                         9f659b05-72bd-5bb6-9605-5e05d5f71906 ← PCI\VEN_8086&DEV_46A8&SUBSYS_10280B06
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│  
├─BC711 NVMe SK hynix 1TB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    41002131
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      FJB7N689511907S66
│     GUIDs:              b8a120c5-b2af-540f-905e-b6136d3920a7 ← STORAGE-DELL-110002
│                         0a413c20-88f2-11ea-bc55-0242ac130003
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│  
├─Dell WD15/TB16/TB18 wired Dock:
│     Device ID:          d762543f8c20f636e6fff031a000078d3e10c600
│     Summary:            Multi-stream transport device
│     Current version:    3.12.02
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB, DRM_DP_AUX_DEV:0x8086)
│     GUIDs:              87f8cd71-1e6b-5622-94e9-0e9d11bdde33 ← MST-274
│                         c1e81f96-234e-597d-b9cb-6477d0652868 ← MST-wd15-274
│                         c0f73159-4fc0-509b-80f8-5f871f3751a8 ← MST-wd15-vmm3330-274
│                         c1e2bea5-4cf2-5f84-b965-533e8e5fea45 ← MST-tb16-274
│                         be025b25-ca5c-546c-97c6-ee2160ba489d ← MST-tb16-vmm3330-274
│                         82460515-30d3-5cdf-be4c-571340842a10 ← MST-tb18-274
│                         190e8d7c-6bcf-5fab-94e7-4f8739eb90aa ← MST-tb18-vmm3330-274
│                         6ca2b423-3cc7-509a-89fd-3329e244e11f ← MST-leaf-vmm3330-274
│                         4e4e06b6-8217-5cf0-8eff-ce5e987d531a ← MST-leaf-274
│     Device Flags:       • Updatable
[... 145 lines cut ...]
│     GUID:               0a413c20-88f2-11ea-bc55-0242ac130003
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│  
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            Dell Inc. Platform Key
│     Vendor:             Dell Inc.
│     GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872
│  
└─USB4 Retimer:
      Device ID:          8f92f06a1ff9413dbdd56b2337683e333ab56574
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    10.00
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ← THUNDERBOLT\VEN_8087&DEV_15EE
                          fd6822bc-9ef9-578e-8cd5-b322fe4e64ef ← TBT-808715ee-retimer0-0:3.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
```

andrejpodzimek, 2024-09-12:
<details>
<summary><code>fwupdmgr --version</code></summary>

```
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.24
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.10.8-arch1-1
runtime   org.freedesktop.fwupd         1.9.24
```
</details>

<details>
<summary><code>fwupdmgr get-devices --show-all</code></summary>

```
Dell Inc. XPS 13 9370
│
├─Unknown Device:
│     Device ID:          533f8dcc55554ba0d48582a72ece77dc0dd59d69
│     GUID:               c35c2451-ada5-556c-93eb-368ad00355a4 ← GPIO\ID_INT344B:00
│   
├─8XDHY:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               c0de1d13-a2d9-5f69-9935-55c7fd7c070b ← DRM\VEN_SHP&DEV_148B
│     Device Flags:       • Internal device
│   
├─Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f6
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─KXG50ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:          f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Summary:            NVM Express solid state drive
│     Current version:    AADA4107
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      88RS10J9T8KQ
│     GUIDs:              4d0aed03-a30c-52c6-99e7-a8977797c3d9 ← NVME\VEN_1179&DEV_0116
│                         23e442d8-2af4-5c60-81f5-ac41aaee72c7 ← NVME\VEN_1179&DEV_0116&SUBSYS_11790001
│                         2ce5641c-516e-5c6b-908f-6279409807b5 ← KXG50ZNV1T02 NVMe TOSHIBA 1024GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          274024292ef3688eea2d2351d9b1e2de259e9c9c
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    1.21.0
│ │   Minimum Version:    1.21.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUID:               7ceaf7a8-0611-4480-9e30-64d8de420c7c
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
[... 93 lines cut ...]
      GUIDs:              8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
                          9526268c-ccab-5ef4-9e81-d97c01988504 ← PCI\VEN_8086&DEV_5917&SUBSYS_102807E6
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
```
</details>

It may be time to ditch this dock, because the manufacturer shows no support for the free desktop and the dock has never worked too well. A firmware upgrade may solve some of its issues, but the dock is still unsupported by `fwupdmgr` (most likely by Dell’s fault). [ArchWiki confirms](https://wiki.archlinux.org/title/Dell_TB16#Firmware_Updates) that updates of the device are **infeasible** on Linux, so whatever `--force`d flashes could possibly do (if they worked, which they don’t) would yield an inconsistent state with a mixture of possibly incompatible firmware versions in the 3 devices in the dock.

Sadly, Dell misinforms customers on their download page, which has just cost me an hour of time. Apparently, there is [a January 3, 2019 version](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=7tj1h&oscode=w732&productcode=dell-thunderbolt-dock-tb16&lwp=rt) of the firmware with **critical** fixes, which should be applied ASAP. Let’s try to apply it then:

* The downloaded file, `DellDockingStationFwUp_1.0.2_12052018_TB16.exe`, is not in a format known to `fwupdmgr`.
* The download page claims (**incorrectly**): *“This file format consists of a BIOS executable file. The Universal (Windows/MS DOS) format can be used to install from any Windows or MS DOS environment.”* **No. Just NO:**
  * There is no such thing like a *BIOS executable file*. BIOS doesn’t execute (or know about) files or file systems.
  * The file is **not** a valid UEFI binary. UEFI cannot boot it and won’t recognize it.
  * There is **nothing** MS DOS-related in that file. The download page contains invalid statements from decades ago.
* I first tried to boot the file as a UEFI binary, which didn’t work.
* I booted [FreeDOS](https://www.freedos.org/download/) — which comes with disabling SecureBoot (which wipes all custom SecureBoot keys without warning, a nice Dell “feature”), enabling all legacy nonsense etc. — only to find out that it doesn’t work, because: ***This program cannot be run in DOS mode.***

This is incredibly, unnecessarily annoying. What should I do?

* Can this bug be reopened, please, or should I file a new one?
* Are any people from Dell following these reports?
* Is there a Dell-specific bug tracker where I could file something?
* What is preventing Dell from surfacing [this update](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=7tj1h&oscode=w732&productcode=dell-thunderbolt-dock-tb16&lwp=rt) so that `fwupdmgr` can just automatically find it?

andrejpodzimek, 2024-09-12:
By the way: [This bug](https://github.com/fwupd/firmware-dell/issues/1) claims that the firmware for the dock is in “testing” (whatever that means, I assume `lvfs-testing`), but that is simply not the case.

andrejpodzimek, 2024-09-12:
Another piece of context: At some point this [got stuck on **one** of the three firmware updates](https://github.com/fwupd/fwupd/issues/1351) that need to happen (preferably) all at once. The issue has never been fixed, it seems. In any case, both the dock and the archaic active cable are listed as unsupported by `fwupdmgr` to this day.

mightymos, 2024-10-10:
I'm not sure if this helps but I've collected more output from a second TB16 dock.
To add to the confusion...I am having to use a Samsung Galaxy 360 to connect to the dock.

I notice that the entry for the Dell WD15/TB16/TB18 wired Dock does not appear in the output for the second dock when using the Samsung. I am not sure if this is expected or unexpected. However, some of the GUIDS for that entry match the CAB file details previously posted. So that seems relevant.

I'll post a blog entry next that contains yet more GUIDs.

```
SAMSUNG ELECTRONICS CO., LTD. 730QFG
│
├─ELAN0B00:00 04F3:324D:
│     Device ID:          752ad79c0332b36b36f73b6fd9ad4fc44c6717f7
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0000
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              826c410a-922d-5095-bf94-b58e28cfeb03 ← HIDRAW\VEN_04F3&DEV_324D
│                         a124d2b9-a314-5992-81dd-66fa3dac38e8 ← HIDRAW\VEN_04F3&DEV_324D&MOD_0066
│                         6b681645-897e-5644-8f38-650c9f416d14 ← ELANTP\ICTYPE_0D&MOD_0066
│                         a9208d1a-5b24-594c-a729-29d5ae3fcb84 ← ELANTP\ICTYPE_0D&MOD_0066&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─13th Gen Intel Core™ i7-1360P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00004121
│     Vendor:             Intel
│     GUIDs:              c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
│                         557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
│     Device Flags:       • Internal device
│   
├─Raptor Lake-P [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              0e8b1f1a-8c2b-53b6-89d7-d1a05dbc039d ← PCI\VEN_8086&DEV_A7A0
│                         7b1121bf-9e76-5eb7-b2df-4779fbdbdbd6 ← PCI\VEN_8086&DEV_A7A0&SUBSYS_144DCA06
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─System Firmware:
│ │   Device ID:          1e7ca58ef4cd60c33f086e231b42fa37a51ce51b
│ │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
│ │   Current version:    539
│ │   Minimum Version:    539
│ │   Vendor:             SAMSUNG ELECTRONICS CO., LTD. (DMI:American Megatrends International, LLC.)
│ │   Update State:       Success
│ │   GUID:               bd03dc7d-37cf-4a69-978a-ef672f03c710
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ ├─BootGuard Configuration:
│ │     Device ID:        b0d4430dfa6bde9f0c22680df36dbc8c15c80753
│ │     Current version:  01
│ │     Vendor:           Intel Corporation (MEI:0x8086)
│ │     GUIDs:            dd17041c-09ea-4b17-a271-5b989867ec65
│ │                       fccad2fe-62ae-5879-b7a9-4ead7bce50f4 ← MEI\VEN_8086&DEV_51E0
│ │                       5dc40fbc-dd7d-52d6-bb0e-9f126cc71793 ← MEI\VEN_8086&DEV_51E0&SUBSYS_144DCA06
│ │     Device Flags:     • Internal device
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
[... 69 lines cut ...]
│     Current version:    10.00
│     Vendor:             Intel (THUNDERBOLT:0x8087)
│     GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ← THUNDERBOLT\VEN_8087&DEV_15EE
│                         6d71ac25-b368-5e7b-ac62-32df365fa640 ← TBT-808715ee-retimer0-0:1.1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Device stages updates
│   
└─WDSN740-SDDPNQD-512G-1004:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    73100004
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      2326MR400363
      GUIDs:              1524d43d-ed91-5130-8cb6-8b8478508bae ← NVME\VEN_15B7&DEV_5017
                          87cfda90-ce08-52c3-9bb5-0e0718b7e57e ← NVME\VEN_15B7&DEV_5017&SUBSYS_15B75017
                          838fb1b6-8c51-58bf-bc81-60fbb2678971 ← WDSN740-SDDPNQD-512G-1004
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    

```

mightymos, 2024-10-10:
Here is more output:
https://whatdoineed2do.blogspot.com/2021/12/dell-tb16-docking-station-with-dell.html

```
├─Thunderbolt Cable:
│     Device ID:          2315cbb258f43caf4677117e8dfbb6ce68f60f88
│     Current version:    16.00
│     Vendor:             Dell (THUNDERBOLT:0x00D4, TBT:0x00D4)
│     GUIDs:              99102381-23e8-5ff5-9767-8bcda2aaa864 ← THUNDERBOLT\VEN_00D4&DEV_B051&REV_00
│                         6634407c-6706-5fe9-907e-37efcbc8098a ← THUNDERBOLT\VEN_00D4&DEV_B051
│                         b4fd3cdf-4e3a-5090-a583-45367cfd6421 ← TBT-00d4b051
│                         8564922d-2c7a-5169-9cff-d3e73f0bd807 ← TBT-00d4b051-controller0-1
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─Thunderbolt Dock:
│     Device ID:          c9f174d381c66aab3dea447decfe5df418a2d22f
│     Current version:    16.00
│     Vendor:             Dell (THUNDERBOLT:0x00D4, TBT:0x00D4)
│     GUIDs:              8e801c01-c7bf-5de2-85e3-185b2afe3b10 ← THUNDERBOLT\VEN_00D4&DEV_B054&REV_00
│                         1fa96dfa-7407-50b2-87d4-4ae4351c3867 ← THUNDERBOLT\VEN_00D4&DEV_B054
│                         76cc74d4-f062-5b93-a11c-8d2a58a25848 ← TBT-00d4b054
│                         f5a71973-58f2-5638-9c9a-d9c7538d6772 ← TBT-00d4b054-controller0-301
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
```

mightymos, 2024-10-10:
So I've confirmed that the DellDockingStationFwUp_1.0.5_TB16_20210716 file contains at least the same EEPROM file that can be viewed by extracting the CAB file. This can be confirmed by extracted the [0] file contained in the exe, then running the following command to copy bytes from the 9D210 file produced. I was able to find the file initially by searching for the string "Non-PnP" which is near the file beginning:
```
$ dd skip=2651384 count=65486 if=9D210 of=candidate.eeprom bs=1
```
This should be the checksum calculated for the file:
SHA-512: 28093D2D4F049C99A6C0667669CFFF472E5B77D380907FACBC31219D2E6790D91996590739DE24940A2C0CE4B68F28A7713AB3DE0E45294869AD12DB53B5A36F

I realize this does not help exactly.
However, it demonstrates that the cab is apparently the latest version denoted 1.0.5 by Dell.

I realize it's a questionable use of time but if all else failed it would seem possible to manually flash chips with a socket.
I'm sure there's some reason this won't work for some chips...but anyway.

superm1, 2024-10-10:
The WD15/TB16 can only be fully updated from a Dell system.  It uses a proprietary mechanism to do it.  For a period of time fwupd had the ability to trigger this mechanism, but it has since been removed since no updates were to be continued for these docks.

mightymos, 2024-10-10:
> The WD15/TB16 can only be fully updated from a Dell system. It uses a proprietary mechanism to do it. For a period of time fwupd had the ability to trigger this mechanism, but it has since been removed since no updates were to be continued for these docks.

Ahh okay, I see it here now:
https://github.com/fwupd/fwupd/commit/952edaaf9ab2873530343bd0a911b2c0fabf7dc6

It's a real shame they make things so cumbersome for developers.
Thanks for the further explanation and other discussion in related issues, I learned a little bit in the process.

superm1, 2024-10-10:
> Ahh okay, I see it here now:
> https://github.com/fwupd/fwupd/commit/952edaaf9ab2873530343bd0a911b2c0fabf7dc6
>
> It's a real shame they make things so cumbersome for developers.
> Thanks for the further [...]

Yeah fwiw the follow on docks all are very well supported and can be updated on any device Dell or not.

mightymos, 2024-10-10:
> > > The WD15/TB16 can only be fully updated from a Dell system. It uses a proprietary mechanism to do it. For a period of time fwupd had the ability to trigger this mechanism, but it has since been removed since no updates were to be continued for these docks.
> > 
> > 
> > Ahh okay, I see it here now:
> > [952edaa](https://github.com/fwupd/fwupd/commit/952edaaf9ab2873530343bd0a911b2c0fabf7dc6)
> > It's a real shame they make things so cumbersome for developers.
> > Thanks for the further explanation and other discussion in related issues, I learned a little bit in the process.
> 
> Yeah fwiw the follow on docks all are very well supported and can be updated on any device Dell or not.

Great, progress. Thanks for the advice.

flhofer, 2024-11-11:
Hi! I just created a repo with updated firmware for TB16 (probably also WD15 and TB18) and some officially unavailable bins. I have 3 Docks and successfully flashed them.

Spend hours to figure out especially the Wake-up issue with the USB controller. Issue now disappeared.

[Repo](https://github.com/flhofer/dell_TB16_firmware)