fwupd/fwupd #1672: Update missed for the Thunderbolt controller on XPS 13 9370

andrejpodzimek, 2019-12-25:
Dell shows [an update](https://www.dell.com/support/home/us/en/04/drivers/driversdetails?driverid=2670n&productcode=xps-13-9370-laptop) for the Thunderbolt 3 controller on XPS 9370, version `4.40.24.002, A03`, but my system appears to be stuck at the previous version, `4.33.18.004, A02`. (At least I assume that's what the `33.00` in the listing means.) Of course I've tried `refresh` and all, but the new version is nowhere to be found.

- Operating system and version: **ArchLinux, kernel `5.4.6-arch3-1`**
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): **`pacman`**
- Have you tried rebooting? **:grinning: Sure.**
- Are you using an NVMe disk? **Yes, two of those, PCIe and Thunderbolt.**
- Is secure boot enabled (only for the UEFI plugin)? **No, I was too lazy to install `shim` or the like.**

```shell
$ fwupdmgr --version
client version: 1.3.5
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.5
```

```shell
$ fwupdmgr get-devices
XPS 13 9370
│
├─Dell Thunderbolt Cable:
│     Device ID:           a4a18a4366372e8a0eed1347b1ca0185d3a23931
│     Current version:     16.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUID:                b4fd3cdf-4e3a-5090-a583-45367cfd6421 ← TBT-00d4b051
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Dell Thunderbolt Dock:
│     Device ID:           10f968084d300dbc2c93a92e57f59ad51692cf0d
│     Current version:     16.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUID:                76cc74d4-f062-5b93-a11c-8d2a58a25848 ← TBT-00d4b054
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Portable SSD X5:
│     Device ID:           f99d0eff1001b4c29940f5df8dcd903c778b723e
│     Current version:     35.00
│     Vendor:              SAMSUNG ELECTRONICS CO., LTD (TBT:0x0175)
│     GUID:                ed2454a2-0f06-5e01-9708-ff961fa07a92 ← TBT-01754000
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─Thunderbolt Controller:
│     Device ID:           91cb4290ca585a69eb6c7155ca9f84ac633cb6f0
│     Souhrn:              Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a ← TBT-00d407e6
│                          31dedd69-1429-5224-87d4-a8fb8072a50e ← TBT-00d407e6-0000:05:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─KXG50ZNV1T02 NVMe TOSHIBA 1024GB:
│     Device ID:           f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
│     Souhrn:              NVM Express Solid State Drive
│     Current version:     AADA4107
│     Vendor:              Toshiba Corporation (NVME:0x1179)
│     Serial Number:       88RS10J9T8KQ
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab ← NVME\VEN_1179&DEV_0116&REV_00
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9 ← NVME\VEN_1179&DEV_0116
│                          2ce5641c-516e-5c6b-908f-6279409807b5 ← KXG50ZNV1T02 NVMe TOSHIBA 1024GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Samsung Portable SSD X5:
│     Device ID:           6ffc09023f3810cce192d24b6358877369bfc30b
│     Souhrn:              NVM Express Solid State Drive
│     Current version:     1P3QEXE7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S423NV0K800180V
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          b1913cf1-29b4-58a0-9079-893f307b2464 ← Samsung Portable SSD X5
│     Device Flags:        • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
├─System Firmware:
│     Device ID:           8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
│     Current version:     0.1.11.1
│     Minimum Version:     0.1.11.1
│     Vendor:              Dell Inc.
│     GUID:                7ceaf7a8-0611-4480-9e30-64d8de420c7c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│     Device ID:           370e10407b1f04ade798a9f1d3e1fa57c67750c3
│     Souhrn:              Platform TPM device
│     Current version:     7.2.0.1
│     Vendor:              Dell Inc.
│     GUIDs:               cb8da68d-cd80-5f5b-8fef-038383adbb83 ← 07e6-2.0
│                          7d65b10b-bb24-552d-ade5-590b3b278188 ← DELL-TPM-2.0-NTC-NPCT
│                          6f5ddd3a-8339-5b2a-b9a6-cf3b92f6c86d ← DELL-TPM-2.0-NTC-NPCT75x
│                          fe462d4a-e48f-5069-9172-47330fc5e838 ← DELL-TPM-2.0-NTC-NPCT75xrls
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
└─Touchpad:
      Device ID:           88861d85ec873ffdf484ac0b981c640781bd6937
      Current version:     1.3.2375007
      Bootloader Version:  54.0
      Vendor:              Dell Inc. (HIDRAW:0x06CB)
      GUIDs:               f4384034-9243-5334-8075-a534be913e46 ← HIDRAW\VEN_06CB&DEV_76AF&REV_00
                           424bd00e-9789-5cdf-a12a-3c81bc4676d6 ← HIDRAW\VEN_06CB&DEV_76AF
                           140f4458-951b-5bb9-85e2-879bd5b02615 ← SYNAPTICS_RMI\TM3038-003
                           b29d3c85-cd0e-503e-9c7e-f6731c1eaf2d ← SYNAPTICS_RMI\TM3038
      Device Flags:        • Internal device
                           • Updatable
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,0002
Boot0000  Windows Boot Manager  HD(1,GPT,10c51780-d1a2-46d7-a9f1-4ef50d6b52da,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...AM...............
Boot0001* arch  HD(1,GPT,36696fe8-b87e-b742-a4dc-19a80959f9bd,0x800,0x100000)/File(\EFI\arch\grubx64.efi)
Boot0002* Linux Firmware Updater        HD(1,GPT,36696fe8-b87e-b742-a4dc-19a80959f9bd,0x800,0x100000)/File(\EFI\arch\fwupdx64.efi)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-7ceaf7a8-0611-4480-9e30-64d8de420c7c-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
/boot
├── efi
│   ├── DellDockingStationFwUp_1.0.2_12052018_TB16.exe
│   ├── EFI
│   │   ├── Dell
│   │   │   └── Bios
│   │   │       └── Recovery
│   │   │           ├── BIOS_CUR.RCV
│   │   │           └── BIOS_PRE.rcv
│   │   ├── arch
│   │   │   ├── fw
│   │   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   │   └── fwupd-7ceaf7a8-0611-4480-9e30-64d8de420c7c.cap
│   │   │   ├── fwupdx64.efi
│   │   │   └── grubx64.efi
│   │   └── archlinux
│   │       └── grubx64.efi
│   └── grub
│       ├── ...
├── grub -> efi/grub
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

20 directories, 433 files
```

superm1, 2020-01-06:
The 33 in the listing doesn't necessarily represent the NVM version number - no.  Confusingly the versioning system on support.dell.com is versioning for the executable.  It's entirely possible that no NVM change happens between executables.

superm1, 2020-01-08:
NVM 40 is in the 'testing' channel for LVFS as listed here: https://fwupd.org/lvfs/devices/com.dell.tbt4eeb9d07.firmware

You can enable the testing channel to install it and try it out.

closed 2020-02-14 by stale[bot]