fwupd/fwupd #1674: Version string in the result of get-details is changed after 1.0.9

ltmvic51, 2019-12-27:
In Ubuntu Bionic, I downloaded a cab file from https://fwupd.org/downloads/d7baa890b5d2642ebc2eb40a7cb8f990b1f6e42c-Bandon_1.4.1.cab by `fwupdmgr get-releaess`, using fwupdmgr get-details xxx.cab command and found different version string in the result with different fwupd version.
fwupd 1.0.9: "Version:              0.1.4.1"
fwupd 1.2.5: "Version:              0.1.1025"

Please refer to following for full log:
  --------
ubuntu@Latitude-5300:~/Downloads$ sudo fwupdmgr get-details d7baa890b5d2642ebc2eb40a7cb8f990b1f6e42c-Bandon_1.4.1.cab 
Decompressing…           [***************************************]
Latitude 5300 System Firmware
  DeviceId:             b03588a447340830af90b63cd5830a54455504fb
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  
  [Release]
  AppstreamId:          com.dell.uefi93634829.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell Latitude 5X00
  Version:              0.1.1025
  Checksum:             SHA1(aa6bcf22339c449ac5ce85eb9478373ebb689c63)
  License:              proprietary
  Size:                 21.1 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  TrustFlags:           none

ubuntu@Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.5

//=================================================================================================

ubuntu@Latitude-5300:~/Downloads$ sudo fwupdmgr get-details d7baa890b5d2642ebc2eb40a7cb8f990b1f6e42c-Bandon_1.4.1.cab 
Decompressing…         [***************************************]
Latitude 5300 System Firmware
  DeviceId:             b03588a447340830af90b63cd5830a54455504fb
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  
  [Release]
  AppstreamId:          com.dell.uefi93634829.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell Latitude 5X00
  Version:              0.1.4.1
  Filename:             firmware.bin
  License:              proprietary
  Size:                 21.1 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  TrustFlags:           payload

ubuntu@Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

hughsie, 2019-12-28:
If you try this with fwupd >= 1.3.5 do you get the expected version?

ltmvic51, 2020-01-03:
```
u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr get-details ffbf931509f73fd26b410ff8358b51f3eaa20e7f-firmware.cab 
Decompressing…         [***************************************]
Latitude 5300 System Firmware
  DeviceId:             b03588a447340830af90b63cd5830a54455504fb
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  
  [Release]
  AppstreamId:          com.dell.uefi93634829.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell Latitude 5X00
  Version:              0.1.5.0
  Filename:             firmware.bin
  License:              proprietary
  Size:                 21.1 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  Flags:                trusted-payload

u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.5
u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr get-details ffbf931509f73fd26b410ff8358b51f3eaa20e7f-firmware.cab 
Decompressing…           [***************************************]
Latitude 5300 System Firmware
  DeviceId:             b03588a447340830af90b63cd5830a54455504fb
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  
  [Release]
  AppstreamId:          com.dell.uefi93634829.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell Latitude 5X00
  Version:              0.1.1280
  Checksum:             SHA1(51b9269d0dc93635610f86ce149a7103bc6e5b68)
  License:              proprietary
  Size:                 21.1 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  Flags:                none

u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.10
u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr get-details ffbf931509f73fd26b410ff8358b51f3eaa20e7f-firmware.cab 
Decompressing…           [***************************************]
Latitude 5300 System Firmware
  DeviceId:             7a786fb720c6983f8813ab110f17a5e50393c75e
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  VersionFormat:        quad
  
  [Release]
  AppstreamId:          com.dell.uefi93634829.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell Latitude 5X00
  Version:              0.1.5.0
  Checksum:             SHA1(51b9269d0dc93635610f86ce149a7103bc6e5b68)
  License:              proprietary
  Size:                 21.1 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  Flags:                none

u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.5
u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr get-details ffbf931509f73fd26b410ff8358b51f3eaa20e7f-firmware.cab 
Decompressing…           [***************************************]
Latitude 5300
│
└─System Firmware:
  │   Device ID:           7a786fb720c6983f8813ab110f17a5e50393c75e
  │   Description:         Updating the system firmware improves performance.
  │   GUID:                93634829-10b8-4098-a617-d52d056a4dd8
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─Latitude 5X00 System Update:
        New version:       0.1.5.0
        Remote ID:         lvfs
        Summary:           Firmware for the Dell Latitude 5X00
        License:           Proprietary
        Size:              21.1 MB
        Vendor:            Dell Inc.
      

```

ltmvic51, 2020-01-03:
Tried fwupd 1.3.5, the version string of get-details reports correct string now.
Close this issue.

closed 2020-01-03 by ltmvic51