fwupd/fwupd #1675: Without --allow-reinstall the same version of BIOS could still be updated by fwupd

ltmvic51, 2019-12-27:
In Ubuntu 18.04, I found the same version of BIOS could still be updated by `sudo  fwupdmgr install xxx.cab` without --allow-reinstall.

fwupd 1.0.9 (in -update channel):
   BIOS update would be blocked by fwupdmgr without --allow-reinstall.
fwupd 1.2.5:
   BIOS update would NOT be blocked by fwupdmgr without --allow-reinstall.
fwupd 1.2.10 (in -proposed channel):
   BIOS update would NOT be blocked by fwupdmgr without --allow-reinstall.
fwupd 1.3.5:
   BIOS update would NOT be blocked by fwupdmgr without --allow-reinstall.

u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr install d7baa890b5d2642ebc2eb40a7cb8f990b1f6e42c-Bandon_1.4.1.cab 
Decompressing…         [***************************************]
Specified firmware is already installed '0.1.4.1'
u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

//=================================================================================================
u@u-Latitude-5300:~/Downloads$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.2.5
u@u-Latitude-5300:~/Downloads$ sudo fwupdmgr install d7baa890b5d2642ebc2eb40a7cb8f990b1f6e42c-Bandon_1.4.1.cab 
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Latitude 5300 System Firmware…                     ]
Scheduling…              [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]:

hughsie, 2020-01-02:
Can you attach the output of `fwupdmgr get-devices --verbose` please.

ltmvic51, 2020-01-03:
`u@u-Latitude-5300:~$ sudo fwupdmgr --version
[sudo] password for u: 
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.5
u@u-Latitude-5300:~$ sudo fwupdmgr get-devices --verbose
(fwupdmgr:3315): FuMain-DEBUG: 10:49:28.043: PC401 NVMe SK hynix 512GB
  DeviceId:             3743975ad7f64f8d6575a9ae49fb3a8856fe186f
  Guid:                 12c8974c-99cd-5f9a-b310-c28d565a47f8 <- NVME\VEN_1C5C&DEV_1527&REV_00
  Guid:                 2260c743-963b-566e-93f1-8223c3e6b87b <- NVME\VEN_1C5C&DEV_1527
  Guid:                 c28fdadd-607d-578c-b6ed-60ac49682904 <- PC401 NVMe SK hynix 512GB
  Serial:               MI88N047510608S6M
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  Vendor:               Intel Corporation
  VendorId:             NVME:0x1C5C
  Version:              80007E00
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-01-03

(fwupdmgr:3315): FuMain-DEBUG: 10:49:28.043: System Firmware
  DeviceId:             b03588a447340830af90b63cd5830a54455504fb
  Guid:                 93634829-10b8-4098-a617-d52d056a4dd8
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|can-verify|usable-during-update
  Checksum:             SHA1(b922e3f0b7ed52b24bf0b121d1ac42478ca4bb)
  Checksum:             SHA256(36d565a688137160451ee25450afcaed7ad1c90185abde5baf440f403c33283c)
  Vendor:               Dell Inc.
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2020-01-03

(fwupdmgr:3315): FuMain-DEBUG: 10:49:28.043: TPM 2.0
  DeviceId:             f205eb1376813f00b3e909a97c54cbbcd4199369
  Guid:                 670b5381-3fd0-5653-8610-f446d2bedd10 <- 08b7-2.0
  Guid:                 73730635-f6c2-53da-9df2-948bb5ac1022 <- DELL-TPM-2.0-STM-
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Version:              0.74.0.8
  VersionFormat:        quad
  Icon:                 computer
  Created:              2020-01-03
  UpdateError:          Updating disabled due to TPM ownership

(fwupdmgr:3315): FuMain-DEBUG: 10:49:28.047: lvfs has no RemoteURI
(fwupdmgr:3315): FuMain-DEBUG: 10:49:28.047: no unreported devices
`

hughsie, 2020-01-06:
Version:              0.1.5.0
    VersionLowest:        0.1.5.0
    VersionFormat:        quad

I did expect to see "VersionFormat: dell" there for 1.3.5... Are you using the snap to get that?

ltmvic51, 2020-01-06:
I installed fwupd from apt.

hughsie, 2020-01-06:
> I did expect to see "VersionFormat: dell" there for 1.3.5

Urgh, dell-bios was added in 1.3.6 --- could you install 1.3.6 and see if the issue is fixed there? Thanks.

superm1, 2020-01-06:
If it helps, 1.3.6 is in the snap stable channel now and once an Ubuntu archive administrator releases it, it will be in the proposed channel for Ubuntu focal (https://launchpad.net/ubuntu/+source/fwupd/1.3.6-1)

superm1, 2020-01-06:
@hughsie is https://github.com/fwupd/fwupd/commit/1fbcb1a1808adebbe80a9c45f156bb695b02fb43 enough to backport to 1_2_X for this you think?

hughsie, 2020-01-06:
Not sure; I hadn't thought too much about the non-legacy backport. I think you'd need quite a few more if you want the default per device verfmt on the standalone cab install.

superm1, 2020-01-07:
I've filed a backport here: https://github.com/fwupd/fwupd/pull/1694.  That seems like it should be enough for me from the quick testing I did. ^ @ltmvic51

ycheng, 2020-01-08:
with #1694 applied to 1.2.10, build test failed on "3/15 fu-self-test"

check the below link for detail.

https://launchpad.net/~ycheng-twn/+archive/ubuntu/fwupd-1.2.10-same-version-install/+build/18530288

superm1, 2020-01-08:
Can you check if functionality is okay though? I just didn't backport the self test fix. I'll do that tomorrow.

ltmvic51, 2020-01-08:
I verified fwupd 1.3.6, it can check the version in BIOS capsule and block BIOS update accordingly now.

```
ubuntu@Latitude-5300:~/Downloads$ fwupd.fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.6
ubuntu@Latitude-5300:~/Downloads$ sudo fwupd.fwupdmgr install ffbf931509f73fd26b410ff8358b51f3eaa20e7f-firmware.cab 
Decompressing…           [***************************************]
Specified firmware is already installed '1.5.0'

```

ycheng, 2020-01-08:
Sorry not manage to test patch today. Will do tomorrow.

superm1, 2020-01-08:
#1694 was merged to the 1_2_X branch to backport this fix from 1.3.6.

closed 2020-01-08 by superm1

ycheng, 2020-01-10:
for the record, test passed with the first three patch on merged #1694 applied to 1.2.10 on ubuntu bionic.

superm1, 2020-01-10:
@ycheng you might consider to also include https://github.com/fwupd/fwupd/commit/1f579d82c328b7eac1b95d9e7ca2dfbde79dca77 and 
https://github.com/fwupd/fwupd/commit/a5df5b0e20726985cc971bfdb6f3872cf387f258

for fixing SRU https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1858590 as well as the cleanup behavior that you observed.