fwupd/fwupd #1676: Thunderbolt Controller fails to update

taaem, 2019-12-27:
When updating the Thunderbolt Controller on my Thinkpad X1 Yoga 3rd gen, the update fails:

```shell
$ fwupdmgr update
Upgrade available for Thunderbolt Controller from 33.00 to 43.00
Thunderbolt Controller and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading 43.00 for Thunderbolt Controller...
Entpacken …              [***************************************]
Authentifizierung …      [***************************************]
Thunderbolt Controller wird aktualisiert …                       ]
Gerät wird neu gestartet …[***************************************]failed to update_attach using thunderbolt: update failed (status 6)
``` 

```shell
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5

```

```shell
$ fwupdmgr get-devices
20LES01W00
│
├─Thunderbolt Controller:
│     Device ID:           65c2ab4fbfdd92d531f7b604a22629ab9ae13f66
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     33.00
│     Vendor:              LENOVO (TBT:0x0109)
│     GUIDs:               10bc8610-a3cd-5e74-a187-8d01f3ff345d
│                          fb595a4f-7e77-5030-a518-e4494392378d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─KXG5AZNV512G TOSHIBA:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Zusammenfassung:     NVM Express Solid State Drive
│     Current version:     5106AALA
│     Vendor:              Toshiba America Info Systems (NVME:0x1179)
│     GUIDs:               af8d42da-167c-546a-9294-c1502f6ccbab
│                          4d0aed03-a30c-52c6-99e7-a8977797c3d9
│                          7acac8b6-ad80-5f1a-90c5-165e7938f773
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           d3db6c5633d83a940028fc88d8102d1f1563e3a4
│     Current version:     0.1.37
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     GUID:                fc5552dc-7882-409b-9b98-699361c74fc4
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           a9ca34786c44d8798eb70e2ee146c8e7a6b9c613
│     Current version:     184.70.3626
│     Minimum Version:     184.70.3626
│     GUID:                7c38c98a-c3e9-4c15-b0cd-c634d441cf7f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           12e31bc5df493718c50805292e594ebf744efc9d
      Current version:     0.1.17
      Minimum Version:     0.1.17
      GUID:                6c6c8104-65e5-45c2-8e31-059df92ef3fd
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
    
________________________________________________

```

Please answer the following questions:

- Operating system and version: Fedora 31 Silverblue
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): RPM
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

RAZORXKHARA33, 2019-12-30:
I am also having this issue. I've tried changing Thunderbolt security settings in the BIOS, and I am running the latest BIOS (1.38). I have also tried disabling secure boot. The only differences between my scenario and OP's are that I'm running Fedora 31 Workstation (rather than Silverblue) and that my Thunderbolt firmware version is currently 40.00 (rather than 33.00). `fwupd` came with my Fedora installation.

hughsie, 2019-12-31:
@gicmo any idea what "failed to update_attach using thunderbolt: update failed (status 6)" means?

roblem, 2020-01-03:
Same issue here on Fedora 31.  I have this update scenario (from 30 to 43):
```
20LD0016US
│
└─Thunderbolt Controller:
  │   Device ID:           afb84615ed57a19207236e1fb115ba01727f331f
  │   Summary:             Unmatched performance for high-speed I/O
  │   Current version:     30.00
  │   Vendor:              LENOVO (TBT:0x0109)
  │   GUIDs:               10bc8610-a3cd-5e74-a187-8d01f3ff345d
  │                        fb595a4f-7e77-5030-a518-e4494392378d
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Device stages updates
  │ 
  └─ThinkPad X1 Yoga 3rd Thunderbolt Controller:
        New version:       43.00
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Yoga 3rd Thunderbolt Firmware
        License:           Proprietary
        Size:              262.1 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Yoga 3rd Thunderbolt Firmware
        
        Important Notice[WARNINGS] :
```

Here is a [dmesg -w output](https://gist.github.com/roblem/82ed7d6716553dcd87ddfc8ee4a9fd57) during an upgrade attempt.  This occurs in bash and using gnome software update.

joncamfield, 2020-01-12:
This is also happening in Ubuntu (19.04):

    Operating system and version: Ubuntu 19.04 ; 5.0.0-38-generic #41-Ubuntu SMP Tue Dec 3 00:27:35 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

    How did you install fwupd (ex: from source, pacman, apt-get, etc): apt
    Have you tried rebooting? Yes
    Are you using an NVMe disk? Yes
    Is secure boot enabled (only for the UEFI plugin)? Yes

closed 2020-02-18 by stale[bot]