fwupd/fwupd #1678: "fwupdmgr get-updates" user visible strings are not translatable

tsdgeos, 2019-12-31:
```shell
$ fwupdmgr --version
$ fwupdmgr --version
client version: 1.3.6
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.6
```

For example
```
$ fwupdmgr get-updates 
No hi ha cap actualització per a Unifying Receiver: current version is RQR12.10_B0032: RQR12.10_B0032=same, RQR12.08_B0030=older, RQR12.07_B0029=older
```

The "current version is" and "same" and "older" strings are not marked with _() when looking at the code.

Also i just realized that it seems that "Unifying Receiver" is also part of the codebase, so it'd be great if that was translatable too :)

hughsie, 2019-12-31:
I thought this would be fixed by https://github.com/fwupd/fwupd/commit/7e4949c4f87f1f9957154f0f81e5626d3bacd7fa

tsdgeos, 2019-12-31:
I don't know much (or anything) about the g_stuff so I may be wrong but 
"current version is %s"
is not marked with _() so as far as i understand that's not extracted to the .po files so not translatable?

superm1, 2020-01-06:
The last part of that string comes from the daemon, not the client.  So it won't be aware of the locale in use.  The only way we can really help that is to let the client notify the daemon of the locale over the wire and then the daemon can build translated strings for stuff like this.

superm1, 2020-01-06:
> Also i just realized that it seems that "Unifying Receiver" is also part of the codebase, so it'd be great if that was translatable too :)

This is the same problem - although it's a hardcoded string it's built by the daemon.

hughsie, 2020-01-06:
> is to let the client notify the daemon of the locale over the wire

PackageKit did this, setting the locale for the request and the daemon would return results in the correct language. It was a lot of work, and didn't work very well in most cases.

superm1, 2020-01-06:
>  didn't work very well in most cases.

Curious, what cases did it break?  It seems like a straightforward idea, but if it's a lot of work I agree probably not worth it.

Would it help if we tried to make the systemd unit source `/etc/default/locale` if it exists and then the daemon could at least try to get better strings based upon the language for the owner of a system.

hughsie, 2020-01-07:
> what cases did it break

I guess PackageKit was unusual in that it was calling all kinds of weird package backends, some of which not UTF-8 safe. Some backends were playing games with setlocale too. Perhaps that's less of a problem with the libraries we use in fwupd.

> then the daemon could at least try to get better strings

Also, at the moment we only return en_US versions of the metadata strings -- even if the vendor has provided translations of the metainfo file sections (none have...)

As a general rule I'd much prefer more granular `FWUPD_ERROR_` enums, which we can handle in the front end, rather than making the daemon respect the user locale.

commit 2febde94 (2020-02-18): Revert "trivial: fu-engine: remove some non-translatable strings (Fixes: #1678)"

This reverts commit 616de292c0a02c146d0bdddbac44eff640951432.
commit b800de9a (2020-02-18): Discard the reason upgrades aren't available (Fixes: #1678)
commit e8bcc600 (2020-02-18): Discard the reason upgrades aren't available (Fixes: #1678)
commit 2471fc36 (2020-02-20): Discard the reason upgrades aren't available (Fixes: #1678)

Also fixes: #1792
closed 2020-02-20 by superm1