fwupd/fwupd #1681: 8BitDo SN30 Pro+ controller error: failed to add USB device 2dc8:6002

rhansen, 2020-01-01:
When I start the [8BitDo SN30 Pro+ controller](https://www.8bitdo.com/sn30-pro-plus/) in Android mode via B+Start (USB ID 2dc8:6002) and plug it in, the fwupd daemon logs the following (`VerboseDomains=*`):

```
FuDevice             using 0703a244275ed772f8e276c4b31389afc296101d for usb:00:03
FuEngine             failed to add USB device 2dc8:6002: failed to retrieve from device on ep 0x82: transfer timed out
```

The device does not appear in the output of `fwupdmgr get-devices`.

If I put the device in bootloader mode via L+R+Start (USB ID 2dc8:5750) before plugging it in, `fwupd get-devices` shows the controller:

```
$ fwupdmgr get-devices
<snip>
├─8Bitdo:
│     Device ID:           0703a244275ed772f8e276c4b31389afc296101d
│     Summary:             A redesigned classic game controller    
│     Current version:     3.02  
│     Vendor:              8Bitdo (USB:0x2DC8)
│     Install Duration:    2 minutes                                                                                   
│     GUIDs:               78acf201-a624-5c0b-a521-e204be14e3d5
│                          b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
│     Device Flags:        • Updatable        
│                          • Supported on remote server
│                          • Is in bootloader mode                                                                     
<snip>
```

However, in bootloader mode `get-updates` shows the firmware for the M30, not the SN30 Pro+:

```
$ fwupdmgr get-updates
No upgrades for 8Bitdo, current is 3.02: 1.10=older
<snip>
```

I believe this is because many different 8BitDo controllers all use USB ID 2dc8:5750 when in bootloader mode (see #1161), and several of the 8BitDo controllers list the same 2dc8:5750 GUID (b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e):

```
$ git clone https://github.com/fwupd/8bitdo-firmware.git
$ cd 8bitdo-firmware
$ git grep b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
releases/M30/1.10/m30.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/M30/1.13/m30.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/N30_Pro_2/6.10/n30pro2.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.25/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.26/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.29/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.30/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.32/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SF30_Pro-SN30_Pro/1.33/sf30sn30pro.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
releases/SN30_Pro+/3.01/sn30pro_plus.metainfo.xml:    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
```

```
$ fwupdmgr --version
client version: 1.3.5-615-g7fb6cbc5
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.5-615-g7fb6cbc5
```

- Operating system and version: Custom Debian based on testing
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? yes
- Are you using an NVMe disk? no

hughsie, 2020-01-02:
> However, in bootloader mode get-updates shows the firmware for the M30, not the SN30 Pro+:

Most of the 8bitdo touchpads share a bootloader, and so we have to "see" the device in runtime mode before we can identify it properly.

> failed to retrieve from device on ep 0x82: transfer timed out

I suspect we need to fix that.

rhansen, 2020-01-03:
Is it possible to see the devices and their current firmware version in runtime mode (e.g., B+Start) then issue a command over USB to switch them to update mode when it is time to install a different version?

hughsie, 2020-01-06:
> then issue a command over USB to switch them to update mode

I don't think the hardware has that capability.

rhansen, 2020-02-14:
Still a problem at current master; please reopen.

rhansen, 2020-02-14:
I tried increasing [`FU_EBITDO_USB_TIMEOUT`](https://github.com/fwupd/fwupd/blob/5c7e17276c8eb0c2a845cecb7d644a4bccf80519/plugins/ebitdo/fu-ebitdo-common.h#L21) from 5s to 60s but it didn't help:

```
02:07:24:0709 FuDevice  using 0703a244275ed772f8e276c4b31389afc296101d for usb:00:03
02:08:24:0815 FuEngine  failed to add USB device 2dc8:6002: failed to retrieve from device on ep 0x81: transfer timed out
```

rhansen, 2020-02-14:
Output of `lsusb`:

```
Bus 001 Device 012: ID 2dc8:6002 8Bitdo SF30 Pro   8BitDo SN30 Pro+
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x2dc8
  idProduct          0x6002
  bcdDevice            0.01
  iManufacturer           1 8Bitdo SF30 Pro
  iProduct                2 8BitDo SN30 Pro+
  iSerial                 3 SF30 Pro
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              480mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     136
          Report Descriptor: (length is 136)
            Item(Global): Usage Page, data= [ 0x01 ] 1
                            Generic Desktop Controls
            Item(Local ): Usage, data= [ 0x05 ] 5
                            Gamepad
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Report ID, data= [ 0x01 ] 1
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Physical Minimum, data= [ 0x00 ] 0
            Item(Global): Physical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x0f ] 15
            Item(Global): Usage Page, data= [ 0x09 ] 9
                            Buttons
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            Button 1 (Primary)
            Item(Local ): Usage Maximum, data= [ 0x0f ] 15
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
[... 88 lines cut ...]
                            Preferred_State No_Null_Position Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0000
  (Bus Powered)
```

rhansen, 2020-03-07:
With `LIBUSB_DEBUG=4`:

```
18:57:45:0436 FuDevice             using 0703a244275ed772f8e276c4b31389afc296101d for usb:00:03
[34.847886] [0001590b] libusb: debug [libusb_open] open 1.10
[34.848024] [0001590b] libusb: debug [usbi_add_pollfd] add fd 18 events 4
[34.850106] [0001590f] libusb: debug [handle_events] poll() returned 1
[34.850193] [0001590f] libusb: debug [handle_events] caught a fish on the event pipe
[34.850276] [0001590f] libusb: debug [handle_events] someone updated the poll fds
[34.850354] [0001590f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[34.850427] [0001590f] libusb: debug [handle_events] poll fds modified, reallocating
[34.850505] [0001590f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[34.850588] [0001590b] libusb: debug [libusb_alloc_transfer] transfer 0x5598811d2db0
[34.850661] [0001590b] libusb: debug [libusb_submit_transfer] transfer 0x5598811d2db0
[34.850733] [0001590b] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[34.850824] [0001590b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[34.851452] [0001590f] libusb: debug [handle_events] poll() returned 1
[34.851556] [0001590f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[34.851632] [0001590f] libusb: debug [handle_control_completion] handling completion status 0
[34.851705] [0001590f] libusb: debug [disarm_timerfd]
[34.851781] [0001590f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5598811d2db0 has callback 0x7f6f3a5daba0
[34.851788] [0001590f] libusb: debug [sync_transfer_cb] actual_length=4
[34.851898] [0001590f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[34.851971] [0001590f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[34.852054] [0001590b] libusb: debug [libusb_free_transfer] transfer 0x5598811d2db0
[34.852126] [0001590b] libusb: debug [libusb_alloc_transfer] transfer 0x5598811d2e90
[34.852200] [0001590b] libusb: debug [libusb_submit_transfer] transfer 0x5598811d2e90
[34.852271] [0001590b] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[34.852357] [0001590b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[34.852568] [0001590f] libusb: debug [handle_events] poll() returned 1
[34.852580] [0001590f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=34
[34.852671] [0001590f] libusb: debug [handle_control_completion] handling completion status 0
[34.852747] [0001590f] libusb: debug [disarm_timerfd]
[34.852818] [0001590f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5598811d2e90 has callback 0x7f6f3a5daba0
[34.852863] [0001590f] libusb: debug [sync_transfer_cb] actual_length=34
[34.852945] [0001590f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[34.853158] [0001590f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[34.853334] [0001590b] libusb: debug [libusb_free_transfer] transfer 0x5598811d2e90
[34.853502] [0001590b] libusb: debug [libusb_alloc_transfer] transfer 0x5598811d2e90
[34.853664] [0001590b] libusb: debug [libusb_submit_transfer] transfer 0x5598811d2e90
[34.853835] [0001590b] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[34.853990] [0001590b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[34.854144] [0001590f] libusb: debug [handle_events] poll() returned 1
[34.854290] [0001590f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[34.854424] [0001590f] libusb: debug [handle_control_completion] handling completion status 0
[34.854557] [0001590f] libusb: debug [disarm_timerfd]
[34.854692] [0001590f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5598811d2e90 has callback 0x7f6f3a5daba0
[34.854827] [0001590f] libusb: debug [sync_transfer_cb] actual_length=4
[34.854969] [0001590f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[34.855157] [0001590f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[34.855300] [0001590b] libusb: debug [libusb_free_transfer] transfer 0x5598811d2e90
[34.855434] [0001590b] libusb: debug [libusb_alloc_transfer] transfer 0x5598811d2e90
[34.855566] [0001590b] libusb: debug [libusb_submit_transfer] transfer 0x5598811d2e90
[34.855698] [0001590b] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[34.855842] [0001590b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[34.856232] [0001590f] libusb: debug [handle_events] poll() returned 1
[34.856384] [0001590f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=28
[34.856519] [0001590f] libusb: debug [handle_control_completion] handling completion status 0
[34.856653] [0001590f] libusb: debug [disarm_timerfd]
[34.856786] [0001590f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5598811d2e90 has callback 0x7f6f3a5daba0
[34.856925] [0001590f] libusb: debug [sync_transfer_cb] actual_length=28
[34.857065] [0001590f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[34.857201] [0001590f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[... cut ...]
UDEV  [54749.994285] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:2DC8:6002.0009 (hid)
UDEV  [54749.996168] remove   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:2DC8:6002.0009 (hid)
UDEV  [54749.997682] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)
UDEV  [54750.004792] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)
KERNEL[54754.959858] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)
UDEV  [54754.973478] unbind   /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)
```

rhansen, 2020-03-29:
Anything I can do to help with this issue? Any more information you need or changes I can try?

hughsie, 2020-03-29:
To be honest, I'm not super happy with the recent lack of help I've got from 8bitdo. They were uploading firmware themselves, but then stopped, so I helped them once or twice (with their permission) and they've not done it since.

seancmonahan, 2020-04-03:
> > then issue a command over USB to switch them to update mode
> 
> I don't think the hardware has that capability.

The hardware does support the ability to programmatically enter upgrade mode per the [8BitDo Firmware Updater page](https://support.8bitdo.com/firmware-updater.html).

I'd be interested in helping to figure out how it does that. I have an 8BitDo SN30 Pro+ myself, and I also have this issue. Let me know if there's anything I can do on either front. I'd love to not have to boot up Windows to update my controller.

rhansen, 2020-04-12:
> Let me know if there's anything I can do

I wonder if it would be helpful to capture the USB traffic during an update using the official Windows firmware update tool. Can you set up a Windows VM (https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/) and use Wireshark to capture the traffic (https://wiki.wireshark.org/CaptureSetup/USB#Windows)?

rhansen, 2020-05-12:
Problem still exists.

Sanaki, 2020-07-11:
Sadly, still valid.

rhansen, 2020-08-14:
Still valid.

rhansen, 2020-09-13:
Still valid.

rhansen, 2020-10-14:
Still valid.

henryptung, 2020-10-19:
FWIW, here's a USBPcap capture of an SN30 Pro+ upgrade from v3.02 to v4.02 using the Windows updater on Windows 10.

Controller was started in Android mode (Start+B).

https://www.dropbox.com/s/eyql388meq74yi1/8bitdo%20Sn30%20Pro%20Plus%20Upgrade%20v3.02%20to%20v4.02.pcapng.gz?dl=0

rhansen, 2020-11-19:
Still valid. Hopefully one of these days I'll have the time to look at the capture and figure out what's going on.

rhansen, 2021-03-02:
I squinted at the USB packet capture for a while and fwupd appears to be missing a byte in the packets it sends to the device when it is in runtime mode (Start+B).

The packet capture from @henryptung shows the host sending `81 05 00 21 01 00 00` when querying the version. After that, it sends `81 05 00 51 00 00` to put the device in bootloader mode. (`FuEbitdoPktCmd` does not include an entry for command 0x51, but I'm assuming it is "restart in bootloader mode".) Those are the only two commands I see when the device is in runtime mode. When fwupd queries the version it sends `05 00 21 01 00 00` (it is missing that first byte).

That extra byte only seems to be needed in runtime mode; when the device is in bootloader mode the payloads in the recorded packet capture align with the `FuEbitdoPkt` struct.

@hughsie Would you mind reopening this bug?

rhansen, 2021-03-02:
With the following patch I was able to detect the SN30 Pro+ correctly:

```diff
diff --git a/plugins/ebitdo/fu-ebitdo-device.c b/plugins/ebitdo/fu-ebitdo-device.c
index 3b8581e9..ad05eebb 100644
--- a/plugins/ebitdo/fu-ebitdo-device.c
+++ b/plugins/ebitdo/fu-ebitdo-device.c
@@ -40,6 +40,10 @@ fu_ebitdo_device_send (FuEbitdoDevice *self,
 	/* different */
 	if (fu_device_has_flag (FU_DEVICE (self), FWUPD_DEVICE_FLAG_IS_BOOTLOADER))
 		ep_out = FU_EBITDO_USB_BOOTLOADER_EP_OUT;
+	else {
+		packet[0] = 0x81;
+		hdr = (FuEbitdoPkt *)(packet + 1);
+	}
 
 	/* check size */
 	if (in_len > 64 - 8) {
@@ -160,7 +164,7 @@ fu_ebitdo_device_receive (FuEbitdoDevice *self,
 	}
 
 	/* get-version (firmware) -- not a packet, just raw data! */
-	if (hdr->pkt_len == FU_EBITDO_PKT_CMD_GET_VERSION_RESPONSE) {
+	if (hdr->type == FU_EBITDO_PKT_CMD_GET_VERSION_RESPONSE) {
 		if (out != NULL) {
 			if (out_len != 4) {
 				g_set_error (error,
@@ -171,7 +175,7 @@ fu_ebitdo_device_receive (FuEbitdoDevice *self,
 				return FALSE;
 			}
 			if (!fu_memcpy_safe (out, out_len, 0x0,					/* dst */
-					     packet, sizeof(packet), 0x1,			/* src */
+					     packet, sizeof(packet), 0x2,			/* src */
 					     4, error))
 				return FALSE;
 		}
```

Of course this patch almost certainly breaks other devices so it isn't the right fix, but it's progress I guess. I haven't tried to update the firmware yet because I don't want to brick my controller. :slightly_smiling_face:

zephryn, 2022-02-02:
Can confirm this for my Pro 2 in runtime mode as well; ebitdo reports `05 00 21 01 00 00` while a USB capture from the Windows utility shows `81 05 00 21 01 00 00` (and subsequent padding bytes).

zephryn, 2022-02-06:
Not sure how much it's within the scope of this issue specifically, but I made a capture of USB packets during a firmware update for my Pro 2; it seems that a lot of changes in the protocol are consistent between the SN30+ and my device.

superm1, 2023-01-30:
It seems to me we essentially need to bifurcate the code for the older protocol/responses vs the newer one.  Anyone with the hw want to take a stab?

hughsie, 2023-09-23:
8bitdo doesn't want to support this hardware, sorry.

closed 2023-09-23 by hughsie

diabl0w, 2024-02-15:
> 8bitdo doesn't want to support this hardware, sorry.

why was this closed? is manufacturer support required for devices to be supported by fwupd?

hughsie, 2024-02-15:
> is manufacturer support required for devices to be supported by fwupd

Yes and yes. 8bitdo needs to add the VID/PIDs and any protocol changes in https://github.com/fwupd/fwupd/tree/main/plugins/ebitdo and then the OEM needs to upload firmware to the LVFS.