fwupd/fwupd #1685: Same updates offered repeatedly (Lenovo ThinkPad X1)

LinuxOnTheDesktop, 2020-01-03:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           eded407096031a81370bc25ae01650697a7c72b6
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
│                          09bc8035-c26c-56c8-af2d-5b0fd7e1efe1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.8.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               963522d7-bc73-51fa-a70c-cf615f42bcf6
│                          b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.43
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO
│     Update State:        needs-reboot
│     Last modified:       2020-01-03 23:39
│     GUIDs:               a4b51dca-8f97-4310-8821-3330f83c9135
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a
│                          6b3879ff-921f-5bc5-a0c2-5cec28f31d2f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
[... 818 lines cut ...]
The following happens each time I run `fwupdmgr update`. Initially I thought the problem was this one but then even after being sure to answer 'Y' to each prompt (as is represented in the input and output that I paste below), the loop continues, i.e. I keep being offered the same update. Upon rebooting some text flashes up too quickly for me to see and then I enter my Linux desktop.

```
~ $ fwupdmgr update
Upgrade available for System Firmware from 0.1.43 to 0.1.44
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 0.1.44 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[ -                                     ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for UEFI Device Firmware, current is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
Upgrade available for UEFI Device Firmware from 0.1.18 to 0.1.19
20KHCTO1WW must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: Y
Downloading 0.1.19 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…                                   ]
Scheduling…              [***************************************]
Successfully installed firmware
No upgrades for Unifying Receiver, current is RQR24.10_B0036: RQR24.10_B0036=same, RQR24.06_B0030=older, RQR24.05_B0029=older

An update requires a reboot to complete. Restart now? [Y|n]: Y
```

LinuxOnTheDesktop, 2020-01-04:
After five or so reboots, the first of the two upgrades that was being offered actually installed itself. After running `fwupdmgr update` once again, and rebooting, the second update installed itself too. Weird.

hughsie, 2020-01-06:
> After five or so reboots

!!!

> After running fwupdmgr update once again, and rebooting

I'm guessing you can't reproduce this by installing the old firmware versions? I do wonder if we should limit the number of UEFI updates to just one at a time, as this seems to be something vendors keep getting wrong. @superm1 comments?

LinuxOnTheDesktop, 2020-01-06:
I am sorry but, given that things work at present, and given that I've filed another Linux bug report today already, I'm not going to try to reproduce the problem in that way.

One option you have is to give the user a notification ('This installation of > 1 update may not work') and a choice ('Install all in one go or one after another?').

superm1, 2020-01-06:
> !!!
>
>
> I'm guessing you can't reproduce this by installing the old firmware versions? I do wonder if we should limit the number of UEFI updates to just one at a time, as this seems to be something [...]

That's really unfortunate, it basically means that if the vendor splits up the updates to say 4 ESRT entries that's 4 reboots to get them all installed!   How big is this problem?
I would think this should either be a flag we can add into quirks for known problematic systems if it's just a few.

Aberrantfox, 2020-01-08:
I get a very similar problem that I documented here: https://github.com/pop-os/pop/issues/818 
I was under the false impression that it was the fault of Pop!_os at first but it seems this binary is the problem. My issue is for the UEFI firmware. I have a t490s so that's a little different. 

I believe the version it's try to update from/to is 1.1.15 to 1.1.55 - although I am not at my machine now and I'm just remembering, I documented some other versions on the original issue I filed. Probably the most annoying thing is that anytime I manage to trick it into getting into the process of updating the firmware (because sometimes it just completely skips over the firmware update and boots back like nothing happened) it starts the firmware update and immediately kills itself. What exactly is going on here, is there anywhere I can refer to for a log file to help?

Aberrantfox, 2020-02-13:
Still experiencing this ,  if there is any additional testing I can do let me know.

LinuxOnTheDesktop, 2020-02-15:
This is happening again, with a new update. Something does happen on reboot, but I don't know what: a message appears but it is tiny on my HiDPI screen (can't something be done about that?) and is there only for a second; and, when I reach the desktop and run a terminal, the same update - the one I detail below - gets offered (i.e. the same one that I tried to install already).
```

~ $ fwupdmgr get-updates
No upgrades for Thunderbolt Controller: current version is 43.00: 43.00=same
No upgrades for UEFI Device Firmware: current version is 184.70.3626: 184.70.3626=same, 184.65.3590=older, 184.60.3561=older, 184.55.3510=older
No upgrades for Unifying Receiver: current version is RQR24.10_B0036: RQR24.10_B0036=same, RQR24.06_B0030=older, RQR24.05_B0029=older
20KHCTO1WW
│
├─System Firmware:
│ │   Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│ │   Current version:     0.1.44
│ │   Minimum Version:     0.0.1
│ │   Vendor:              LENOVO (DMI:LENOVO)
│ │   Update State:        success
│ │   GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
│ │
│ └─ThinkPad X1 Carbon 6th System Update:
│       New version:       0.1.45
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 6th System Firmware
│       License:           Proprietary
│       Size:              9.5 MB
│       Vendor:            Lenovo Ltd.
│       Flags:             is-upgrade
│       Description:       Lenovo ThinkPad X1 Carbon 6th System Firmware Version 1.45
│
│       The computer will be restarted automatically after updating BIOS completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.
│
│       This stable release fixes the following issues:
│
│        • Fixed an issue where Thunderbolt controller device might disappear from
│
│       the Device Manager at 2nd boot.
│
│       (Note) ECP version 1.20 or later is required for complete solution.
│
└─UEFI Device Firmware:
  │   Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  │   Current version:     0.1.19
  │   Minimum Version:     0.1.19
  │   Vendor:              DMI:LENOVO
  │   Update State:        success
  │   GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Device is usable for the duration of the update
  │
  └─ThinkPad X1 Carbon 6th Embedded Controller Update:
        New version:       0.1.20
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
        License:           Proprietary
        Size:              767.1 kB
        Vendor:            Lenovo Ltd.
        Flags:             is-upgrade
        Description:       Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware 1.20

        The computer will be restarted automatically after updating EC completely. Do NOT turn off your computer or remove the AC adaptor while update is in progress.

        This stable release fixes the following issues:

         • Fixed an issue where Thunderbolt controller device might disappear from

        the Device Manager at 2nd boot.

        (Note) UEFI BIOS version 1.45 or later is required for complete solution.
```

As another poster asks (on this same page): ' What exactly is going on here, is there anywhere I can refer to for a log file to help?' Is this _yet another_ Lenovo mess up?

LinuxOnTheDesktop, 2020-02-15:
I have a partial diagnosis and a workaround.

What is happening is that somehow the update mechanism can't lauch the fwupd boot environment. I ran that environment manually via the grub boot menu, and then the process of installing the update began and it completed successfully.

hebrewd, 2020-02-18:
@LinuxOnTheDesktop Thanks, I was facing the same issue, rebooting again and again to no avail.
It sure appears like the booting into the fwupd entry allow fwupd to complete the update and now everything seems to be up to date.

closed 2020-03-26 by stale[bot]