fwupd/fwupd #1687: Dell BIOS Upgrade Fails

dupondje, 2020-01-05:
When trying to update BIOS on my system, I get the following:

```
$ sudo fwupdmgr update
No upgrades for System Firmware: No releases found: Firmware version formats were different, device was 'dell-bios' and release is 'quad'
```
Even using --force doesn't override the error.

```shell
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.6
```

```shell
$ fwupdmgr get-devices
Precision 5520
│
├─Thunderbolt Controller:
│     Device ID:           3628aef4ccee7f1e32fd62c5a7d8c076f3f3329c
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     21.00
│     Vendor:              Dell (TBT:0x00D4)
│     GUIDs:               c5364422-2ec3-5092-b30e-03bdb1a53e4a ← TBT-00d407bf
│                          f4836136-af2e-5c18-bdf9-8785fe32c496 ← TBT-00d407bf-0000:08:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─PC300 NVMe SK hynix 512GB:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     20005A00
│     Vendor:              SK hynix (NVME:0x1C5C)
│     Serial Number:       ES7BN50451140BB48
│     GUIDs:               fd7d6d0b-40d7-50cb-9bc9-8dfa398cee33 ← NVME\VEN_1C5C&DEV_1284&REV_00
│                          66cc0d83-cfdc-5602-9581-3637989f0c64 ← NVME\VEN_1C5C&DEV_1284
│                          c93201aa-cbec-55d4-bafb-f7c488646839 ← PC300 NVMe SK hynix 512GB
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           f49999b6e3af5b4bf8ad1f2ed2922b76df457271
│     Current version:     1.16.0
│     Minimum Version:     1.16.0
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     GUID:                34578c72-11dc-4378-bc7f-b643866f598c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
└─Touchpad:
      Device ID:           7d3efd1506b7280469190a79abd799cea3467681
      Current version:     1.7.2378871
      Bootloader Version:  54.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               1d6705d0-3fa6-58d2-bd0e-806ed6792c18 ← HIDRAW\VEN_06CB&DEV_7A13&REV_00
                           2aa17264-7ef0-5cc5-9368-8595974a5e31 ← HIDRAW\VEN_06CB&DEV_7A13
                           741692a4-ae19-5ecc-9e48-e644604c2345 ← SYNAPTICS_RMI\TM3125-001
                           a90f8ef5-de8d-5810-a6c2-d0b88b1e0be8 ← SYNAPTICS_RMI\TM3125
      Device Flags:        • Internal device
                           • Updatable
    

```

```shell
$ efibootmgr -v
BootCurrent: 0008
Timeout: 0 seconds
BootOrder: 0008,0000,0001,0002,0003,0004,0005,0007
Boot0000* ubuntu	HD(1,GPT,6ce02f89-242e-40bc-8287-3e229d32931a,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux Firmware Updater	HD(1,GPT,6ce02f89-242e-40bc-8287-3e229d32931a,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0002* Diskette Drive	BBS(Floppy,Diskette Drive,0x0)..BO
Boot0003* M.2 PCIe SSD	BBS(HD,P0: PC300 NVMe SK hynix 512GB,0x0)..BO
Boot0004* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0005* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0006* USB NIC	BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0007* Onboard NIC	BBS(Network,Onboard NIC,0x0)..BO
Boot0008* ArchLinux	HD(1,GPT,6ce02f89-242e-40bc-8287-3e229d32931a,0x800,0x100000)/File(\EFI\ArchLinux\grubx64.efi)

```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2020-01-06:
@hughsie do these specific releases on LVFS need to be updated to have both `dell-bios` and `quad`?  I thought they were all updated a few weeks ago?

hughsie, 2020-01-06:
> I thought they were all updated a few weeks ago?

Urgh, they were. For all existing firmware. 0.1.18.0 was uploaded a few days ago. I've changed the vendor default to "dell-bios" which I probably should have done when I deployed the change. I've rebuilt the dell, testing and stable remotes which should build in about an hour. After that, a "fwupdmgr refresh" should fix everything. Please let us know either way.

dupondje, 2020-01-07:
Success! :)

Thanks

```
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 1 local device supported
$ sudo fwupdmgr update
Upgrade available for System Firmware from 1.16.0 to 1.18.0
Precision 5520 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 1.18.0 for System Firmware...
Fetching firmware https://fwupd.org/downloads/bc334d8b098f2e91603c5f7dfdc837fb01797bbe-Dell%20XPS%2015%209560&Precision%205520%20System%20BIOS_Ver.1.18.0.cab
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[  -                                    ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: 

```

closed 2020-01-07 by dupondje

hughsie, 2020-01-07:
> Success! :)

Phew! Thanks for testing.