fwupd/fwupd #1689: Implement flashing for System76 Laptop ECs

jackpot51, 2020-01-06:
I have been working on GPLv3 firmware for System76 Laptop EC's, which are usually ITE IT8587E ECs, and the flashing protocol is yet to be defined. Some problems I noticed with other protocols is that they require re-using a channel already used by the OS, such as the keyboard controller ports 0x60 and 0x64, or the ACPI EC ports 0x62 and 0x66. This can result in unexpected bytes being sent by the kernel during the flashing process that userspace cannot detect.

To counter this, I would like to work on a new protocol for flashing from Linux userspace using fwupd that is immune to the kernel's communication with the keyboard controller and ACPI EC ports. If there are any recommendations or there are other LPC EC's with reliable flashing code in fwupd, please let me know.

Here is the source for the new firmware: https://github.com/system76/ec. I hope to finalize the support for updates in the firmware and in fwupd within the next two months.

superm1, 2020-01-06:
Are these UEFI or coreboot systems? If they're UEFI, the safest way is probably in a UEFI executable or DXE driver.

jackpot51, 2020-01-06:
The system firmware is coreboot with a Tianocore payload.

superm1, 2020-01-06:
If you're running tianocore payload I would think you could still go the UEFI approach then too.

hughsie, 2020-01-06:
> If there are any recommendations or there are other LPC EC's with reliable flashing code in fwupd

There's https://github.com/fwupd/fwupd/tree/master/plugins/superio which is for some of the ITE chips, although it's somewhat fragile, and we run this in a special update target to try and isolate the EC as much as possible. Star labs have been using this for some time. It is however using port IO, which is going to stop working Real Soon Now for people running new kernels and with secure boot turned on. Long term I did want to rewrite the ITE loader as an EFI binary. If you want to take a stab at that it would be very welcome...

superm1, 2020-01-06:
> stop working Real Soon Now for people running new kernels and with secure boot turned on

Well in this case with coreboot + tiano core I don't think expect secure boot to be available.  However the kernel lock down policies enacted by secure boot can be set manually too.  So I do think that moving superio based updating to something in UEFI would be good.

For development purpose a standalone EFI executable is fine, but I think would like to see superio and UEFI capsule updating come from a "single" fwupd EFI binary if it's possible.  I know this will require some shuffling in the source tree.

My reasoning is that there is a bunch of infrastructure that has to happen to:
1. install EFI binaries to ESP
2. get new EFI binaries built properly, and signed with distro keys
3. split up packaging properly

Maybe superio plugin could act like Dell plugin where it just passes payloads that aren't part of ESRT into UEFI plugin.

Then `fwup$arch.efi` would just need to look for a header to indicate it's a superio payload not a capusle and do different logic.

hughsie, 2020-01-06:
> I know this will require some shuffling in the source tree.

Another plugin is perhaps going to be writing to the ESP at some point too, so I guess we need to move out some of the ESP handling code to libfwupdplugin.

> install EFI binaries to ESP

Well, I did think that the efi binary would be shipped in the .cab file, i.e. the EFI binary would be a loader and a blob of data, and then signed as an EFI binary by the hardware vendor (if applicable).

> Then fwup$arch.efi would just need to look for a header to indicate it's a superio payload not a capusle and do different logic.

I don't think one binary that does multiple different flashing methods buys us anything; we can just create and use a different EFI binary that's specifically designed for superio.

superm1, 2020-01-06:
> Well, I did think that the efi binary would be shipped in the .cab file, i.e. the EFI binary would be a loader and a blob of data, and then signed as an EFI binary by the hardware vendor (if applicable).

EFI binary shipped by CAB file?  That seems like a slippery slope of precedent that can lead to non-free code directly in fwupd becoming common.

> and then signed as an EFI binary by the hardware vendor (if applicable).

At least with secure boot the EFI binary has to be signed by something with a cert already registered.  In practice this is distro private keys (which are verified by shim).

> I don't think one binary that does multiple different flashing methods buys us anything; we can just create and use a different EFI binary that's specifically designed for superio.

It's not long ago that we had that fiasco where a BIOS vendor was looking for `fwupx64.efi` and we renamed to `fwupdx64.efi` and it caused major problems for them.

hughsie, 2020-01-10:
>  That seems like a slippery slope of precedent that can lead to non-free code directly in fwupd becoming common

True. I was thinking more about signing but the OEMs but that's probably less of a concern for this kind of thing.

superm1, 2020-01-10:
> True. I was thinking more about signing but the OEMs but that's probably less of a concern for this kind of thing.

I think if you want to have an OEM signature solution the right place for it is on the binary image that is _flashed_ not the flashing tool/executable.

superm1, 2020-07-10:
Nothing against this directionally, but at least from upstream fwupd team there is no intent to work on this directly ourselves so I'm going to close it.
If the System76 team still wants this but doesn't have the manpower to do it, then I would recommend looking at some of the consulting teams that can work on it here: https://fwupd.org/lvfs/docs/consulting

Certainly happy to review a pull request and accept the code when applicable.  As mentioned in the above comments kernel lockdown is going to prohibit anything from communicating over Port IO.  If you can boot a Tianocore payload from coreboot, then I would recommend that this be done as a UEFI application.

My preference for it would be that it's a modification to the existing UEFI application shipped with fwupd (fwupdx64.efi for 64 bit x86 systems).  Currently it's stored in the tree as `plugins/uefi/efi` and only enabled when you build `plugin_uefi` for UEFI capsule support.

I think that realistically if we end up growing support for that thing to flash more than just UEFI capsules, then we instead will want to move it somewhere else in tree that is enabled if you have `plugin_system76_ec` or `plugin_uefi` enabled.  Some design ideas:
* Splitting out ESP handling code to be part of `libfwupdplugin` instead
* Perhaps to have `plugin_system76_ec` "Register" a UEFI device with some special device metadata, and then modify plugin_uefi
* Split `plugin_uefi` into `plugin_uefi_capsule` and `plugin_uefi`.  `plugin_uefi_capsule` registers devices, as does `plugin_system76_ec` and then `plugin_uefi` handles flashing for both.
* The actual meson.build in `plugins/uefi/efi/` compiles in an extra source file for systems76 support when the daemon is built with it.

I don't think that any of them is more right than the other until we have a pull request and see what really needs to be changed and maintained.

The structure that gets put into efivars currently to relay information between Linux and the next boot to trigger the capsule will need a special flag to indicate it's working with a System 76 EC.  Some ideas here:
* If *all* System 76 machines will always take the same EC codebase, reserve and use a GUID.  Hardcode that in the code, and if it's found flash it with the UEFI flasher rather than the `UpdateCapsule` handler.
* If there is a family of machines, then reserve a set of GUIDs, do the same.
* If it's going to change frequently (IE 1 EC build -> 1 model), then modify the structure to differentiate between types of UEFI payloads.

closed 2020-07-10 by superm1