fwupd/fwupd #1698: MOTD messages are repeated

OddBloke, 2020-01-10:
When SSHing into my desktop, I was greeted with _182_ lines of MOTD content from fwupd. This content was repeated 3 times:

```
B450M DS3H
│
└─Unifying Receiver:
  │   Device ID:           cf3685ba249d3d98602047341d6f5a5556a6ac05
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR12.07_B0029
  │   Bootloader Version:  BOT01.02_B0014
  │   Vendor:              Logitech, Inc. (USB:0x046D)
  │   Install Duration:    30 seconds
  │   GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
  │                        fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
  │                        279ed287-3607-549e-bacc-f873bb9838c4
  │   Device Flags:        • Updatable
  │                        • Supported on remote server
  │ 
  ├─Unifying Receiver (RQR12) Device Update:
  │     New version:       RQR12.10_B0032
  │     Remote ID:         lvfs
  │     Summary:           Firmware for the Logitech Unifying Receiver (RQR12.xx)
  │     License:           Proprietary
  │     Size:              56.8 kB
  │     Vendor:            Logitech
  │     Duration:          30 seconds
  │     Flags:             is-upgrade
  │     Description:       This release addresses an encrypted keystroke injection vulnerability sent by pointing devices. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
  │     
  │     A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
  │     
  │      • Wireless Mouse M335
  │      • Zone Touch Mouse T400
  │      • Wireless Mouse M545
  │      • Wireless Mouse M560
  │      • Touch Mouse M600
  │      • Touch Mouse T620
  │      • Wireless Rechargeable Touchpad T650
  │     
  │     Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
  │   
  └─Unifying Receiver (RQR12) Device Update:
        New version:       RQR12.08_B0030
        Remote ID:         lvfs
        Summary:           Firmware for the Logitech Unifying Receiver (RQR12.xx)
        License:           Proprietary
        Size:              72.7 kB
        Vendor:            Logitech
        Duration:          30 seconds
        Flags:             is-upgrade
        Description:       This release addresses an encrypted keystroke injection issue known as Bastille security issue #13. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
        
        A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
        
         • Wireless Mouse M335
         • Zone Touch Mouse T400
         • Wireless Mouse M545
         • Wireless Mouse M560
         • Touch Mouse M600
         • Touch Mouse T620
         • Wireless Rechargeable Touchpad T650
        
        Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
```

I've attached [/run/private/motd.d/85-fwupd](https://github.com/fwupd/fwupd/files/4046424/85-fwupd.txt). (You can see that this has the same content _4_ times now, so clearly something is appending when it should be replacing.)

(This was originally reported in Ubuntu at https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1859180.)

## Boilerplate

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.6
```

```shell
$ fwupdmgr get-devices
<snip>
├─Unifying Receiver:
│     Device ID:           cf3685ba249d3d98602047341d6f5a5556a6ac05
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.07_B0029
│     Bootloader Version:  BOT01.02_B0014
│     Vendor:              Logitech, Inc. (USB:0x046D)
│     Install Duration:    30 seconds
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│                          • Supported on remote server
<snip>
```
(Just included the relevant portion here.)

Please answer the following questions:

- Operating system and version: Ubuntu 20.04 (development release)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): it's in the base system, so apt
- Have you tried rebooting? Don't think it would change matters, as this is intentional behaviour.
- Are you using an NVMe disk? No

hughsie, 2020-01-10:
@superm1 ?

OddBloke, 2020-01-10:
As requested in the Ubuntu bug:

```
$ systemctl show fwupd-refresh.service | cat
Type=oneshot
Restart=no
NotifyAccess=none
RestartUSec=100ms
TimeoutStartUSec=infinity
TimeoutStopUSec=1min 30s
TimeoutAbortUSec=1min 30s
RuntimeMaxUSec=infinity
WatchdogUSec=0
WatchdogTimestampMonotonic=0
RootDirectoryStartOnly=no
RemainAfterExit=no
GuessMainPID=yes
SuccessExitStatus=2
MainPID=0
ControlPID=0
FileDescriptorStoreMax=0
NFileDescriptorStore=0
StatusErrno=0
Result=success
ReloadResult=success
CleanResult=success
UID=[not set]
GID=[not set]
NRestarts=0
OOMPolicy=stop
ExecMainStartTimestamp=Fri 2020-01-10 09:30:21 EST
ExecMainStartTimestampMonotonic=169999562244
ExecMainExitTimestamp=Fri 2020-01-10 09:30:21 EST
ExecMainExitTimestampMonotonic=169999651794
ExecMainPID=548705
ExecMainCode=1
ExecMainStatus=0
ExecStart={ path=/usr/bin/fwupdmgr ; argv[]=/usr/bin/fwupdmgr refresh --no-metadata-check ; ignore_errors=no ; start_time=[n/a] ; stop_time=[n/a] ; pid=0 ; code=(null) ; status=0/0 }
ExecStart={ path=/usr/bin/fwupdmgr ; argv[]=/usr/bin/fwupdmgr get-updates --log 85-fwupd ; ignore_errors=no ; start_time=[n/a] ; stop_time=[n/a] ; pid=0 ; code=(null) ; status=0/0 }
ExecStartEx={ path=/usr/bin/fwupdmgr ; argv[]=/usr/bin/fwupdmgr refresh --no-metadata-check ; flags= ; start_time=[n/a] ; stop_time=[n/a] ; pid=0 ; code=(null) ; status=0/0 }
ExecStartEx={ path=/usr/bin/fwupdmgr ; argv[]=/usr/bin/fwupdmgr get-updates --log 85-fwupd ; flags= ; start_time=[n/a] ; stop_time=[n/a] ; pid=0 ; code=(null) ; status=0/0 }
Slice=system.slice
MemoryCurrent=[not set]
CPUUsageNSec=[not set]
EffectiveCPUs=
EffectiveMemoryNodes=
TasksCurrent=[not set]
IPIngressBytes=[no data]
IPIngressPackets=[no data]
IPEgressBytes=[no data]
IPEgressPackets=[no data]
IOReadBytes=18446744073709551615
IOReadOperations=18446744073709551615
IOWriteBytes=18446744073709551615
IOWriteOperations=18446744073709551615
Delegate=no
CPUAccounting=no
CPUWeight=[not set]
StartupCPUWeight=[not set]
CPUShares=[not set]
StartupCPUShares=[not set]
CPUQuotaPerSecUSec=infinity
CPUQuotaPeriodUSec=infinity
AllowedCPUs=
AllowedMemoryNodes=
IOAccounting=no
IOWeight=[not set]
StartupIOWeight=[not set]
BlockIOAccounting=no
BlockIOWeight=[not set]
[... 154 lines cut ...]
RefuseManualStop=no
AllowIsolate=no
DefaultDependencies=yes
OnFailureJobMode=replace
IgnoreOnIsolate=no
NeedDaemonReload=no
JobTimeoutUSec=infinity
JobRunningTimeoutUSec=infinity
JobTimeoutAction=none
ConditionResult=yes
AssertResult=yes
ConditionTimestamp=Fri 2020-01-10 09:30:19 EST
ConditionTimestampMonotonic=169998186712
AssertTimestamp=Fri 2020-01-10 09:30:19 EST
AssertTimestampMonotonic=169998186713
Transient=no
Perpetual=no
StartLimitIntervalUSec=10s
StartLimitBurst=5
StartLimitAction=none
FailureAction=none
SuccessAction=none
InvocationID=a1dfabcccca9457f974791697fb1d844
CollectMode=inactive
```

OddBloke, 2020-01-10:
Oh, and apologies for not reading your request correctly: I haven't enabled `fwupd-refresh.service` manually.

superm1, 2020-01-10:
Thanks for reporting.  Yes, I can reproduce this.  I think there are 3 distinct things going on that we should address.

1. The output is repeated multiple times.  This appears to be because the default behavior is to append the file: https://github.com/fwupd/fwupd/blob/master/src/fu-util.c#L2795.  I think we should change that to wipe the file by default.

2. The logging doesn't indicate what this is.  For someone logging into a system via SSH seeing a tree output doesn't exactly make sense without context.  We probably at a minimum need a header saying something along the lines of "You have firmware updates available for the following devices".   

3. Even though `fwupd-refresh.service` has the preset set to false, `fwupd-refresh.timer` doesn't.
This was intended to be disabled by default, but something that users can opt into.

```
$ systemctl status fwupd-refresh.timer
● fwupd-refresh.timer - Refresh fwupd metadata regularly
   Loaded: loaded (/lib/systemd/system/fwupd-refresh.timer; enabled; vendor preset: enabled)

$ systemctl status fwupd-refresh.service
● fwupd-refresh.service - Refresh fwupd metadata and update motd
   Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: disabled)
```

OddBloke, 2020-01-10:
I've also filed #1699, as this append behaviour seems like it could be fixed independent of addressing the actual content that's getting appended. :-)

superm1, 2020-01-16:
This was fixed in #1700

closed 2020-01-16 by superm1