fwupd/fwupd #1699: MOTD messages are very verbose

OddBloke, 2020-01-10:
When SSHing into my desktop, I was greeted with the following content (repeated several times, but that's captured in #1698) in the MOTD:

```
B450M DS3H
│
└─Unifying Receiver:
  │   Device ID:           cf3685ba249d3d98602047341d6f5a5556a6ac05
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR12.07_B0029
  │   Bootloader Version:  BOT01.02_B0014
  │   Vendor:              Logitech, Inc. (USB:0x046D)
  │   Install Duration:    30 seconds
  │   GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
  │                        fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
  │                        279ed287-3607-549e-bacc-f873bb9838c4
  │   Device Flags:        • Updatable
  │                        • Supported on remote server
  │ 
  ├─Unifying Receiver (RQR12) Device Update:
  │     New version:       RQR12.10_B0032
  │     Remote ID:         lvfs
  │     Summary:           Firmware for the Logitech Unifying Receiver (RQR12.xx)
  │     License:           Proprietary
  │     Size:              56.8 kB
  │     Vendor:            Logitech
  │     Duration:          30 seconds
  │     Flags:             is-upgrade
  │     Description:       This release addresses an encrypted keystroke injection vulnerability sent by pointing devices. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
  │     
  │     A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
  │     
  │      • Wireless Mouse M335
  │      • Zone Touch Mouse T400
  │      • Wireless Mouse M545
  │      • Wireless Mouse M560
  │      • Touch Mouse M600
  │      • Touch Mouse T620
  │      • Wireless Rechargeable Touchpad T650
  │     
  │     Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
  │   
  └─Unifying Receiver (RQR12) Device Update:
        New version:       RQR12.08_B0030
        Remote ID:         lvfs
        Summary:           Firmware for the Logitech Unifying Receiver (RQR12.xx)
        License:           Proprietary
        Size:              72.7 kB
        Vendor:            Logitech
        Duration:          30 seconds
        Flags:             is-upgrade
        Description:       This release addresses an encrypted keystroke injection issue known as Bastille security issue #13. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
        
        A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
        
         • Wireless Mouse M335
         • Zone Touch Mouse T400
         • Wireless Mouse M545
         • Wireless Mouse M560
         • Touch Mouse M600
         • Touch Mouse T620
         • Wireless Rechargeable Touchpad T650
        
        Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
```

Compare this with what I see for available apt updates:

```
14 packages can be updated.
0 updates are security updates.
```

fwupd shouldn't take up more than one or two lines in the MOTD. I would suggest that something like:

```
3 firmware updates are available. See `fwupdmgr get-updates` for details.
```

would be more appropriate. (I don't know if that's exactly what the text should be, but I think a single line pointing people to get more detail is definitely better than including the detail with no context.)

## Boilerplate

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.6
```

```shell
$ fwupdmgr get-devices
<snip>
├─Unifying Receiver:
│     Device ID:           cf3685ba249d3d98602047341d6f5a5556a6ac05
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR12.07_B0029
│     Bootloader Version:  BOT01.02_B0014
│     Vendor:              Logitech, Inc. (USB:0x046D)
│     Install Duration:    30 seconds
│     GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│                          • Supported on remote server
<snip>
```
(Just included the relevant portion here.)

Please answer the following questions:

- Operating system and version: Ubuntu 20.04 (development release)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): it's in the base system, so apt
- Have you tried rebooting? Don't think it would change matters, as this is intentional behaviour.
- Are you using an NVMe disk? No

superm1, 2020-01-10:
> 3 firmware updates are available. See `fwupdmgr get-updates` for details.

That's a very good suggestion.  I like this idea.  I wonder if we should actually leave it enabled by default then as well with this modification.

superm1, 2020-01-16:
This was fixed in #1700

closed 2020-01-16 by superm1