fwupd/fwupd #1704: Requires an installer for the win32 build

hughsie, 2020-01-15:
I've had some ODMs ask me about the win32 builds of fwupd. At the moment we don't have a good story on how to deploy this onto your own system. I've done this before using `makensis` but that kinda implies making a Fedora mingw pacakge for fwupd which seems like a lot of work.

From a 40,00ft point of view, I'm wondering if this is something we *want* to support.

superm1, 2020-01-15:
> I've had some ODMs ask me about the win32 builds of fwupd.

Have you asked: _"Why?"_ yet?  I think understanding that should help drive the direction.
 
>I've done this before using makensis but that kinda implies making a Fedora mingw pacakge for fwupd which seems like a lot of work.

Well making a mingw package is 90% there I think, everything needed ends up in `dist/` as part of CI.  So I would think a simple installer that takes all that and dumps it into `C:\Program Files\fwupd` shouldn't be too bad right?

> From a 40,00ft point of view, I'm wondering if this is something we want to support.

I think that why question should be illustrative.  It's still super power user tool, you need to be an admin to use it - not something that would be able to turn into a "useful" exe on Windows with a clicky GUI.  So if the reason that this is wanted is to help transform it into that, then yes i think it makes sense.

Maybe a crawl walk run approach is to capture `dist/` on the mingw build and turn it into a zip file on the next fwupd release.  Let folks use that and play with it.  If it's useful for someone then maybe an installer later after that.

superm1, 2020-01-16:
GUID verification is a good idea, I've heard the same. Perhaps we'll need to start trying to use Windows API for other devices than USB to at least read GUID at some point.

They used fwupd on Linux to verify OEM block on Dell NVME and actually discovered an endian bug in the implementation!

closed 2020-01-16 by superm1