fwupd/fwupd #1711: "fwupdmgr get-updates" fails with "No upgrades for System Firmware: No releases found: cannot handle firmware requirement 'firmware'"

christianhorn, 2020-01-18:
```shell
$ fwupdmgr --version
client version: 1.3.6
compile-time dependency versions
        gusb:   0.3.1
        efivar: 37
daemon version: 1.3.6
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,0000,0018,0019,001A,001B,001C,001D,001E,001F
Boot0000* Windows Boot Manager  HD(1,GPT,8bd14393-8c07-44bd-9afe-738bda5ce430,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...B................
Boot0001* Fedora        HD(1,GPT,8bd14393-8c07-44bd-9afe-738bda5ce430,0x800,0x82000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information        FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001C* ATA HDD2      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001D* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001E* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001F* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0020* USBR BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
$
```

```shell
$ tree /boot |egrep 'fwup|cap'
$
```

Please answer the following questions:

- Operating system and version: Fedora 31
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf, Fedora repos
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes, see below
- Is secure boot enabled (only for the UEFI plugin)? no

# fwupdmgr get-devices 
20LS001AGE
│
├─SAMSUNG MZVLB256HAHQ-000L7:
│     Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:             NVM Express Solid State Drive
│     Current version:     0L2QEXD7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     Serial Number:       S41GNF1K634417
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                          47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                          9657ce89-450f-58d2-ade0-2c6ae667541a ← SAMSUNG MZVLB256HAHQ-000L7
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1917c35d1d47dfb7126aaee11348c0a565cdc9c9
│     Current version:     0.1.35
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                f0fad8e6-0d0c-4743-8ffd-16e0918c998e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9466b55cd6b7537d5dc3890f61bc780266d5c5f6
│     Current version:     184.50.3425
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                9de55de6-0a81-48fc-a6c9-0e13b38bdc94
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           e6948cf157248078c46bf3ee2888f46a414fe583
      Current version:     0.1.29
      Minimum Version:     0.0.1
      Vendor:              DMI:LENOVO
      GUID:                954070c3-1f4b-49ac-9c98-8dd5b8e49bc0
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

hughsie, 2020-01-20:
Debugging now.

hughsie, 2020-01-20:
This was an incorrect requirement added by Lenovo on the LVFS. I've fixed it up now, and in two hours the metadata will regenerate.

christianhorn, 2020-01-20:
That works:
```shell
[root@電脳 ~]# fwupdmgr refresh; fwupdmgr get-updates
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 1 local device supported
No upgrades for System Firmware: current version is 0.1.35: 0.1.29=older
[root@電脳 ~]# 
```
I conclude that the firmware metadata for the L480 is not up2date.. but that's a different topic. Thanks!

hughsie, 2020-01-20:
> I conclude that the firmware metadata for the L480 is not up2date

There's https://github.com/fwupd/missing-firmware-lenovo-thinkpad/issues if that helps.