fwupd/fwupd #1712: fwupd power usage

nyalldawson, 2020-01-18:
I've read up on some of the earlier reports on fwupd power usage, but none seem to describe my situation.

On a brand new, updated, Fedora 31 install on a Dell Precision 5540, I'm consistently seeing powertop report fwupd usage of 2-3W:

```
The battery reports a discharge rate of 30.4 W
The power consumed was 610 J
The estimated remaining time is 2 hours, 4 minutes

Summary: 2522.0 wakeups/second,  0.0 GPU ops/seconds, 0.0 VFS ops/sec and 20.0% CPU use

Power est.              Usage       Events/s    Category       Description
  3.86 W     21.0 ms/s     969.0        Timer          tick_sched_timer
  2.92 W     20.0 ms/s     728.5        Process        [PID 25506] /usr/libexec/fwupd/fwupd
```

(yes - the overall battery usage on this is extreme (30W!!!) -- that's ultimately what I'm trying to get to the bottom of)

This happens all the time, and reboots/etc don't help.

fwupd --verbose reports:
```
22:55:02:0690 FuDebug              Verbose debugging enabled (on console 1)
22:55:02:0690 FuConfig             loading config values from /etc/fwupd/daemon.conf
22:55:02:0691 FuRemoteList         loading /usr/share/fwupd/metainfo
22:55:02:0691 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
22:55:02:0691 XbSilo               file: eb7cf309-93a9-2200-889b-734593bd66cb, current:eb7cf309-93a9-2200-889b-734593bd66cb, cached: (null)
22:55:02:0691 XbSilo               loading silo with file contents
22:55:02:0691 Fwupd                looking for os-release at /etc/os-release
22:55:02:0691 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
22:55:02:0691 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
22:55:02:0691 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
22:55:02:0691 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
22:55:02:0691 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
22:55:02:0691 FuRemoteList         ignoring unfound remote fwupd
22:55:02:0691 FuRemoteList         ordering lvfs-testing=lvfs+1
22:55:02:0691 FuRemoteList         ignoring unfound remote fwupd
22:55:02:0691 FuRemoteList         ignoring unfound remote fwupd
22:55:02:0691 FuRemoteList         ignoring unfound remote fwupd
22:55:02:0696 FuEngine             client certificate exists and working
22:55:02:0696 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
22:55:02:0696 FuHistory            got schema version of 5
22:55:02:0696 FuIdle               setting timeout to 7200s
22:55:02:0696 FuHwids              smbios property Manufacturer=Dell Inc.
22:55:02:0696 FuHwids              smbios property EnclosureKind=a
22:55:02:0696 FuHwids              smbios property Family=Precision
22:55:02:0696 FuHwids              smbios property ProductName=Precision 5540
22:55:02:0696 FuHwids              smbios property ProductSku=0906
22:55:02:0696 FuHwids              smbios property BiosVendor=Dell Inc.
22:55:02:0696 FuHwids              smbios property BiosVersion=1.5.0
22:55:02:0696 FuHwids              smbios property BiosMajorRelease=01
22:55:02:0696 FuHwids              smbios property BiosMinorRelease=05
22:55:02:0696 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
22:55:02:0696 FuHwids              smbios property BaseboardProduct=0FMYX6
22:55:02:0698 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
22:55:02:0698 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
22:55:02:0698 XbSilo               file: b272695d-2060-ce80-090e-d1916d84f790, current:b272695d-2060-ce80-090e-d1916d84f790, cached: (null)
22:55:02:0698 XbSilo               loading silo with file contents
22:55:02:0698 FuEngine             remote lvfs-testing not enabled, so skipping
22:55:02:0698 FuEngine             remote vendor not enabled, so skipping
22:55:02:0698 FuEngine             remote vendor-directory not enabled, so skipping
22:55:02:0698 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
22:55:02:0698 XbSilo               0.01ms      parse blob
22:55:02:0698 XbSilo               0.03ms      loaded file
22:55:02:0698 XbSilo               file: 50caad2c-48e5-da38-a59d-326511ccb5dd, current:50caad2c-48e5-da38-a59d-326511ccb5dd, cached: (null)
22:55:02:0698 XbSilo               loading silo with file contents
22:55:02:0698 XbSilo               0.00ms      parse blob
22:55:02:0698 XbSilo               0.15ms      query on / with `components/component` limit=0 -> 356 results
22:55:02:0698 FuEngine             356 components now in silo
22:55:02:0698 XbSilo               0.15ms      query on / with `components/component/provides/firmware` limit=0 -> 414 results
[... 575 lines cut ...]
22:55:03:0708 FuHistory            get device
22:55:03:0714 FuEngine             31 devices with subsystem pci
22:55:03:0723 FuDevice             using 71b677ca0f1bc2c5b804fa1d59e52064ce589293 for PCI_SLOT_NAME=0000:01:00.0
22:55:03:0723 FuPlugin             emit added from optionrom: 71b677ca0f1bc2c5b804fa1d59e52064ce589293
22:55:03:0724 FuPlugin             performing fu_plugin_device_registered() on uefi
22:55:03:0724 FuPlugin             performing fu_plugin_device_registered() on thunderbolt_power
22:55:03:0724 FuPlugin             performing fu_plugin_device_registered() on dell
22:55:03:0724 FuDeviceList         ::added 71b677ca0f1bc2c5b804fa1d59e52064ce589293
22:55:03:0725 XbSilo               1.40ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='43948b38-3adb-5c42-af77-8619f6b96f4e']/../..|components/component/provides/firmware[@type='flashed'][text()='32757a2c-105b-5e6c-9566-cd704c90f081']/../..` limit=1 -> 0 results
22:55:03:0725 FuHistory            get device
22:55:03:0728 FuDevice             using bbbf1ce3d1cf15550c3760b354592040292415bb for PCI_SLOT_NAME=0000:00:02.0
22:55:03:0729 FuPlugin             emit added from optionrom: bbbf1ce3d1cf15550c3760b354592040292415bb
22:55:03:0729 FuPlugin             performing fu_plugin_device_registered() on uefi
22:55:03:0729 FuPlugin             performing fu_plugin_device_registered() on thunderbolt_power
22:55:03:0729 FuPlugin             performing fu_plugin_device_registered() on dell
22:55:03:0729 FuDeviceList         ::added bbbf1ce3d1cf15550c3760b354592040292415bb
22:55:03:0730 XbSilo               1.40ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='db2a81f9-8177-5e23-8c0b-809f7b3314e3']/../..|components/component/provides/firmware[@type='flashed'][text()='db677673-1102-5481-a11e-6f397e123ced']/../..` limit=1 -> 0 results
22:55:03:0730 FuHistory            get device
22:55:03:0796 FuCommon             not checking 1.4.0 as no version format set
22:55:03:0796 FuEngine             Emitting PropertyChanged('Status'='idle')
22:55:03:0796 FuMain               Emitting PropertyChanged('Status'='idle')
22:55:03:0799 FuMain               Started with locale en_AU.UTF-8
22:55:03:0799 FuMain               Daemon ready for requests
22:55:03:0800 FuMain               FuMain: acquired name: org.freedesktop.fwupd
```

superm1, 2020-01-19:
> 30W

As a general statement for high power consumption at runtime you should check that you have runtime PM set for all your devices in the machine.

The Ubuntu based OEM image does a lot of tuning in this area that doesn't generally happen automatically in general purpose distributions since it can't be safely applied to all machines. If you still have it around you can dig around for finding some specifics. The Ubuntu bssed OEM image does meet the energy star specifications typically.

However using something like `tlp` will get you most of the way there.

> 5540 

For upstream kernels (instead of the OEM tuned kernel) make sure you are using 5.4 or later. There are some significant changes in NVME power management that will save you 2-3W per NVME drive.

You should also look into using the binary nvidia driver and/or turning off discrete GPU with something like bumblebee when not in use.

> fwupd 2W

This is pretty surprising to me. Once it's checked for updates on all the devices the daemon should be remaining idle.

I just checked on another machine of mine with master and at idle nothing from fwupd shows up on the list of top consumers. The biggest are backlight in the order of watts but network, chrome and gnome shell in the order of mW and everything else (including fwupd) in uW or 0W.

Fwupd specifically I see at 0W.

> Ideas

As a hunch - can you please try to remove tpm2-abrmd from your system and try again? I notice it has lots of verbose logging and that is a notable difference in our tests. I'm not using that tcti.

nyalldawson, 2020-01-20:
> However using something like tlp will get you most of the way there.

I'm already running tlp, unfortunately it only gave a minimal drop in the usage...

> For upstream kernels (instead of the OEM tuned kernel) make sure you are using 5.4 or later. 

5.4.10 here :+1: 

> You should also look into using the binary nvidia driver and/or turning off discrete GPU with something like bumblebee when not in use.

That's already done too

> As a hunch - can you please try to remove tpm2-abrmd from your system and try again? I notice it has lots of verbose logging and that is a notable difference in our tests. I'm not using that tcti.

Unfortunately removing tpm2-abrmd hasn't given any noticeable difference

hughsie, 2020-01-20:
> 22:55:03:0800 FuMain               FuMain: acquired name: org.freedesktop.fwupd

There's no output other than this? If not, then fwupd is idle, and not doing anything. I assume running `top` doesn't show fwupd as "doing" anything?

hughsie, 2020-01-20:
I guess this is https://github.com/hughsie/libgusb/pull/25 but powertop is insane if it thinks that's 3W...

superm1, 2020-01-20:
Maybe can you make an rpm with that patch for OP to confirm it helps?

superm1, 2020-01-21:
Reading through that linked PR, I'd say it probably was that.

@nyalldawson it might not be immediately obvious but I would suspect that would prevent the USB controllers from going into runtime PM.  USB controllers not going into runtime PM will prevent PCH from reaching the deeper package states.  Not reaching the deeper package states will show quite a bit of power consumption.

superm1, 2020-01-22:
In the duplicate issue it www confirmed that patch ing libgusb helped fwupd spinning.

So yes that number of 3W was likely correct.

closed 2020-01-22 by superm1