fwupd/fwupd #1713: 8Bitdo SN30 Pro+ update - device id not found

mrt181, 2020-01-19:
Can´t update the controller to 3.01.

When I download the file directly 
https://fwupd.org/downloads/da605483fbc6c7b2bfdddfdb5197c3b83e7609d4-8Bitdo-SN30_Pro+-3.01.cab
and open it with GNOME software I get this message:

> Unable to update "SN30 Pro+":
> device ID 4aa5d06234e4d30aec453144bbe944d89a614c7c was not found

```shell
$ fwupdmgr refresh                                              
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 0 local devices supported
$ fwupdmgr update 
No updatable devices
```

```shell
$ fwupdmgr --version
client version: 1.3.6
compile-time dependency versions
        gusb:   0.2.11
        efivar: 34
daemon version: 1.3.6```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

This output is missing one ssd drive
```shell
$ fwupdmgr get-devices
HP EliteBook 1050 G1
│
├─Thunderbolt Controller:
│     Device ID:           6fd4a8a5fa3b97d110a3b13e5a61ec5534756690
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     35.00
│     Vendor:              HP, Inc. (TBT:0x00F0)
│     GUIDs:               0abe9964-e2bf-5059-8c47-a14aad943b16
│                          997da4b3-ed66-508f-afd6-29c8285c5451
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Device stages updates
│   
├─8BitDo SN30 Pro+:
│     Device ID:           73ef80b60058b4f18549921520bfd94eaf18710a
│     Summary:             A redesigned classic game controller
│     Current version:     1.02
│     Vendor:              8Bitdo (USB:0x2DC8)
│     Install Duration:    2 minutes
│     GUIDs:               043c7673-bd45-5e2e-b8b9-c22df2dfec2a
│                          b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
│                          64eb3e9c-e79c-50d0-984b-21a1641f37c5
│                          a9a82ee0-7fdc-5d80-bd73-6b39f7a4f061
│     Device Flags:        • Updatable
│                          • Requires a bootloader
│                          • Device will not re-appear after update completes
│   
├─System Firmware:
│     Device ID:           123fd4143619569d8ddb6ea47d1d3911eb5ef07a
│     Current version:     Q72 Ver. 01.08.00
│     Vendor:              HP
│     Update Error:        Firmware can not be updated in legacy mode, switch to UEFI mode
│     GUID:                230c8b18-8d9b-53ec-838b-6cfc0383493a
│     Device Flags:        • Internal device
│                          • Requires AC power
│                          • Needs a reboot after installation
│   
├─TS256GMTS400:
│     Device ID:           e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:             ATA Drive
│     Current version:     N1126KB
│     Vendor:              Intel Corporation
│     GUIDs:               fcdb38c6-e2a4-5cdf-8b73-9b2b84ad116f
│                          771fa184-26bd-541e-9e02-3bb6a635a620
│                          e3ec7c1e-dfc5-5e11-9610-10c49c1007e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Touchpad:
      Device ID:           771fb77e7bbe357c1979d364d8b6354b6d0f03e0
      Current version:     1.7.2763040
      Bootloader Version:  8.0
      Vendor:              Synaptics (HIDRAW:0x06CB)
      GUIDs:               a716eeeb-c9e2-5b72-90fb-218dd0d75933
                           8a12a4a7-0c08-5978-92b7-59271c9dc982
                           e46412fa-15e1-57a0-b500-64c4e7fe5762
                           5e4ba50a-1593-5397-b328-57e2a98b127a
      Device Flags:        • Internal device
                           • Updatable
    
```

Please answer the following questions:

- Operating system and version:
```shell
Distributor ID: Ubuntu
Description:    Ubuntu 19.10
Release:        19.10
Codename:       eoan
```

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
```shell
sudo snap install fwupd --classic
```
- Have you tried rebooting? `Yes`
- Are you using an NVMe disk? `don't know`
- Is secure boot enabled (only for the UEFI plugin)? `No`

superm1, 2020-01-19:
Can you please run fwupdmgr get-devices with sudo instead? We'll be able to see a little more verbose information about how those GUIDs were built and compare.

mrt181, 2020-01-21:
This is the output from running  `sudo fwupdmgr get-devices`

```shell
├─8BitDo SN30 Pro+:
│     Device ID:           73ef80b60058b4f18549921520bfd94eaf18710a
│     Summary:             A redesigned classic game controller
│     Current version:     1.02
│     Vendor:              8Bitdo (USB:0x2DC8)
│     Install Duration:    2 minutes
│     GUIDs:               043c7673-bd45-5e2e-b8b9-c22df2dfec2a
│                          b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
│                          64eb3e9c-e79c-50d0-984b-21a1641f37c5 ← USB\VID_2DC8&PID_6002&REV_0001
│                          a9a82ee0-7fdc-5d80-bd73-6b39f7a4f061 ← USB\VID_2DC8&PID_6002
│     Device Flags:        • Updatable
│                          • Requires a bootloader
│                          • Device will not re-appear after update completes
```

superm1, 2020-01-21:
Well the GUID is the proper one; that's the one fwupd expects to use.  The CAB file you're referencing however is using a different GUID:
```
    <!-- USB\VID_2DC8&PID_5750&REV_020 -->
    <firmware type="flashed">78acf201-a624-5c0b-a521-e204be14e3d5</firmware>
    <!-- USB\VID_2DC8&PID_5750 -->
    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
```

It's tied specifically to the bootloader, which seems like a mistake:
https://github.com/fwupd/fwupd/blob/master/plugins/ebitdo/ebitdo.quirk#L5

mrt181, 2020-01-21:
Ok, so is that the reason why there is no updateble device found?
```shell
$ fwupdmgr refresh                      
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Successfully downloaded new metadata: 0 local devices supported
```

superm1, 2020-01-21:
Yes I think it's a metadata error. Let @hughsie confirm.

hughsie, 2020-01-22:
Agree, it shouldn't just list the bootloader. I'm away from home at the moment, so don't have any 8bitdo stuff handy.

superm1, 2020-01-22:
Can you tear that out of stable at least while away from home?  I think that should go back to embargo and we can re-promote the right thing.