fwupd/fwupd #1715: X390 fails to install ThinkPad T490s Corporate ME Update

foshaug, 2020-01-20:
My system is unable to install the "ThinkPad T490s Corporate ME Update". 
Every time I try to install it the system offers the same update again. 

```shell
$ fwupdmgr get-updates
UEFI Device Firmware has firmware updates:
GUID:                    aeac19da-0c78-4b86-a718-1a043947d83b
ID:                      com.lenovo.ThinkPadN2JRM.firmware
Update Version:          192.47.1524
Update Name:             ThinkPad T490s Corporate ME Update
Update Summary:          Lenovo ThinkPad T490s Corporate ME Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(afa2196c27376214d475213bfec45570aa260ab3)
Update Location:         https://fwupd.org/downloads/b2c442db905872d9d4725a038b29450edaea8361-Lenovo-ThinkPad-T490s-CorporateMEFirmware-12.0.47.1524.cab
Update Description:       • 0 Q2'19 Intel Platform Update (Hot Fix Release)
                         
                         Version 12.0.47.1524 (LVFS: 192.47.1524)
                         
                         Problem Fixes
                          • Intel CSME PSIRT-TA-201905-011 Q2 2019 IPU:
                          • Mitigated the following security vulnerabilities under issues:
                         
                         Security issues fixed:
                          • CVE-2019-0086
                          • CVE-2019-0090
                          • CVE-2019-0092
                          • CVE-2019-0093
                          • CVE-2019-0094
                          • CVE-2019-0096
                          • CVE-2019-0097
                          • CVE-2019-0098
                          • CVE-2019-0153
                          • CVE-2019-0170
No upgrades for UEFI Device Firmware, current is 0.1.15: 0.1.15=same, 0.1.12=older
No upgrades for 20Q0CTO1WW System Firmware, current is 0.1.51: 0.1.51=same
```

[image]
[image]

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X390/T490s Thunderbolt Controller
  DeviceId:             d13dc094e8febb10211a5395d0961b529c340495
  Guid:                 e773c51e-a20c-5b29-9f09-6bb0e0ef7560
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              18.00
  Icon:                 computer
[... 381 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-37-generic
├── initrd.img-5.3.0-26-generic
├── lost+found
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-37-generic
├── System.map-5.3.0-26-generic
├── vmlinuz-5.0.0-37-generic
└── vmlinuz-5.3.0-26-generic

10 directories, 297 files
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04.3 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Came installed
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Not sure - Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

neuged, 2020-01-22:
I think I have the same issue. This update just won't stick. Also on an X390, Ubuntu 19.10.

In case it is interesting, here is [a pastebin with additional info like the OP provided it](https://pastebin.com/9f7qEFVw).

Tried the update both with secure boot enabled and disabled.

The output on startup is a bit different for me however than the photos above. I get:

```
Found update fwupd-ux-capsule
Found update fwupd-ef5cdc85-9cf6-469d-9cb7-920b7dd6672b-0
Adding new capsule
Adding new capsule
Reset system
```

Then darkness and afterwards an "Installing new firmware..." screen.

Aberrantfox, 2020-01-24:
Precisely the same boat here, sadly.