fwupd/fwupd #1718: fwupd spins in a poll(2) loop, consuming CPU

joakim-tjernlund, 2020-01-22:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.6

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20L7004MMX
│
├─Thunderbolt Controller:
│     Device ID:           ab7d5105abea00e669b468df7882afc0b34e133e
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               061f2847-061b-5cfb-9c39-3dbfe2b49d69 ← TBT-01091708-native
│                          0cdc303c-b1f9-54a1-8c67-c8eb9b20b6d3 ← TBT-01091708-native-0000:06:00.0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─SanDisk SD7SN6S-256G-1006:
│     Device ID:           64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Summary:             ATA Drive
│     Current version:     X3511006
│     Vendor:              Intel Corporation
│     Serial Number:       162587402065
│     GUIDs:               d1dd479e-36e4-5617-bae8-5d0c8a4b0a6c ← IDE\SanDisk_SD7SN6S-256G-1006_______________X3511006
│                          93c93663-6f6d-5833-9e59-373a750ef51b ← IDE\0SanDisk_SD7SN6S-256G-1006_______________
│                          b28f2002-d07a-5f21-96bf-be0e0833fb7b ← SanDisk SD7SN6S-256G-1006
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           e7cc234e82d08b6a14e246beb6b5f8142091e570
│     Current version:     0.1.37
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                ebfe8df8-dee7-4692-a721-cbcf5095c5cf
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
│     Current version:     184.70.3626
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                676af093-2a5c-4238-9c29-db8063a33532
│     Device Flags:        • Internal device
[... 105 lines cut ...]
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
[pid  2247] 19:31:09 poll([{fd=10, events=POLLIN}, {fd=12, events=POLLIN}], 2, 1) = 0 (Timeout)
....

So why has fwupd got stuck in 1 ms timeout loop?

joakim-tjernlund, 2020-01-22:
Found https://github.com/hughsie/libgusb/pull/25 and patched my libgusb with 5 sec tmo
and the loop was gone :)

superm1, 2020-01-22:
Dup of #1712

closed 2020-01-22 by superm1