fwupd/fwupd #1722: chained fw update fails on wrong metainfo file

cnolan3, 2020-01-23:
- OS: Fedora 31
- Installed through copr, fwupd build #1137816 (fwupd-1.3.6-999.544.20200103git.fc31.src.rpm)

When testing a chained firmware update it fails when checking for the wrong current firmware version. My .cab file has six files in it, three firmware files and one metainfo xml for each.

multi_test.cab:
- FW_A.bin
- FW_A.metainfo.xml
- FW_B.bin
- FW_B.metainfo.xml
- FW_C.bin
- FW_C.metainfo.xml

The drive I'm testing has fw version L02C and needs to go to version BR003C (FW_A.bin) then to version 2262 (FW_B.bin) and finally to version 004C (FW_C.bin). My FW_A.metainfo.xml contains a <requires> tag, requiring that the current firmware on the drive be L02C.

```xml
<requires>
    <firmware compare="eq" version="L02C"/>
</requires>
```
and FW_B.metainfo.xml has a similar line requiring the current firmware version to be BR003C.
When I run the install command, it fails on the firmware comparison because it's using the comparison from FW_B.metainfo.xml instead of the comparison from FW_A.metainfo.xml, the current fw on the drive is L02C and it's failing on the check for version BR003C.

```shell
$fwupdmgr install multi_test.cab
Not compatible with firmware version: failed predicate [BR003C eq L02C]
```

Seems to me that the tool is running the checks from all metainfo files in the cab file at once and fails if any of the comparisons are false, but I may also be misunderstanding something about how chained updates are designed to work.

superm1, 2020-01-23:
A few things:
1. This device needs to have the quirk `install-all-releases` set
2. You need a very recent fwupd build, can you please confirm that you have https://github.com/fwupd/fwupd/commit/af140735220cb769f398cdf346af23a96c4b7a51 in your build?  I would guess based on that date it doesn't, but please confirm.

superm1, 2020-01-23:
(That commit landed on Jan 7, 2020)

cnolan3, 2020-01-23:
ah yeah I forgot to mention that I added ```install-all-releases``` to a .quirk file that I added to /usr/share/fwupd/quirks.d/

```
[DeviceInstanceId=NVME]
Plugin = nvme

[DeviceInstanceId=NVME\VEN_8086&DEV_F1A6]
Flags = force-align,needs-shutdown,install-all-releases
```

hopefully that was the correct way to do that.

Yeah the version of fwupd may be the issue actually, looking at the fwupd copr page it looks like the build that I'm using was run on the 6th, sorry I should've realized that. Is there a way for me to easily get a build of the current code or do I need to build it myself?

superm1, 2020-01-23:
I always build by ✋, I'm not sure how those builds get made. @hughsie, can you get a new copr built somehow for @cnolan3 ?

hughsie, 2020-01-23:
@cnolan3 my apologies, I've been away on training for the last few days. I've just done a fwupd of git master in the copr just now.

cnolan3, 2020-01-23:
@hughsie no problem! as always thanks for your help! I'll close this since it's not actually an issue

superm1, 2020-01-23:
Well make sure it actually works :) you are the first tester of the feature I believe.

cnolan3, 2020-01-23:
got ya, yeah I'll reopen if the same issue pops up again

cnolan3, 2020-01-24:
Not the same issue, but after updating to the latest build the fwupdmgr commands can't connect to the daemon
```shell
$ sudo fwupdmgr get-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```
so I checked the fwupd daemon service and it's failing to start

```shell
$ sudo service fwupd status
Redirecting to /bin/systemctl status fwupd.service
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: failed (Result: exit-code) since Fri 2020-01-24 10:56:19 EST; 4min 12s ago
     Docs: https://fwupd.org/
  Process: 3617 ExecStart=/usr/libexec/fwupd/fwupd (code=exited, status=226/NAMESPACE)
 Main PID: 3617 (code=exited, status=226/NAMESPACE)
      CPU: 8ms

Jan 24 10:56:19 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jan 24 10:56:19 localhost.localdomain systemd[3617]: fwupd.service: Failed to set up mount namespacing: /run/systemd/unit-root/run/motd.d: Permission denied
Jan 24 10:56:19 localhost.localdomain systemd[3617]: fwupd.service: Failed at step NAMESPACE spawning /usr/libexec/fwupd/fwupd: Permission denied
Jan 24 10:56:19 localhost.localdomain systemd[1]: fwupd.service: Main process exited, code=exited, status=226/NAMESPACE
Jan 24 10:56:19 localhost.localdomain systemd[1]: fwupd.service: Failed with result 'exit-code'.
Jan 24 10:56:19 localhost.localdomain systemd[1]: Failed to start Firmware update daemon.
```
tried changing permissions on the files that the status mentioned but it didn't help.

I can make this a new issue if you would prefer that rather than just having it here, unless there's just something obvious that I'm missing.

superm1, 2020-01-24:
Can you show the output of `ls -R /run/systemd/unit-root/run`?  This looks like some fallout from https://github.com/fwupd/fwupd/commit/d81ea2e3fc0de9cf73a63fcd9f7360e54e03ed31 that we need to sort out.

cnolan3, 2020-01-24:
looks like /run/systemd/unit-root exists on my system, but /run/systemd/unit-root/run does not, I just tried creating run/ and then restarting the service and got the same message. 
Running ```ls -Rlh /run/systemd/unit-root/run``` after adding that directory and attempting to restart the service gives
```
/run/systemd/unit-root/run/:
total 0
```

superm1, 2020-01-24:
OK well for now I think you can comment out `RuntimeDirectory` in the systemd unit and set `UpdateMotd=false` in `/etc/fwupd/daemon.conf` to turn off this feature while we figure out what's wrong with it on F31.

@hughsie can you repro?  I haven't seen this yet on my side anywhere with master.

cnolan3, 2020-01-24:
ha alright making some progress, commenting that out and setting that to false worked, the daemon now runs, but I'm actually still getting the same comparison issue that I originally filed this issue for, it's failing on the check for the current firmware in the wrong metainfo xml

hughsie, 2020-01-27:
> The drive I'm testing has fw version L02C and needs to go to version BR003C (FW_A.bin) then to version 2262 (FW_B.bin) and finally to version 004C (FW_C.bin). My FW_A.metainfo.xml contains a tag, requiring that the current firmware on the drive be L02C.

I noticed this on the plane. The version numbers are actually `L02C`, `BR003C`, `2262` and `004C`? If so they don't look much like version numbers to me. For NVMe and ATA we kinda expect the hardware isn't going to be using `1.2.3.4` style version numbers, but we do expect that the firmware updates can be `strcmp'd` to get the correct order. e.g. `BR003C` -> `BR003D` -> BR004A` would be more what I expected...

cnolan3, 2020-01-27:
Ah I see, yes, it's a bit annoying but our fw version numbers are generally not perfectly sequential, especially when it comes to the middle firmwares.

hughsie, 2020-01-27:
Is there anything we can use instead, i.e. a build date or anything like that? @superm1 ideas?

superm1, 2020-01-27:
Or anything in the OEM embedding area that can be used?

superm1, 2020-01-30:
@cnolan3 what's the scope of the need for this?  Are we talking a one time occurrence for a handful or less of drives?  Or is this going to be every single drive ever, and will potentially be needed to be in the future and very scalable?

superm1, 2020-02-03:
@cnolan3  (or anyone else watching). I suspect the failure in https://github.com/fwupd/fwupd/issues/1722#issuecomment-578260692 was the same one mentioned in https://github.com/fwupd/fwupd/issues/1738 which is caused by SELinux policies.

closed 2020-03-12 by stale[bot]